arendst/Tasmota #833: Soundoff Touch double tap

laurentdinclaux, 2017-09-05:
Hello,

I use firmware 5.6.1 20170818 on a Sonoff Touch and Home Assistant.

I want double tap to not trigger the relay but send a specific MQTT message, is that possible ?

The use case: the sonoff touch controls a smart bulb. I want it to turn on/off on single tap, and to toggle between two dim light levels on double tap

PeggyFree, 2017-09-05:
Yes it is possible, I use the double tap for dim light and long press for
the blinds. Search for it in the previous issues, it is explained.

laurentdinclaux, 2017-09-05:
I did and didn't found, thus why I am asking

ionciubotaru, 2017-09-05:
look for `setoption11` Button single and double press functionality swapped
and `buttontopic`

closed 2018-05-07 by stale[bot]