arendst/Tasmota #834: [REQUEST] Support for the new Sonoff RF Bridge 433

Gamester17, 2017-09-05:
Request firmware support for the new Sonoff RF Bridge 433

https://www.itead.cc/sonoff-rf-bridge-433.html

WiFi to 433MHz converter and vice versa to convert 433MHz RF sensors and remote to App via WiFi

https://www.itead.cc/wiki/Sonoff_RF_Bridge_433

Based on ESP8285 (ESP8266 chipset) from Itead.

https://www.youtube.com/watch?v=wTqQ_cmweuo

arendst, 2017-09-05:
See readme and wiki

mitkodotcom, 2017-09-11:
Is it possible to send an arbitrary command, without first learning it and storing it to a key? As long as as I understood, the 0xA5 command is actually sending the whole sequence (as shown by 0xA4) and not just the key number? If that's right, than it would be possible to use the bridge to send more than 16 commands.

And some feedback: some "remotes" (for example ARC protocol via rfxcom) may need three consecutive transmissions of the same code in order for it to be detected and learned by the bridge.

gbrault, 2017-09-11:
I have been successful reading remote "keys", I have implemented the 0xA5 command but not sure it does anything. I posted a query to Itead to have a clarification about the protocol...

arendst, 2017-09-13:
Tasmota supports default key codes "out of the box" for all 16 keys without learning any key.

Tasmota also supports learning any key as long as the hardware supports the RF code.

If you want more default key codes you can change the "home" part of the code using undocumented command RFDefault  ABCD where ABCD is a four digit hexadecimal number.

mitkodotcom, 2017-09-14:
The default codes are suitable only for devices supporting Lighting4 protocol. They can not be used for controlling devices that only support Lighting1 or Lighting2. For those devices (virtually all devices support Lighting1 / ARC), the bridge has to be "taught" using original remote control.

What if you don't have a remote control? 

I managed to "teach" the bridge by sending command from RFXcom, but it s not very convenient to use a $100 device to program e $10 device.

What about one of those solutions:

- changing all 8 bytes of sfb_code[0] with RFDefault, not just the last 2 (adding something like if (16 == data_len) after if (4 == data_len) )

- directly storing the 8 bytes to the key with something like:

rfkey3 001122334455667788

- directly sending the 8 bytes with something like:

rfsend 001122334455667788

Example of device that can not be controlled by default codes: https://getdio.com/en/smart-lights/30-smart-plug-5411478547853.html

What's worse, it's original remote control is using AC protocol which is NOT supported by Sonoff RF bridge. This way the smart plug can not be controlled by Tasmota with default codes, and the original remote can NOT be used to teach the bridge. However the device is fully compatible with tasmota + sonoff rf, if you only can somehow transmit a suitable code to it. I.e. with an expensive RFXcom. Directly storing/sending the code without first transmitting it over the air will make virtually all RF-devices compatible with this great little and cheap bridge.

gbrault, 2017-09-14:
I got the answer from Itead: they don't want to support flashing the device... And I confirm the device don't transmit RF signal... (I setup the SDR 802T2 dongle to sniff it)

arendst, 2017-10-30:
See additional RF commands added in v5.9.0

closed 2017-10-30 by arendst