arendst/Tasmota #836: Setting up HC-SR501 PIR MOTION DETECTOR with Home Assistant

dawsmac, 2017-09-05:
If this has been covered and the wrong location by all means remove :) 

So I wanted to setup a PIR sensor that would bring on 3 security light in my back garden. I placed 3 sonoff basic inside [HALF LANTERN WALL LIGHT](http://www.screwfix.com/p/asd-60w-white-slave-half-lantern-wall-light/36964) at the top, flashed them all with the Tasmota firmware and added them to HA following the guides online. this all seems to work fine. then came to adding the PIR sensor... I wired it up went onto Web UI and selected the correct setting Switch1 ( GPOI14) etc..
nothing seems to work. I then came across this post regarding a bypass to [run PIR of 3.3 vDC](http://randomnerdtutorials.com/modifying-cheap-pir-motion-sensor-to-work-at-3-3v/)
after this mod PIR started to spring into life. but brought only a single light on and off... 
now because I wanted to control all three and only to come on at night time I needed to understand "Switchtopic". After lots of trying and trying I set the switchtopic in the web UI to "Switchtopic Garage_PIR" and switchmode "SwitchMode1":1" This meant that every time the PIR flagged a movement it would send a MQTT like this cmnd/Garage_PIR/POWER1 = ON. I then updated my HA configuration to include the following in my sensors section 

- platform: mqtt
  name: "Garage PIR"
  qos: 1
  state_topic: "cmnd/Garage_PIR/POWER1"
  payload_on: "ON"
  payload_off: "OFF"
  device_class: motion

Hopefully, this will help others out there :) 

BTW thanks for making this possible with the Sonoff-Tasmota firmware

arcanum246, 2017-09-18:
Thanks Heaps for the help. Could not figure it out on my own.

meirion, 2017-12-16:
Would you mind posting some details or photos of how you installed the mini pir sensor?

I'm interested in building the same setup. Did you replace the PIR in an existing sensor light? How did you power it, and where did you locate the sonoff? 
Thanks for your help.