arendst/Tasmota #837: Sonoff Basic switch double press

icovada, 2017-09-05:
Hello
Ever since one of the previous updates a double tap on a Sonoff Basic turns on the primary relay even if a secondary relay is configured.

Since I was using the double tap to send commands over MQTT to turn on/off something else is there a way to enable this feature again?

icovada, 2017-09-05:
Found it

As explained in [this](https://github.com/arendst/Sonoff-Tasmota/issues/200#issuecomment-306617207) table all I needed to do was typing these commands in the console

`SetOption1 1
SetOption11 1
ButtonTopic othertopic`

Now when I double tap the button I get a TOGGLE command sent to ButtonTopic, and if I hold it 4 seconds I get a HOLD command

closed 2017-09-05 by icovada

PeggyFree, 2017-09-05:
That's right. This table should be in the wiki.