arendst/Tasmota #838: Integration with Openhab2

robbrown92, 2017-09-05:
Firstly i am new to github so please forgive me if this is posted in the wrong place. 

I am having difficulty getting my sonoff basics with tasmota to connect to my openhab2 server running on a rpi. i cant seem to find a decent source how to lay out my configuration files. 

**Sonoff info**
 
AP1 SSId (RSSI)	VM8335559 (100%)
Hostname	sonoff-1505
IP address	192.168.0.51
Gateway	192.168.0.1
Subnet mask	255.255.255.0
DNS server	192.168.4.100
MAC address	5C:CF:7F:AF:65:E1
 
MQTT Host	192.168.0.22
MQTT Port	1883
MQTT Client &
 Fallback Topic	sonoff-A00
MQTT User	openhabian
MQTT Topic	sonoff
MQTT Group Topic	sonoffs
MQTT Full Topic	tasmota/sonoff/cmnd/
 
Emulation	Belkin WeMo
mDNS Discovery	Enabled
mDNS Advertise	Webserver
`

Also my items file code is 

Switch TABLAMP "Table Lamp" <light> ["SWITCHABLE"] { mqtt=">[mosquitto:tasmota/sonoff/sonoff-A00/cmnd/power:command:*:default], <[mosquitto:tasmota/sonoff/sonoff-A00/cmnd/power:state:default]"

agsochi, 2017-09-06:
May be [this tutorial](https://community.openhab.org/t/mqtt-binding-v1-11-getting-started-101/33958) help.

robbrown92, 2017-09-06:
Thanks for the link i followed it all and everything works up until i try to get the switch to turn on and off there must be something wrong with the mqtt command in home.items

Switch TABLAMP "Table Lamp" 
{ mqtt=">[mosquitto:tasmota/sonoff/sonoff-A00/cmnd/power:command:*:default], 
              <[mosquitto:tasmota/sonoff/sonoff-A00/cmnd/power:state:default]"

robbrown92, 2017-09-06:
I have sorted it, I had the topic wrong so if anyone has same issue 

Switch TABLAMP "Table Lamp"
{ mqtt=">[mosquitto:tasmota/sonoff-A00/cmnd/power:command:*:default],
<[mosquitto:tasmota/sonoff-A00/cmnd/power:state:default]"

This fixed it by removing one of the sonoff from the topic

closed 2017-09-06 by robbrown92

thiagoarreguy, 2018-01-09:
A just now put my code to works fine, jus in this moment, so i will share it:
*if you copy this code, make sure to delete the comments -> //read value from device ->  //write value on device.

Switch RoomLight "Room Light" 
{
  mqtt="
              <[mosquitto:stat/RoomLight/POWER:state:ON:1],                      //read value from device
              <[mosquitto:stat/RoomLight/POWER:state:OFF:0],                    //read value from device
              >[mosquitto:cmnd/RoomLight/POWER:command:ON:1],           //write value on device
              >[mosquitto:cmnd/RoomLight/POWER:command:OFF:0]          //write value on device
           "
}