arendst/Tasmota #840: Can't upload all environments

jojost1, 2017-09-06:
I see this:

```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255
========================== [ERROR] Took 14.23 seconds ==========================

 
================================== [SUMMARY] ==================================
Environment sonoff              [SUCCESS]
Environment sonoff-NL           [ERROR]
Environment sonoff-minimal      [ERROR]
Environment sonoff-ds18x20      [ERROR]
========================= [ERROR] Took 100.86 seconds =========================
```

I also have this issue #835 that was not fixed by ee91471.

jojost1, 2017-09-06:
Correction, #835 might be fixed actually. But there is no 'Configure MQTT' button there?

ionciubotaru, 2017-09-07:
```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```
This means you have a problem with FTDI, or with your sonoff
Test your ftdi with a working arduino and check the serial connection
If ftdi works use a different sonoff.

arendst, 2017-09-07:
No mqtt button if mqtt is disabled. Enable via command or button other configuration.

jojost1, 2017-09-07:
> This means you have a problem with FTDI, or with your sonoff
> Test your ftdi with a working arduino and check the serial connection
> If ftdi works use a different sonoff.

I have one FTDI and two sonoffs, both sonoffs have this problem.

What it seems like is that the sonoff restarts after uploading every config, so only the first one it uploads succeeds. When manually uncommenting one of the lines to build/upload only one environment (doesn't matter which of the four) it always succeeds.

> No mqtt button if mqtt is disabled. Enable via command or button other configuration.

MQTT_USE = 1. What else can I do? In the Serial Monitor nothing is stated about MQTT as well..

arendst, 2017-09-07:
Changes to user_config.h have only effect if you change define CFG_HOLDER too as documented in the header of that file.

Apart from that there is no real need to change anything in user_config.h as you can all do this online either with commands as documented in the wiki or via the webinterface (and console).

In your case you may have changed MQTT_USE set to 1 but initially it may have been set to off.

Solution is executing command ``setoption3 1`` or using the web interface select Configure > Configure Other and enable MQTT

davidelang, 2017-09-07:
On Thu, 7 Sep 2017, Joost van den Akker wrote:

> What it seems like is that the sonoff restarts after uploading every config, 
> so only the first one it uploads succeeds.

when sending commands where you want them all to execute before a restart, look 
at the backlog command

closed 2018-05-07 by stale[bot]