arendst/Tasmota #842: Warning regarding FTDI programmers

ulab, 2017-09-07:
My suggestion is to add a warning to the prerequisites page like:

- 3.3V FTDI USB-to-Serial Converter/Programmer (example)
  - Please note that there are FTDI programmers that can be switched between 3.3 V and 5 V for the data pins, but still provide 5V on the power pin. This might fry your chips.

This is the programmer I am talking about:
[image]

Adminius, 2017-09-13:
you can modify it to use 3V3 or 5V on 5V Pin:
[image]

qorron, 2017-10-26:
thanks for the warning!
apparently I was lucky. I flashed about 40 sonoff devices with 5V Vcc.
according to telemetry more like 4.5V.

will modify my flasher.. or get a true 3.3V model.

elbow, 2018-01-09:
Yes - this just happened to me. My cable labeled 3V3 actually connects vcc directly to usb power I.e. 5v.  Unhappy smell from Sonoff, don’t know if it works still.

webtop, 2018-01-09:
I'm not sure about the Sonoff hardware, but the ESP doesn't have a problem with 5v. I flash all my ESPs at 5v, always have done, always will do, and haven't bricked one yet.

Like I said though, the other components on the Sonoff boards may not be 5v tolerant

ulab, 2018-05-10:
I have updated the Wiki Prerequisite page.

closed 2018-05-10 by ulab

stanogustafik, 2021-01-13:
I had the same issue with the same converter and here is the solution: if you switch to 3,3, you also have to repin the cable from 5V to one of the yellow pins in the middle ;)