arendst/Tasmota #843: Show MAC address in Wifi manager mode (4 short presses)

ulab, 2017-09-07:
Since some Wifi networks require you to allow devices using their MAC address, it'd be nice if the Wifi manager did display it.

jetema, 2017-09-08:
It is quite easy to get sonoff MAC address via http, just tap on information ( or type ipaddress/in?  )
as example here's one of my sonoff
[image]

ulab, 2017-09-08:
This only works when the device is already connected to the Wifi in question.

In my case it is not (it can't) and I started the Wifi Manager Mode by pressing the button 4 times. This makes it run its own Wifi access point that you can connect to and enter Wifi username and password. 

But if the Wifi I tried to connect to only allows whitelisted (by mac address) devices. And I had to find it by listing the local machine's arp table which is not something most user are able to.

If the Wifi Manager did show the Mac address directly it'd make this setup a whole lot easier.

closed 2018-05-07 by stale[bot]

badabing2005, 2018-10-27:
It would be nice if it was possible to get the MAC address from console commands.

meingraham, 2018-10-27:
From the Console:
- STATUS 5 for the device MAC
- STATUS 11 for the MAC of the AP the device is connected to

badabing2005, 2018-10-27:
Thank you @meingraham 

Much appreciated.
BB

Cam-ak, 2023-01-24:
@ulab in case you didnt know, The "{IP_ADDRESS}/in?" endpoint does work in the Wifi Manager, you just need to use the IP Address that the Tasmota device sends you to when asking for the wifi credentials. In my case is was 192.168.4.1 ,so the full link would be "http://192.168.4.1/in?"