arendst/Tasmota #844: So off dual can't flash

hwarang098tkd, 2017-09-07:
Please some help . I had flashed sonoff devises all of times . 
So I had a dual for my heater and suddenly stop connecting . It is functionable but no connection with mqtt . The biggest problem is that I can't put it to programming mode to reload the firmware

davidelang, 2017-09-07:
you would put it into programming mode the same way you did origionally 
(connecting gpio0 to ground when starting up), see the wiki for the details of 
where this is (on the dual it's not a nice, easily accessable pin)

hwarang098tkd, 2017-09-07:
Thnx for the interest . I just can't . I follow the instructions properly . I think that just is something wrong and I thinking if there is another way to do it

hwarang098tkd, 2017-09-07:
I just realized that if I let for an hour and connect it with Arduino it works fine. then I connected it with 220 socket and bugged again . Any thoughts?

davidelang, 2017-09-07:
Ok, when you connect a wire between ground and GPIO0, are you holding it to the 
pin on the chip, or the point on the circuit board where a trace goes from one 
side to the other?

hwarang098tkd, 2017-09-07:
to the pin on the chip, as the picture on instructions

closed 2018-05-07 by stale[bot]