arendst/Tasmota #846: Sonoff Pow different telemetry periods?

marcert, 2017-09-08:
Hi,
I am using the Sonoff Pow with fhem. Great device and absolutely great firmware!
One thing I am missing is the possibility to switch to faster telemetry period if the switch is on. 
From my perspective it doesn't make sense to sent lots of MQTT packages if the relay is off (then every 10 minutes are sufficient for me). 
But if I switch on I want to have a more detailed view on the current/Energy and therefore want to have every 10 seconds a package.

Is this possible out of the firmware?

Thanks and Kind Regards,
Marco

andreasloe, 2017-09-10:
Why don't you use
http://<ip-address>/cm?cmnd=Teleperiod
or 
http://<ip-address>/cm?cmnd=Teleperiod%2060
for 60 seconds, for example.

marcert, 2017-09-11:
That is what I am already doing (via MQTT not http). Yes of course this works but it is a little bit crazy that I have to sent two commands every time I switch on or off. So I thought it might be a better idea to have to different periods already stored on the device.
Thanks and Kind Regards,
Marco

ionciubotaru, 2017-09-11:
use `backlog` for multiple commands

marcert, 2017-09-11:
ok. This is also a possibility. So replacing two commands with a double command- have to check how this can be done with fhem.
Nevertheless: from my perspective it is still a better solution to have different periods on the device as I only want to switch on and off and it also doesn't make sense to transfer the telemetry as long as the relay is switched off (then even the voltage is displayed with zero as also all other values of course).
But ok if it doesn't make sense from your side I still can use my workaround.

Thanks and Kind Regards,
Marco

mkh595, 2017-09-11:
Dynamic telemetry period was requested here #688
If previous measure (power, voltage or current) is different from actual, then send telemetry data. If not, use telemetry period as interval.
If I measure voltage or current, I want to see actual values, it has no sense to watch 5 minutes old values.
When switch is off, all measures are 0, therefore no change from previous measure.
I have pumps and UV lamps wich operate base on water temperature. During sommer it is 24 hours, in spring or autumn only during day. But whole night is pow sending data with zeros each 10 seconds.
There was implemented margin to trigger message send but it is unusable because it capture only first trigger.

davidelang, 2017-09-12:
what if you have other sensors connected? under what conditions should the 
period change?

you just cannot make rules that will apply for everyone, so it doesn't make 
sense to use up valuable code space for such a corner case.

marcert, 2017-09-12:
As mkh595 stated there are already a kind of rules available (see request #688). But they are not working as expected (I tried them also without success). So the only difference from todays state is then to add a second timing to the already available rules (like if current is above .5A use secondary period if not primary). That should not be much coding as the rules are already there. 
Maybe they can then be used more general also: like if humidity is below a certain threshold use secondary period.

Kind Regards,
Marco

mkh595, 2017-09-15:
@davidelang: Which sensors do you expect to be connected to POW if configuration of pins is for POW disabled by default?

davidelang, 2017-09-15:
disabled by default != can never be attached

besides, you are asking for timing to be changed depending on the status of a 
relay, that will apply to all sonoff devices, not just the Pow

closed 2018-05-07 by stale[bot]