arendst/Tasmota #847: Sonoff POW resets counters

olenchenko, 2017-09-08:
Hi.
I am using Sonoff POW and I have a following problem: it resets counters (I have noticed reset happens around 6pm - dont know if its relavant).
Firmware version: 5.6.1

Here are some info from console (info sent to MQTT):
```
17:39:26 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T17:39:26", "Total":61.216, "Yesterday":13.407, "Today":41.335, "Period":210, "Power":2304, "Factor":1.00, "Voltage":213, "Current":10.262}
17:44:30 MQTT: tele/sonoff/STATE = {"Time":"2017-09-08T17:44:30", "Uptime":44, "Vcc":3.192, "POWER":"ON", "Wifi":{"AP":1, "SSID":"RADU2", "RSSI":74, "APMac":"16:91:82:37:75:2A"}}
17:44:30 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T17:44:30", "Total":61.413, "Yesterday":13.407, "Today":41.532, "Period":211, "Power":2341, "Factor":1.00, "Voltage":213, "Current":10.289}
17:49:33 MQTT: tele/sonoff/STATE = {"Time":"2017-09-08T17:49:33", "Uptime":44, "Vcc":3.193, "POWER":"ON", "Wifi":{"AP":1, "SSID":"RADU2", "RSSI":78, "APMac":"16:91:82:37:75:2A"}}
17:49:33 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T17:49:33", "Total":61.609, "Yesterday":13.407, "Today":41.728, "Period":211, "Power":2325, "Factor":1.00, "Voltage":212, "Current":10.309}
17:54:37 MQTT: tele/sonoff/STATE = {"Time":"2017-09-08T17:54:37", "Uptime":44, "Vcc":3.194, "POWER":"ON", "Wifi":{"AP":1, "SSID":"RADU2", "RSSI":78, "APMac":"16:91:82:37:75:2A"}}
17:54:37 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T17:54:37", "Total":61.806, "Yesterday":13.407, "Today":41.925, "Period":210, "Power":2325, "Factor":1.00, "Voltage":212, "Current":10.289}
17:59:42 MQTT: tele/sonoff/STATE = {"Time":"2017-09-08T17:59:42", "Uptime":44, "Vcc":3.195, "POWER":"ON", "Wifi":{"AP":1, "SSID":"RADU2", "RSSI":78, "APMac":"16:91:82:37:75:2A"}}
17:59:42 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T17:59:42", "Total":62.004, "Yesterday":13.407, "Today":42.123, "Period":211, "Power":2304, "Factor":1.00, "Voltage":208, "Current":10.583}
18:02:00 MQTT: tele/sonoff/UPTIME = {"Time":"2017-09-08T18:02:00", "Uptime":45}
18:04:48 MQTT: tele/sonoff/STATE = {"Time":"2017-09-08T18:04:48", "Uptime":45, "Vcc":3.196, "POWER":"ON", "Wifi":{"AP":1, "SSID":"RADU2", "RSSI":78, "APMac":"16:91:82:37:75:2A"}}
18:04:48 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T18:04:48", "Total":62.203, "Yesterday":13.407, "Today":42.322, "Period":210, "Power":2335, "Factor":1.00, "Voltage":212, "Current":10.309}
18:09:53 MQTT: tele/sonoff/STATE = {"Time":"2017-09-08T18:09:52", "Uptime":45, "Vcc":3.193, "POWER":"ON", "Wifi":{"AP":1, "SSID":"RADU2", "RSSI":78, "APMac":"16:91:82:37:75:2A"}}
18:09:53 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T18:09:53", "Total":62.401, "Yesterday":13.407, "Today":42.520, "Period":211, "Power":2298, "Factor":1.00, "Voltage":213, "Current":10.236}
18:14:56 MQTT: tele/sonoff/STATE = {"Time":"2017-09-08T18:14:56", "Uptime":45, "Vcc":3.194, "POWER":"ON", "Wifi":{"AP":1, "SSID":"RADU2", "RSSI":76, "APMac":"16:91:82:37:75:2A"}}
18:14:56 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T18:14:56", "Total":62.598, "Yesterday":13.407, "Today":42.717, "Period":211, "Power":2304, "Factor":1.00, "Voltage":210, "Current":10.486}
18:19:59 MQTT: tele/sonoff/STATE = {"Time":"2017-09-08T18:19:59", "Uptime":45, "Vcc":3.195, "POWER":"ON", "Wifi":{"AP":1, "SSID":"RADU2", "RSSI":76, "APMac":"16:91:82:37:75:2A"}}
18:19:59 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T18:19:59", "Total":62.796, "Yesterday":13.407, "Today":42.915, "Period":211, "Power":2342, "Factor":1.00, "Voltage":213, "Current":10.236}
18:25:04 MQTT: tele/sonoff/STATE = {"Time":"2017-09-08T18:25:04", "Uptime":45, "Vcc":3.195, "POWER":"ON", "Wifi":{"AP":1, "SSID":"RADU2", "RSSI":76, "APMac":"16:91:82:37:75:2A"}}
18:25:04 MQTT: tele/sonoff/ENERGY = {"Time":"2017-09-08T18:25:04", "Total":20.044, "Yesterday":13.407, "Today":0.163, "Period":211, "Power":2283, "Factor":1.00, "Voltage":212, "Current":10.410}
```
and some screenshots:
[image]
[image]

I have tried two different Sonoff POWs with the same firmware version and result is the same.
Please let me know how can I get more debug information if needed.

andreasloe, 2017-09-10:
You should first update to 5.7.1 in order to verify that the issue is still existing.

olenchenko, 2017-09-12:
I have upgraded to 5.7.1 as you suggested but issue still persists. Here is MQTT output:
`
21:22:23 MQT: tele/sonoff/STATE = {"Time":"2017-09-12T21:22:23", "Uptime":15, "Vcc":3.195, "POWER":"ON", "Wifi":{"AP":1, "SSId":"RADU2", "RSSI":72, "APMac":"16:91:82:37:75:2A"}}
21:22:23 MQT: tele/sonoff/ENERGY = {"Time":"2017-09-12T21:22:23", "Total":127.437, "Yesterday":12.398, "Today":41.999, "Period":193, "Power":2346, "Factor":1.00, "Voltage":210, "Current":10.403}
21:27:23 MQT: tele/sonoff/STATE = {"Time":"2017-09-12T21:27:23", "Uptime":15, "Vcc":3.196, "POWER":"ON", "Wifi":{"AP":1, "SSId":"RADU2", "RSSI":72, "APMac":"16:91:82:37:75:2A"}}
21:27:23 MQT: tele/sonoff/ENERGY = {"Time":"2017-09-12T21:27:23", "Total":127.631, "Yesterday":12.398, "Today":42.193, "Period":193, "Power":2342, "Factor":1.00, "Voltage":211, "Current":10.356}
21:32:23 MQT: tele/sonoff/STATE = {"Time":"2017-09-12T21:32:23", "Uptime":15, "Vcc":3.194, "POWER":"ON", "Wifi":{"AP":1, "SSId":"RADU2", "RSSI":74, "APMac":"16:91:82:37:75:2A"}}
21:32:23 MQT: tele/sonoff/ENERGY = {"Time":"2017-09-12T21:32:23", "Total":127.825, "Yesterday":12.398, "Today":42.387, "Period":193, "Power":2304, "Factor":1.00, "Voltage":212, "Current":10.356}
21:37:23 MQT: tele/sonoff/STATE = {"Time":"2017-09-12T21:37:23", "Uptime":15, "Vcc":3.193, "POWER":"ON", "Wifi":{"AP":1, "SSId":"RADU2", "RSSI":72, "APMac":"16:91:82:37:75:2A"}}
21:37:23 MQT: tele/sonoff/ENERGY = {"Time":"2017-09-12T21:37:23", "Total":128.019, "Yesterday":12.398, "Today":42.581, "Period":193, "Power":2289, "Factor":1.00, "Voltage":211, "Current":10.309}
21:42:24 MQT: tele/sonoff/STATE = {"Time":"2017-09-12T21:42:24", "Uptime":15, "Vcc":3.196, "POWER":"ON", "Wifi":{"AP":1, "SSId":"RADU2", "RSSI":68, "APMac":"16:91:82:37:75:2A"}}
21:42:24 MQT: tele/sonoff/ENERGY = {"Time":"2017-09-12T21:42:24", "Total":128.214, "Yesterday":12.398, "Today":42.776, "Period":211, "Power":2319, "Factor":1.00, "Voltage":214, "Current":10.190}
21:47:28 MQT: tele/sonoff/STATE = {"Time":"2017-09-12T21:47:28", "Uptime":15, "Vcc":3.196, "POWER":"ON", "Wifi":{"AP":1, "SSId":"RADU2", "RSSI":74, "APMac":"16:91:82:37:75:2A"}}
21:47:28 MQT: tele/sonoff/ENERGY = {"Time":"2017-09-12T21:47:28", "Total":85.461, "Yesterday":12.398, "Today":0.023, "Period":211, "Power":2289, "Factor":1.00, "Voltage":213, "Current":10.329}
21:52:33 MQT: tele/sonoff/STATE = {"Time":"2017-09-12T21:52:33", "Uptime":15, "Vcc":3.196, "POWER":"ON", "Wifi":{"AP":1, "SSId":"RADU2", "RSSI":74, "APMac":"16:91:82:37:75:2A"}}
21:52:33 MQT: tele/sonoff/ENERGY = {"Time":"2017-09-12T21:52:33", "Total":85.658, "Yesterday":12.398, "Today":0.220, "Period":211, "Power":2313, "Factor":1.00, "Voltage":215, "Current":10.210}
21:57:37 MQT: tele/sonoff/STATE = {"Time":"2017-09-12T21:57:37", "Uptime":15, "Vcc":3.195, "POWER":"ON", "Wifi":{"AP":1, "SSId":"RADU2", "RSSI":74, "APMac":"16:91:82:37:75:2A"}}
21:57:37 MQT: tele/sonoff/ENERGY = {"Time":"2017-09-12T21:57:37", "Total":85.854, "Yesterday":12.398, "Today":0.416, "Period":210, "Power":2320, "Factor":1.00, "Voltage":212, "Current":10.164}
`

I dont know if it make sense but I have noticed in both cases today's counter was reseted at value very close to 43kWh.

andreasloe, 2017-09-13:
Well, then I have no idea. You need to search for uptime and Tasmota/sonoff to find out what happens. I do not have the same issue, so it must be related to your config.

closed 2018-05-07 by stale[bot]