arendst/Tasmota #851: Time is not correct

rlust, 2017-09-09:
I have been using this fantastic software for many versions. the last several releases some of my devices are not getting the time set properly.

I am in the US and using TIMEZONE -4
If I just enter timezone I get -4 back but the clock is never getting updated.

Any ideas?

Thanks!

ionciubotaru, 2017-09-09:
try:
```
cmnd/sonoff/ntpserver1
cmnd/sonoff/ntpserver2
cmnd/sonoff/ntpserver3
```
check if the servers are correct and reachable from your network

use `timezone 99` it works better with daylight saving

arendst, 2017-09-10:
Works as expected where my timezone is currently +2:
```
13:56:13 CND: timezone
13:56:13 RSL: Group 0, Index 1, Command TIMEZONE, Data  ()
13:56:13 MQT: stat/wemos5/TIMEZONE = {"Timezone":99}
13:56:18 CND: timezone -4
13:56:18 RSL: Group 0, Index 1, Command TIMEZONE, Data -4 (-4)
13:56:18 MQT: stat/wemos5/TIMEZONE = {"Timezone":-4}
07:56:19 CFG: Saved to flash at FB, Count 75, Bytes 1648
07:57:17 CND: timezone 99
07:57:17 RSL: Group 0, Index 1, Command TIMEZONE, Data 99 (99)
07:57:17 MQT: stat/wemos5/TIMEZONE = {"Timezone":99}
13:57:18 CFG: Saved to flash at FA, Count 76, Bytes 1648
```

rlust, 2017-09-14:
I have run out of ideas for getting 2 of the same POW Sonoffs to set the correct time.
One works and the other does not.
Both are configured the same.
Thanks for any ideas!!

 Working:
16:40:46 CND: status 7
16:40:46 MQT: stat/stereo/STATUS7 = {"StatusTIM":{"UTC":"Thu Sep 14 20:40:46 2017", "Local":"Thu Sep 14 16:40:46 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":-4}}
16:44:54 CND:  cmnd/sonoff/ntpserver1
16:44:54 MQT: stat/stereo/RESULT = {"NtpServer1":"pool.ntp.org"}

Not:
0:00:30 MQT: stat/fan/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:30 1970", "Local":"Thu Jan 01 00:00:30 1970", "StartDST":"Thu Jan 01 00:00:00 1970", "EndDST":"Thu Jan 01 00:00:00 1970", "Timezone":99}}
00:02:00 MQT: tele/fan/UPTIME = {"Time":"1970-01-01T00:02:00", "Uptime":1}
00:00:57 RSL: Group 0, Index 1, Command NTPSERVER, Data  ()
00:00:57 MQT: stat/fan/RESULT = {"NtpServer1":"pool.ntp.org"}

arendst, 2017-09-14:
What is the result of command status 5?

ionciubotaru, 2017-09-15:
One of your sonoffs can reach pool.ntp.org, the other cannot
Check your wifi network, router, firewall

If everything is fine, upload again the software to pow2 and reset it to defaults `reset 1`

rlust, 2017-09-15:
I found the issue THANKS!
It was the DNS address!
Here is the result of status 5 for non working POW module for time:

16:35:41 CND: status 5
16:35:41 RSL: Group 0, Index 1, Command STATUS, Data 5 (5)
16:35:41 MQT: stat/fan/STATUS5 = {"StatusNET":{"Hostname":"fan", "IPAddress":"192.168.1.205", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.1", "Mac":"5C:CF:7F:A0:2C:C9", "Webserver":2, "WifiConfig":3}}

Working
09:21:27 CND: status 5
09:21:27 MQT: stat/stereo/STATUS5 = {"StatusNET":{"Hostname":"stereo", "IPAddress":"192.168.1.120", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "DNSServer":"8.8.8.8", "Mac":"5C:CF:7F:A0:2D:CF", "Webserver":2, "WifiConfig":3}}

TommySharpNZ, 2017-10-18:
Random question.... How do I get the timezone correct for Auckland, New Zealand?

I've tried "timezone 12" but I need to to actually be one more hour into the future.... i.e. The sonoff says it's 11:59pm but it's actually 01:59am here now. Is daylight savings messing with things?

ionciubotaru, 2017-10-18:
Set `Timezone 99` and set correct values for:
#define TIME_DST               Last, Sun, Mar, 2, +180  // Last sunday in march at 02:00 +180 minutes
#define TIME_STD               Last, Sun, Oct, 3, +120  // Last sunday in october at 03:00 +120 minutes

jat80, 2017-10-29:
@ionciubotaru how do I set those values ?  Do I do that in console?

ionciubotaru, 2017-10-29:
Download Tasmota latest version.
Change TIME_DST and TIME_STD in user_config.h, recompile and upload

TommySharpNZ, 2017-12-04:
Am I right in saying that if I use Sonota to upload Tasmota to a Sonoff POW then I have no way of setting the timezone to correctly handle New Zealand? How would I edit the user_config.h file?

TommySharpNZ, 2017-12-04:
Okay looks like timezone 13 is now possible.... Is it possible for someone to update the documentation?

http://192.168.1.31/cm?cmnd=timezone 13

jat80, 2017-12-04:
Yeah it was updated in a firmware update.  It was in the change log but may not be in the wiki.

closed 2018-05-07 by stale[bot]

ascillato, 2018-07-10:
Please, read the wiki. There is explained in commands. See timedst and timestd

royaljaat111, 2019-11-16:
For India: timezone +5:30
use this command