arendst/Tasmota #852: DHT22 failure keeps reporting the same values

mzealey, 2017-09-09:
I had a DHT22 fail - it keeps logging `Timeout waiting for start signal low pulse`. The issue is that for a day after it failed the software was just reporting the same temp/humid:

```
ESP-MQTT: domoticz/in = {"idx":16,"nvalue":0,"svalue":"26.3;50.2;1"}
```

After reboot it doesnt report anything. I think sensor values should time out after a while - perhaps 2* send ?

ionciubotaru, 2017-09-09:
Some suggestions:

- check the pullup resistor
- use short wires between esp and dht
- replace dht with a new one
- check messages on serial or mqtt instead of domoticz

mzealey, 2017-09-09:
Sure, however this issue is about the DHT22 but with the way the software keeps reporting out of date values when a device has been disconnected.

arendst, 2017-10-30:
Updated again in v5.9.0

closed 2017-10-30 by arendst