arendst/Tasmota #855: Home Assistant doesn't accept HEX codes for LED colors

jtroberts1, 2017-09-10:
Awesome work Theo.  I'm now using Tasmota on 5 different devices and thinking of now projects all the time.  Led and specifically WS2812 color modification is important to the success of Tasmota.  LED fun really gets people involved.  Given that Home Assistant is by far the fastest growing Home automation software I think Tasmota needs to add the option to choose sending Colors via HEX or numerical 255,255,255 like HA uses?  I'm thinking...

colormode 1 (1=HEX, 0=HA-compatible)

Thoughts?

automrk, 2017-09-11:
I have this problem too, I tried to use the fastled lib, which is even more complete and uses numerical instead of hex, but I was unsuccessful, maybe someone has accomplished this?

jtroberts1, 2017-09-27:
This has now been solved by HA.  add this to your config...  NOTE: requirev version 0.53.1

rgb_command_template: "{{ ‘%02x%02x%02x’ | format(blue, green, red)}}"

See my [thread](https://community.home-assistant.io/t/tasmota-rgb-convert-to-hex/26684) in the HA forums.

closed 2017-10-24 by jtroberts1