arendst/Tasmota #856: Can't get Tasmota on my Sonoff Dual. Seems to jump out of program mode DURING flashing.

EttVenter, 2017-09-10:
Every time I try to flash my Dual with Tasmota, I get a result like this:

`[SUMMARY]
Environment sonoff              [SUCCESS]
Environment sonoff-minimal      [ERROR]
Environment sonoff-ds18x20      [ERROR]`

Sometimes, it'll be the first two that succeed and the 3rd fails. I even had one instance where the 1st and 3rd succeed, but the second fails. 

If I go back through through the log, I see that the Sonoff seems to jump out of Program mode, because of this:

`Uploading 488000 bytes from .pioenvs\sonoff-ds18x20\firmware.bin to flash at 0x00000000
...................................warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: didn't receive command response
error: espcomm_upload_mem failed
*** [upload] Error -1`

I've connected GPIO0 and GND properly, so that's not the issue.

Any ideas?

EttVenter, 2017-09-11:
Anyone?

EmilySmith0, 2017-09-11:
you didn't mentioned compiler
I assume its arduino
First make BIN file
[image]

Then find that bin file in project folder
and upload using ESP download Tool
http://espressif.com/en/support/download/other-tools
[image]
[image]

I don't know about ATOM(PlatformIO).I think you can create BIN file with ATOM as well

EttVenter, 2017-09-11:
Thanks @EmilySmith0! I'm compiling with Atom. 

I'll give this a shot!

closed 2018-05-07 by stale[bot]