arendst/Tasmota #857: WS2811

Gary202, 2017-09-11:
Will WS2811 led strips work with the WS2812 selection?

chepa92, 2017-09-12:
I'm trying to get it work, no luck.

arendst, 2017-09-12:
It probably won't work as the timing is different between WS2811 and WS2812.

Perhaps you find more info on the NeoPixelBus library forum used in Tasmota and provides the WS2812 drivers.

joecotton, 2017-09-12:
Likely the only difference is how the NeoPixelBus object is instantiated. The `xdrv_ws2812.ino` file, lines 27-39. Try changing `Neo800KbpsMethod` with `Neo400KbpsMethod`

chepa92, 2017-09-12:
That what I did, not working

chepa92, 2017-09-12:
NeoEsp8266BitBang400KbpsMethod and Neo400KbpsMethod

chepa92, 2017-09-12:
Maybe problem in [low voltage? ](https://github.com/Makuna/NeoPixelBus/wiki/FAQ-%232)

chepa92, 2017-09-12:
I've added 4.7k on the gpio14 can this be a problem?

Adminius, 2017-09-13:
try to use GPIO3

chepa92, 2017-09-14:
not working, same reaction

Adminius, 2017-09-14:
are they are working without Tasmota? e.g. with NeoPixelBus test sketch only?

chepa92, 2017-09-14:
Didn't tested on sonoff, on arduino mini its working well

Adminius, 2017-09-14:
328p is not ESP8266 ;)
1. Try it on any 3.3V Arduino
2. Try it on any ESP8266
3. Try it with another Lib (e.g. Adafruit NeoPixel)
3. Different modes: DMA, UART, Bitbang...

skatun, 2017-11-16:
Did anyone get it to work?

chepa92, 2017-11-16:
nope

birca1987, 2018-01-28:
see here how I have them working on tasmota [https://github.com/arendst/Sonoff-Tasmota/issues/1690](url)

markcarline, 2018-06-05:
Anyone else having issues with WS2811? This is a circuit diagram and parts list of my setup. I'm wondering if I can use Tasmota in my setup rather than the code in his example:
https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs

ascillato, 2018-06-05:
Hi, please, could you test the solution on https://github.com/arendst/Sonoff-Tasmota/issues/1690   ?
Thanks

markcarline, 2018-06-05:
If I'm using one of the prebuilt binaries is there a way I can test this plus that fix looks like just a re order of the RGB values will that help I'm happy to test if possible

markcarline, 2018-06-05:
Sorry forgot to add is there a link I can use to get hold of one of the pre-built binaries that includes this fix thank you

markcarline, 2018-06-05:
Also is there any reason why fast LED isn't included in Tasmota? It would be a great addition and I've seen it used in a few other uses and seems to have good compatibility

arendst, 2018-06-05:
Life is all about choices. Shall I buy a car or would a bicycle do too? I choose for the bicycle as it gets me there too.

markcarline, 2018-06-09:
So yes, suggested the changes in:

https://github.com/arendst/Sonoff-Tasmota/issues/1690

Of editing user_config.h and no change.

The issue seems to be related to communication rather than a mix up of RGB colours.

Shall I open a separate bug report ? It does relate to WS2811 12v strip though, wired up like:

https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/blob/master/ESP%20MQTT%20Digital%20LEDs%20Wiring%20Diagram.png

skatun, 2018-08-06:
I got it to work without any modifications on a wemos mini, however colors are wrong, rbg, which seems to be standard for ws2811, either fix it in ur sketch or handle it in the smart home software. I did mention for @arendst that we maybe should add ws2811 as a sensor option where the colors are swapped compared to ws2812..

ascillato2, 2018-09-04:
Hi,

@skatun 

You said that colors are swapped but you did not specify which ones?

ascillato2, 2018-09-07:
Hi,

@skatun 

Can you share your configuration and your results?

ascillato2, 2018-09-09:
Hi,

Closing issue as there is no feedback. Please, when you have time, share your findings. Thanks.

closed 2018-09-09 by ascillato2

skatun, 2018-09-11:
@ascillato and @arendst , sorry I was away for the last 2 weeks. So I have tested with 4 different suppliers of ws2811 ledstrips and they all seems to be rbg coded in comparison to ws2812 that the user can select from the drop down menu, so my suggestion was to add ws2811 to the drop down menu. Can you point me out how to do this? The code that I use to fix this currently is:

Which then is defined in the user config as 
`  #define USE_WS2812_CTYPE     NEO_RBG // WS2812 Color type (NEO_RGB, NEO_GRB, NEO_BRG, NEO_RBG, NEO_RGBW, NEO_GRBW)`

But I would like to have it defined through the tasmota webpage, because if I autoupdate the software this manual setting will be lost. It is also easier for beginners, because then they dont have to fiddle around with the arduino code..

arendst, 2018-09-11:
Won't work as the NeoPixelBus library inits it's space and Color type during compile time.

skatun, 2018-09-11:
@arendst How did you do it for the ws2812 type that you can select then? Why would ws2811 be different from the same approach? Maybe I am missing something

arendst, 2018-09-11:
For the WS2812 you will have to re-compile if you need another RGB sequence.

skatun, 2018-09-11:
@arendst For the WS2812 you will have to re-compile if you need another RGB sequence. 
This is exactly what I ment for WS2812 the sequence is RGB, for WS2811 the sequence is RBG, so if we add this option to the drop down meny we have two hardcoded strips that will cover 90% of all strips on the market. For those using the rest 10% we will leave it as it is today.. Or add optional configuration as you do have for some of the other sensors

skatun, 2018-11-05:
I am not a programmer expert, but why is it not possible to copy the ws2812 sketch to ws2811 and change that one line to #define USE_WS2811_CTYPE NEO_RBG, can you please clearify why that is not possible?