arendst/Tasmota #860: Multiple DS18B20 in Domoticz MQTT - idx setup.

Bodzio122333, 2017-09-12:
Hello,
I connected two ds18b20 to Wamos D1 mini but I see one idx temperature in Configuration Domiticz menu. So module send one temprature to Domoticz. Please give me information how add two ds18b20 to domoticz.

[image]
[image]

arendst, 2017-09-13:
Currently not supported.
If you want it you'll have to edit all xsns files and xdrv_domoticz.ino

Bodzio122333, 2017-09-13:
I wanted to monitor the temperature of the stove and water heater with one Wemos D1 mini. I think that a lot of people have used a few ds18b20 thermometers in homes. Could you show how to make an additional temperature sensor from another sensor? Or add a new one. I think it's worth taking care of this option

Damme, 2018-01-15:
I think this is an important change to be made!

gj23, 2018-03-03:
has somebody done it? i have tried to get a second ds18x20 sensor under the domoticz... first i broke the whole tasmota and now i have it...but domoticz dont showing the temperature :( has somebody a tip to get it working?

[image]

Damme, 2018-03-04:
I've tried once and failed, haven't looked into it very much though

gj23, 2018-03-04:
ok, thx for the feedback. i will give it a new try the next days....and hope to get it workling

gj23, 2018-03-15:
hey guys,
it's almost done... but not 100% working. it's almost a little bit bugy. domoticz is showing me alternately the temperature from sensor1 on 2 idx and then the temperature from sensor 2 on the same 2 idx, not the both sensors seperately on each configured idx :(

sorry of the bad coding...i have done this simple way to check how i can intervent this and get a first feedback how it works. when its done i will change it in a clean code of using a for loop, arrays etc for example, to get this working for individual numbers and kinds of sensors automatically.

here are my changes:
______________________________________________________
de-DE.h

// added Lines
347 #define D_DOMOTICZ_TEMP2 "Temp2"
348  #define D_DOMOTICZ_TEMP3 "Temp3"
349  #define D_DOMOTICZ_TEMP4 "Temp4"
______________________________________________________________
______________________________________________________________
sonoff.h

//DZ_TEMP2 - DZ_TEMP4 added
140 enum DomoticzSensors {DZ_TEMP, DZ_TEMP2, DZ_TEMP_HUM, DZ_TEMP_HUM_BARO, DZ_POWER_ENERGY, DZ_ILLUMINANCE, DZ_COUNT, DZ_VOLTAGE, DZ_CURRENT, DZ_AIRQUALITY, DZ_MAX_SENSORS};
__________________________________________________________________
__________________________________________________________________
xdrv_05_domoticz.ino

//added D_DOMOTICZ_TEMP2 - 4
47 const char kDomoticzSensors[] PROGMEM =
48  D_DOMOTICZ_TEMP "|" D_DOMOTICZ_TEMP2 "|" D_DOMOTICZ_TEMP3 "|" D_DOMOTICZ_TEMP4 "|" D_DOMOTICZ_TEMP_HUM "|" D_DOMOTICZ_TEMP_HUM_BARO "|" D_DOMOTICZ_POWER_ENERGY "|" D_DOMOTICZ_ILLUMINANCE "|" D_DOMOTICZ_COUNT "|" D_DOMOTICZ_VOLTAGE "|" D_DOMOTICZ_CURRENT "|" D_DOMOTICZ_AIRQUALITY ;
  
//added Function DomoticzSensor2
  void DomoticzSensor2(byte idx, char *data)
{
  if (Settings.domoticz_sensor_idx[idx]) {
    char dmess[64];

    memcpy(dmess, mqtt_data, sizeof(dmess));
    if (DZ_AIRQUALITY == idx) {
      snprintf_P(mqtt_data, sizeof(dmess), PSTR("{\"idx\":%d,\"nvalue\":%s}"), Settings.domoticz_sensor_idx[idx], data);
    } else {
      snprintf_P(mqtt_data, sizeof(dmess), PSTR("{\"idx\":%d,\"nvalue\":0,\"svalue\":\"%s\"}"), Settings.domoticz_sensor_idx[idx], data);
    }
    MqttPublish(domoticz_in_topic);
    memcpy(mqtt_data, dmess, sizeof(dmess));
  }
}

void DomoticzSensor2(byte idx, uint32_t value)
{
  char data[16];
  snprintf_P(data, sizeof(data), PSTR("%d"), value);
  DomoticzSensor2(idx, data);
}
____________________________________________________________________
____________________________________________________________________

xsns_05_ds18x20_legacy.ino

//added i++ and Function DomoticzSensor2
190 #ifdef USE_DOMOTICZ
191        /*if (1 == dsxflg) {
192          DomoticzSensor(DZ_TEMP, temperature);
193        }*/
194        if (1 == dsxflg || dsxflg >= 1)
195        {
196          DomoticzSensor(DZ_TEMP, temperature);
197          i++;
198          DomoticzSensor2(DZ_TEMP2,temperature);
199        }  
200 #endif  // USE_DOMOTICZ

[image]
[image]
[image]

when somebody knows how to fix my problem, his knowledge is welcome to get it finished so that can it may be integrated in the official tasmota from arendst...that can be used by everyone.

gj23, 2018-03-16:
ok, its done!!!! :) i have 3 sensors now working in domoticz, all over gpio 14 and each sensor shows its temperature in domoticz seperatly on its given idx ;)

i have changed this:

xsns_05_ds18x20_legacy.ino
_______________________________________________

//changing dsxlfg++ to give the fix value of 1
186 dsxflg = 1;

//switch statement to seperate the temperature on right idx
191 #ifdef USE_DOMOTICZ
192       /*if (1 == dsxflg) {
193         DomoticzSensor(DZ_TEMP, temperature);
194        }*/
195
196       if (1 == dsxflg)
197        {
198         switch (i)
199          {
200           case 0:  DomoticzSensor(DZ_TEMP, temperature);
201                     break;
202
203            case 1:  DomoticzSensor(DZ_TEMP2, temperature);
204                     break;
205
206            case 2:  DomoticzSensor(DZ_TEMP3, temperature);
207                     break;
208          }
209        }
210 #endif  // USE_DOMOTICZ

so in xdrv_05_domoticz.ino you can delete DomoticzSensor2.

now i can change it to make it all dynamicly and automatic that the domoticz menu in tasmota is changing on depending how many sensors are really connected on sonoff. then i will change it for all sensortypes.

omegasa, 2018-05-13:
Im registered just for ask: is any chance to see this option (multiple idx for miltiple one-wire sensors) in next firmware update? It is difficult for me to update it manualy. Or may be there is some instruction?

gj23, 2018-05-15:
hi omegasa,
what sensors do you have? for the ds18x20 sensor i have descripted where i changed the code and what...you only need to paste the lines at the rows i have given. but when you use english you have to change the en-GB.h and put the lines in it where i have chosen the de-DE.h file. for the i2c sensors its different...but i can look it up and post it here if you want

omegasa, 2018-05-15:
I have 4 ds18x20 sensors on one wire. 
Just copy-paste strings in "xsns_05_ds18x20_legacy.ino", "xdrv_05_domoticz.ino", "sonoff.h" and en-GB.h ?

gj23, 2018-05-15:
ok, in xsns_05_ds18x20_legacy.ino put the cursor after break; at line 207 and hit enter. then paste or type this in:

case 3:     DomoticzSensor(DZ_TEMP4, temperature);
               break;

and in sonof.h you must add DZ_TEMP3 and DZ_TEMP4 at line 140 like so:

enum DomoticzSensors {DZ_TEMP, DZ_TEMP2, DZ_TEMP3, DZ_TEMP4, DZ_TEMP_HUM, DZ_TEMP_HUM_BARO, DZ_POWER_ENERGY, DZ_ILLUMINANCE, DZ_COUNT, DZ_VOLTAGE, DZ_CURRENT, DZ_AIRQUALITY, DZ_MAX_SENSORS};

and then it should work fine with your 4 sensors ;)

mikeo999, 2018-05-21:
I tried your great solution and it is almost working .  I connect two ds18x20 sensors to D5 and working  fine one the Wemos.

I do get both IDX field in Tasmota --> Domoticz settings and see both temp in the home screen of Tasmota. I filled both IDX numbers from Domoticz in the right IDX fields but i only get the first temp in Domoticz. The second one is not updated.

The Console show it well i think.
tele/sonoff229/SENSOR = {"Time":"2018-05-21T08:08:15","DS18B20-1":{"Id":"800000011960","Temperature":20.8},"DS18B20-2":{"Id":"800000011ADA","Temperature":19.9},"TempUnit":"C"}

So i think i am missing some code. I adjust to code for just 2 sensors.

You wrote: **so in xdrv_05_domoticz.ino you can delete DomoticzSensor2**
Can you explain which part of the code can be removed because it is not clear what you can skip now. 

Maybe an idea to make an update with all adjustments and remove the things that not needed any more.
And linenumbers are not correct anymore with the new source code so don't know if it is better to change it search for the code.

I hope you solution will be in the source quickly because this was the part we needed.

gj23, 2018-05-22:
hi mikeo999,

yes i have seen that with arendst latest updates, many on the sourcecode has changed.

in my first playing with the sourcecode i tried to get more sensors when i created the function DomoticzSensor2 to let the original sourcecode untouched. but it didnt worked, because i had to change so many things. so for me it was the easiest way to use the original function DomoticzSensor and change only the small thing in adding the switch statement without changing too much of the sourcecode.
and because of that switch statement i wrote later that the function DomoticzSensor2 can be deleted.

have you give a look to the Domoticz log? are the 2 sensor idx values listed in the log from domoticz? tasmota is showing all sensors automaticly...the changes i have made are only to get the new idx fields on the domoticz setting in tasmota that you can set the idx values so that tasmota send it to domoticz.

at the moment i'm really busy...but i will give it an update on the latest tasmota to make it easier for all ;)

mikeo999, 2018-05-22:
Hi gj23,

Would be great if you will do that. Problem is that DomoticzSensor2 is found a few times and the it is hard to know which parts you need and which parts can be skip.

So if you can make a small summery of the needed code is more the great.

In the log of Domoticz indeed it goes wrong so i think i missed some code.
MQTT: Topic: domoticz/in, Message: {"idx":3417,"nvalue":0,"svalue":"26.2","Battery":42,"RSSI":9}

mikeo999, 2018-05-23:
Great thanks for your help.

Tried the files but still same problem.

When i look at the console of Tasmota is see. The first line only send the first IDX 3417 but not send the other (3418). 

07:57:13 MQT: domoticz/in = {"idx":3417,"nvalue":0,"svalue":"22.3","Battery":41,"RSSI":9}
07:57:13 MQT: tele/sonoff229/SENSOR = {"Time":"2018-05-23T07:57:13","DS18B20-1":{"Id":"800000011960","Temperature":22.3},"DS18B20-2":{"Id":"800000011ADA","Temperature":23.1},"TempUnit":"C"}

I will remove everything and try it with a new project.

mikeo999, 2018-05-24:
Correct, i had ik included but because i start all over is was gone also. That was the moment i thought to mention it so other that are trying this will know that they need the adjust that part also.

mikeo999, 2018-06-03:
Still got a problem.

When i start the Wemos everything is working fine. Tasmota Gui shows both temp sensors and Domoticz gets updated. But after few moments the Tempsensors are gone in the Gui and Domoticz is also not updating any more until i remove the power. Then is all starts over again. A restart doens't fix it so may be the Wemos got problems so will test this soon with a other one. But maybe you know something else what fix the problem.

gj23, 2018-06-05:
hm, i have no wemos... but tested it on sonoff basic, th16, esp-12e and esp-32. on all devices its working without problems.

and on the basics and th16 it's working perfect for months, now

the only problem i have....is still loosing connection sometimes for a few minutes that my damn fritzbox kicks them out :D

closed 2018-08-04 by stale[bot]

omegasa, 2018-09-18:
I`m trying do this mod with latest tasmota (6.2.1.4), but no luck :-(

I'am edited all files exept **xsns_05_ds18x20.ino**

I dont understand how to edit **Ds18x20Show** function to send multiple IDX data to MQTT...

Can somebody help me?

omegasa, 2018-09-18:
Hmmm, probably working (for tasmota **6.2.1**)

In file **sonoff.h** replase string 197 by this:
`enum DomoticzSensors {DZ_TEMP, DZ_TEMP2, DZ_TEMP3, DZ_TEMP4, DZ_TEMP_HUM, DZ_TEMP_HUM_BARO, DZ_POWER_ENERGY, DZ_ILLUMINANCE, DZ_COUNT, DZ_VOLTAGE, DZ_CURRENT, DZ_AIRQUALITY, DZ_MAX_SENSORS};
`

In file **language\en-GB.h**
Insert after string 371
`  #define D_DOMOTICZ_TEMP2 "Temp2"
  #define D_DOMOTICZ_TEMP3 "Temp3"
  #define D_DOMOTICZ_TEMP4 "Temp4"`

If file **xdrv_07_domoticz.ino** replase string 50 by this:
`  D_DOMOTICZ_TEMP "|" D_DOMOTICZ_TEMP2 "|" D_DOMOTICZ_TEMP3 "|" D_DOMOTICZ_TEMP4 "|" D_DOMOTICZ_TEMP_HUM "|" D_DOMOTICZ_TEMP_HUM_BARO "|" D_DOMOTICZ_POWER_ENERGY "|" D_DOMOTICZ_ILLUMINANCE "|" D_DOMOTICZ_COUNT "|" D_DOMOTICZ_VOLTAGE "|" D_DOMOTICZ_CURRENT "|" D_DOMOTICZ_AIRQUALITY ;`

In file **xsns_05_ds18x20.ino** replase code block from ifdef USE_DOMOTICZ till  #endif  // USE_DOMOTICZ by this:

`#ifdef USE_DOMOTICZ
        if (0 == tele_period) {
  switch(i)
  {

    case 0:	DomoticzSensor(DZ_TEMP, temperature);
            break;

    case 1:	DomoticzSensor(DZ_TEMP2, temperature);
            break;

    case 2:	DomoticzSensor(DZ_TEMP3, temperature);
            break;

    case 3:	DomoticzSensor(DZ_TEMP4, temperature);
            break;
  }
        }
#endif  // USE_DOMOTICZ`

omegasa, 2018-09-18:
Hmmm, i am put my sonoff to better wifi area, and probleme gone, as i can see. No any reboots for 3+ hours.

cherowley, 2019-12-18:
Is this functionality ever to be implemented as standard ?

gj23, 2019-12-18:
> 
> 
> Is this functionality ever to be implemented as standard ?

i think no, there have been so many tasmota updates since i have it done...it's always on you to implement it in the new version.

i'm working almost with tasmota 5.14.0a... cause i act like never change a running system if it works well. there is no functionality in the further updates that i needed to change everything from new every time.

joba-1, 2019-12-18:
have you tried my tasmoticz plugin?

It discovers tasmota devices and creates domoticz devices for them.

No domoticz support on tasmota side required at all, but can coexist.

I don’t have tested this kind of double or multiple sensor setup, but would be interested to fix issues, if any.

omegasa, 2019-12-18:
I ve solved this via tasmota ruleses.

tasmota -> console -> 
rule1 on tele-DS18B20-1#temperature do publish domoticz/in {"idx":69,"svalue":"%value%;1"} endon on tele-DS18B20-2#temperature do publish domoticz/in {"idx":107,"svalue":"%value%;1"} endon
rule1 on

where 69 - Domoticz dummy sensor1, and 107 - sensor2, if u need, u can add the same way sensor3, sensor4, etc...

gj23, 2019-12-19:
@omegasa thats a good idea, i have read about rules longer time ago but didn't try it out. great if its working...thanks for the tip ;)