arendst/Tasmota #861: SwitchTopic

ionciubotaru, 2017-09-13:
Tasmota version 5.6.1k:

`cmnd/sonoff/SwitchTopic` returns `{"SwitchTopic":"0"}`
`cmnd/sonoffs/SwitchTopic` returns `{"Command":"Unknown"}`

switchMode returns `{"Command":"Unknown"}` in both cases

I suggest that switchmode and switchtopic returns 0 if no switch defined

arendst, 2017-09-13:
SwitchTopic issue is as designed; I only allowed SwitchTopic commands in dedicated device mode and not in group mode.
I will change that in next release so it will respond with the same result and leave it to the user if he/she wants to change all switchTopics at once... 

SwitchMode returns Unknown as there are no switches defined.
I will change that too in next release but will return the current SwitchMode setting as 0 is just one of them and might confuse the user.

ionciubotaru, 2017-09-13:
I have a database with all sonoffs configuration and run a daily script to check the configuration. That's why I send commands on grpupTopic instead of topic

closed 2017-09-14 by ionciubotaru