arendst/Tasmota #864: Cayenne/mydevices.com MQTT

dragondaud, 2017-09-13:
I was planning to do some testing using Cayenne/mydevices.com with some Sonoff modules, and discovered that Tasmota's MQTT configuration fields are hardcoded to a size of 33, and always chops off 4 characters of USERNAME, PASSWORD and CLIENTID, which are all always 36 characters long as generated by their software.

I am using the current github source, verified I could flash my module from the .bin using the web interface with the current version, and that worked fine. I modified settings.h where the char arrays are defined, after doing a quick check that the code was using sizeof() and not another hard coded value. I found an instance where it was using sizeof(mqtt_user) in a copy, so I recompiled, and tried to upload. The upload fails:
Upload Failed
Upload buffer miscompare

I'm going to look thru the code more, but thought I should report the issue, since Cayenne is a popular and free IoT platform that this fine code won't currently work with.

ulab, 2017-09-13:
I had a similar problem today that I wanted to look into. The goal was to use an FQDN instead of an IP for the syslog host and I noticed that it got cut at 32 chars when entered in the web frontend.

dragondaud, 2017-09-13:
According to the MQTT spec, username and such should allow 65535 characters (16bit). However, the DNS system only allows 250 characters for a FQDN. Limiting either to only 32 is very limiting.

arendst, 2017-09-13:
LOL

I bet you want to use TLS too.

ulab, 2017-09-13:
Now that you suggest that... ;)

Of course I understand the limitations, but would it be an option to inform the user instead of just cutting it off? 

Or for the webservice a "maxlength" for the <input> tags for example? You seem to be using "length" which is not defined according to the HTML standard?

arendst, 2017-09-13:
Regarding string length; See the wiki.

dragondaud, 2017-09-14:
I'm aware the length is coded to the inadequate limit of 32 characters. I don't see anything else in the wiki regarding this issues. Unless you'd like to point me to something I missed, it sounds like you just aren't interested in interoperability.

davidelang, 2017-09-14:
it's not "not interestedin interoperability", it's "ram is very limited and 32 
characters works for most people, making it larger would break other configs due 
to lack of ram, so this is the current compromise"

I know there was something on the wiki about this limit, but I don't remember 
where.

David Lang

patbeirne, 2017-09-17:
The parameter string lengths are in the code and duplicated here:

https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview

davidelang, 2017-09-18:
37 character userids and passwords are a waste of time, this is not going to be 
the only software that doesn't like them being that long.

dragondaud, 2017-09-18:
Using fixed length buffers is a waste of time and space from my point of view, but to each their own. I am not affiliated with mydevices.com, and was simply trying to test their service, which is wholly incompatible with your software.

petrfaitl, 2017-11-17:
As I understand there is an imposed (soft) limit on length of ClientID of 23 chars, in the standards. Not sure if the same applies to username and password.

Whilst I'd love to be able to use Cayenne, I think that's where we should be putting some pressure.

closed 2018-05-07 by stale[bot]

eadin79, 2018-12-14:
find out a possibility ... on Tasmota 6.3.0 we have free_1D5 since version 5.12.0e  20 bytes.... can be reused to MQTT and also you can comment by "//" some sensors not used by you in # define to save more memory ....

[image]

andtsin, 2018-12-28:
> 
> 
> find out a possibility ... on Tasmota 6.3.0 we have free_1D5 since version 5.12.0e 20 bytes.... can be reused to MQTT and also you can comment by "//" some sensors not used by you in # define to save more memory ....
> on cayenne page is clear: standard lenght of Cayenne MQTT settings (?) are 36 + 40 + 36 = 112 so if the total size not equal to 112 reset
> [image]

So what you have to do to increase the length of the characters? I increased it manually for username to 36, password to 40, and client to 36, but nothing changed.. Please tell me what to do

eadin79, 2019-01-05:
unfortunately I discovered that Tasmota do not support specific MQTT Cayenne publishing methods.... do no chance ... I will upload a trial made from different sources which I test now for 7 days already....
I made it based on:
https://community.mydevices.com/t/sonoff-or-other-esp8266-device-with-cayenne-mqtt-ota-ap/6948
https://gist.github.com/erikpena/a01ace32f7bef4a5ec52
https://github.com/regi18/Sonoff-Blynk
AND .... 
https://community.mydevices.com/t/sonoff-or-other-esp8266-device-with-cayenne-mqtt-ota-ap/6948

then I addapt Cayenne ....

THIS is example for TH10+AM2301+CAYENNE (it is a  tryout for IteadSI7021 but not successfully)
NOT FINALIZED , but working on that ...
/*
   Heavily inspired by: https://github.com/idreamsi/SonoffBoilerplate
   and https://github.com/tzapu/WiFiManager
   Converted from Blynk to Cayenne by WJ4IoT
   Preparation:
   - follow instructions: https://github.com/mirko/SonOTA
   - create hex from verified sketch and upload though: https://github.com/arendst/Sonoff-Tasmota/wiki
   - continue inside Arduino IDE
   Remarks:
   - no need to connect first to eWelink
   - DO NOT INSTALL LATEST FIRMWARE eWelink (did once result divice could not use SonOTA)
   - Make sketch non distructive comment out things like reset(), restart() if you loop into this you are screwed!
   - preferred testing sketch on easy access ESP8266 device like Wemos D1 R2 
   - a restart normally goes into error and freeze so it is also a Cayenne button
   Objectives: 
   - Solderless programming
   - Over the Air updating sketch
   - No private credentials in sketch 
   - On the fly change of Wi-Fi and MQTT credentials and store in EEPROM (sometimes required a second time)
   - Synchronisation of Sonoff Relay, button and Cayenne Actuator (regardless initiator)
   - Including:
     - time (from NTP-server)
     - Dailight Saving Time / Summertime
     - Scheduling based on sunset and sunrise
     - Display Wifi signal
     - uptime in days, hours minutes (bit stupid)
     - version displayed (bit more stupid)
   HOLDING PHYSICAL BUTTON SONOFF:
   - shorter than  1.5 seconds -> toggle relay and sync
   - between 1.5 - 5.0 seconds -> going into AP-modus
   - longer                    -> restart SonOff 
   SONOFF details:
   - Upload as Generic ESP8266 - Flash mode: DOUT - size: 1MB (64 SPIFFS)
   - Sonoff header (if you are not working solderless anymore)
     1 - vcc 3v3
     2 - rx
     3 - tx
     4 - gnd
     5 - gpio 14
*/
// standard available in SonOff Basic:
#define SONOFF_BUTTON    0 // gpio  0 - button
#define SONOFF_RELAY    12 // gpio 12 - relay
#define SONOFF_LED      13 // gpio 13 - green led - active low
#define SONOFF_INPUT    14 // gpio 14 - pin 5 on header

//Cayenne Virtual channels definitions
#define UpTime_Channel  37
#define UpDays_Channel  36
#define TimeNow_Channel 40
#define check_DST_Channel 43
#define TimeSunSet_Channel 38
#define TimeSunRise_Channel 39
#define SketchVersion_Channel 31
#define Schedule_Channel 26
#define Relay_Channel 0
#define ButtonStatus_Channel 51
#define ButtonPressTime_Channel 50
[... 384 lines cut ...]
}

CAYENNE_IN(0)  { toggle()  ; }                                     // toggle relay and sync leds
CAYENNE_IN(1)  { restart() ; }                                     // do not use!!
CAYENNE_IN(2)  { reset()   ; }                                     // into AP modes
//CAYENNE_IN(26) { digitalWrite(int_scheduling, getValue.asInt()); } // set internal scheduling 

void transmitData()
{
   float h = dht1.readHumidity();
   float t = dht1.readTemperature();
   Serial.print("Temperature: ");
   Serial.print(t);
   Serial.println("C");
   Serial.print("Humidity: ");
   Serial.print(h);
   Serial.println("%");
   Cayenne.virtualWrite(V4, h);
   Cayenne.virtualWrite(V5, t);

   if (digitalRead(SONOFF_BUTTON) == LOW) {    // button on SonOff is pressed
    PRESS_SONOFF_BUTTON();
  }

}