arendst/Tasmota #866: Noob question about Switchmode

sfanicola, 2017-09-14:
Hi guys,
don't get bored about my maybe noob question, but I really started from 0 and learned a lot thanks to the guide and the wiki.
But there's one thing I can't understand...
I've got a Sonoff Basic and would like to configure it from simple turn ON/OFF in Switchmode 6 so in configuration module I selected GPIO 01 09 Switch1 and sent the command via mqtt, receiving {"SwitchMode1":6} as answer. 
Here comes the noob, how can I use this switch mode? Is a command enough? 
Should I connect something to the GPIO 01? 
I would to open my house gate with it and don't want to risk to burn something down.
I hope something I've done is right :)
Thanks in advance

ionciubotaru, 2017-09-14:
Easy way - connect your switch to gpio0

Hard way:
- set `switchtopic 0`
- set `switchmode 0`
- connect switch1 to desired gpio (gpio1 and gpio3 are TX/RX pins - if you defined one of the as switch1 your serial interface will loose TX or RX capabilities during program operation)

sfanicola, 2017-09-14:
So you mean I should, after those settings, connect the wires of the switch directly on the gpio and not on the normal output?
What about voltage? Input is 220V and output on the gpio?

ionciubotaru, 2017-09-15:
It is a misunderstanding, connect 220V to relay output, not to GPIO
The switch has no electric potential, it is just a switch.

sfanicola, 2017-09-15:
Ok that was a stupid question about voltage.
I think I get it. Last question, turning almost on my first post ... Once I set the GPIO1 as a switch, can I control this switch through command without attaching a physical switch?

ionciubotaru, 2017-09-15:
Switch1 is input device. I think you want to control RELAY1 - this is possible `POWER1 ON`

sfanicola, 2017-09-15:
yep, but with the sonoff basic input 220V output will be 220V and will blow my gate circuits :)
I have to change relay and take a "1 Channel Inching /Self-lockIing Mode WiFi Wireless Switch 5V/12V" one. 
Thank you for your kind answer, even if my questions wasn't exactly on focus.

ionciubotaru, 2017-09-15:
https://www.banggood.com/DC-5V-24V-DIY-WIFI-Wireless-Switch-Sonoff-SV-Module-For-Smart-Home-p-1063987.html?rmmds=search - this is better

or this 

http://www.electrodragon.com/product/wifi-iot-relay-board-spdt-based-esp8266/
SPDT 5 pins relay, Songle 10A/250VAC/30VDC, NOT connected to the AC mains. Three states: normally ON, normally OFF and Common pin.

sfanicola, 2017-09-15:
so this one too or not?
http://www.electrodragon.com/product/d1-mini-wifi-dev-board-esp8266-esp12f/

sfanicola, 2017-09-15:
Is the last one compatible with Sonoff-Tasmota?
On the compatibility list I found this:
http://www.electrodragon.com/product/wifi-iot-relay-board-based-esp8266/

ionciubotaru, 2017-09-15:
all electrodragon models are compatibile with tasmota - I use more than 10 different electrodragons in my house

sfanicola, 2017-09-15:
Wow! Would love to see that :)
Thank you for your kind answers and patience ;)

sfanicola, 2017-09-27:
sorry that I continue bothering you... but the relay Electrodragon SPDT arrived this morning.
Don't want to burn it! Is there something I need to know how to flash it?

ionciubotaru, 2017-09-27:
Just be careful to use a 3V3 FTDI and supply power to ELD via FTDI
do not connect it to 220V during programming or when ftdi is nonnected

sfanicola, 2017-09-27:
Ok ... so I have to power it just through FTDI while the flash.
But after, can I connect the input to 220V or do I need a input power module?

ionciubotaru, 2017-09-27:
After flash look at the messages on the serial interface and configure Tasmota
In the end remove FTDI and connect it to 220V

sfanicola, 2017-09-29:
Great! Everything works now! Thank you very much ionciubotaru ;)
Just a final question. I connected the Tasmota ECT instead of the phisical button I had. 
But I don't want to loose this button...
So I turn back to the first confusing question made on top...
Is there a way to connect the old button to the relay making it a sort of wired remote button?
I guess i have to connect this button to a GPIO, or maybe two? the button to work needs 2 cables...

ionciubotaru, 2017-09-29:
Connect the button between gpio0 and gnd

sfanicola, 2017-09-29:
ehm... where is gpio0?

sfanicola, 2017-09-30:
I understand it finally... sorry for the ignorant questions... I connected the Button on Gpio1.
thank you once again

sfanicola, 2017-09-30:
Can’t understand switchmode / SwitchTopic page... with the real switch connected the relay starts randomly by itself...
Can’t understand why, neither watching the console

davidelang, 2017-10-01:
look at the logs, they will tell you what is happening

what's probably happening is that stray electricity and RF is being picked up by 
the switch wires. You probably need to hook up a pull-up resistor and bypass 
capacitor.

On Sat, 30 Sep 2017, sfanicola wrote:

> Can’t understand switchmode / SwitchTopic page... with the real switch connected the relay starts randomly by itself...
> Can’t understand why, neither watching the console

sfanicola, 2017-10-02:
could bigger wires, maybe shielded ones, could solve the problem?

davidelang, 2017-10-02:
bigger wires won't help, shielded may make a difference (may make it less 
neccessary to put in a bypass capacitor)

but you really should just add a pull-up resistor and try a bypass capacitor, 
it's simple and 99% of the time your problems go away.
k

sfanicola, 2017-10-02:
Ok, I will try the resistor, but my electronic welding abilities are really bad :) 
Where exactly should I hook it up?

davidelang, 2017-10-02:
between the GPIO pin and 3.3v, use a resistor in the ~4-10Kohm range.

sfanicola, 2017-10-02:
huh... I thought to put it on the wire... now I'm confused.
why the 3.3V... there is nothing attached

davidelang, 2017-10-02:
you connect the resistor between the wire that is attached to the GPIO and 3.3v 
so that when the switch is not closed, the wire/GPIO will be pulled up to 3.3v 
by the resistor.

when you close the switch, you connect the wire/GPIO to ground, and while some 
current flows through  the resistor, it's not enough to counteract the direct 
path to ground.

closed 2018-05-07 by stale[bot]