arendst/Tasmota #868: 5 relays

ionciubotaru, 2017-09-14:
I have one Electrodragon board with 2 (onboard) + 3 (extra) relays, so I need one more output PIN,

I didn't find any reference in the wiki about how GPIO_LED1 works.
Is it related to relay1 ?
Can I  toggle it independently ?

davidelang, 2017-09-14:
several people have posted patches to handle more than 4 relays. the LED can be 
managed separately, but you have to set the LED mode (see the commands page of 
the wiki).

you are probably better off digging up and applying the patch to increase the 
number of relays.

ionciubotaru, 2017-09-15:
@davidelang cannot find ledmode in wiki / supported commands
```
07:54:46 CND: ledmode
07:54:46 RSL: Group 0, Index 1, Command LEDMODE, Data  ()
07:54:46 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
```
Please provide a small example howto toggle gpio_led1

davidelang, 2017-09-15:
look at all the LED commands on the command page. It may not be ledmode, but 
there are commands to set how the led works.
k

ionciubotaru, 2017-09-15:
That commands are for default led, not for led1 to led4
Please write the command to toggle led4, I am sorry I didn't found it on [supported commands section](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

ionciubotaru, 2017-09-20:
Solved with this trick:
I defined the GPIO as PWM1 and I use it like `PWM1 1023` for ON and `PWM1 0` for OFF
Unfortunately there is no `pulsetime` for PWM

closed 2017-09-20 by ionciubotaru