arendst/Tasmota #869: Sonof pow continuos power cicling

ivandal77, 2017-09-15:
Dear,
I suspect there is some problem with the firmware.
My Sonoff Pow works great integrated with HA via mqtt. But if for some cause the wifi signal goes down (power off of the AP or ESSD during the night for example...) consequently the Sonoff lost the connection with HA ,it start to cycle the relay every 1 minute approximately... power on....power off....power on.... power off.....
I think is not normal. Maybe there is some flag in the firmware code? I think is not an HA problem since when it happen the HA is no connected to the Sonoff.
Thanks,

Ivan

davidelang, 2017-09-15:
what shows up in the logs?
k

ivandal77, 2017-09-15:
If I don't I have WiFi I don't have syslog available, serial not available when ac connected... So is difficult having log....
Some suggestions?
Thanks

arendst, 2017-09-15:
If pow is configured with default settings it will currently toggle between two access points wbich it cannot connect to as you have no wifi. This is as designed.

Solution is connecting serial and configure wificonfig set to 4. This will still toggle between non accessable access points but wil no longer reboot.

What's the use of a pow without wifi?

ivandal77, 2017-09-15:
Thanks,
But if the WiFi goes down it start to reboot by design????
I use with WiFi and works well thanks to you. 
But when I power off the WiFi om the ap for some reason, it continuos toogle on and of the relay....
Is it normal??? No WiFi mean continuos reboot and toogle on off on off?
I don't think is normal.
Thanks 
Ivan

arendst, 2017-09-15:
Read the solution again.....

ivandal77, 2017-09-15:
Ok but really I don't understand why you will it start to reboot when it lost wifi connection.
There is a way to set wificonfig 4 in the code without having to use serial?
The setting is permanent or get lost at the first manual reboot?
Thanks again, you can close since is not a bug.

Ivan

davidelang, 2017-09-16:
without looking at the logs, we don't know that it's rebooting, just that the 
power is cycling.

you can see the logs in the web interface, beyond that you need to look at the 
serial port (with an opto-isolator to keep from blowing up your equipment), or 
setup a syslog server somewhere.

without logs from the system to tell us what it thinks is happening, we cannot 
guess what's happening

closed 2018-05-07 by stale[bot]