arendst/Tasmota #874: cmnd/power not responding to on/off, only to 1/0

boyd91, 2017-09-15:
My Tasmota flashed sonoff touch is not responding to commands with payload `OFF`, `ON`, `BLINK`, `BLINKOFF`. It only works when I use their numeric counterparts `0`, `1`, `3`, `4`. The only string command that works is `TOGGLE`.

The MQTT status messages sent out by Tasmota do have the string payloads as expected (`ON`, `OFF`, ...)

- I know I have to send commands to the cmnd/sonoff/power topic.
- `MQTT_STATUS_OFF` and `MQTT_STATUS_ON` are set to `OFF` and `ON`

arendst, 2017-09-16:
What do you use to send commands?

boyd91, 2017-09-16:
An mqtt client for Mac, home assistant and the console on the Tasmota web interface. They all show the same behavior.

arendst, 2017-09-16:
Strange. It works fine with me. Pls reflash using the latest version.

boyd91, 2017-09-16:
It works after reflashing. I used the sonoff environment this time instead of the nl env. So it could have been related to that, not sure though.

arendst, 2017-09-16:
Ah that's it!

I'll change on/off/toggle check from just plain translations to the ones configured by the user in an updated release.

Blink would still be knipper though.

Nice problem.

closed 2017-09-21 by arendst