arendst/Tasmota #875: Hue Bridge/ Harmony Remote

AUyttendaele, 2017-09-16:
Hi,
I succesfully connected a Wemos D! mini enabled as Hue Bridge to my Logitech Harmony remote.

However I don't see any MQTT topics when toggling the lights with my Harmony remote.
Any idea's?

thanks

arendst, 2017-09-16:
Did you enable MQTT?

Pls provide output of command ``status 0``.

AUyttendaele, 2017-09-16:
Hi, yes, I can toggle the wemos via mqtt. Communication from the wemos to my harmony works. If I switch the output via mqtt I see a status change on the screen of my harmony remote. But when I toggle the output on my remote nothing happens and nothing appears on my broker. I am not home right now but I'll try the status message later. Thanks

AUyttendaele, 2017-09-16:
Hello, I must be doing something wrong.
I sent the command
/cmnd/WemosHue/status 0
the return is command unknown 

But as I mentionned I do get respons on Power commands etcetera

arendst, 2017-09-16:
Did you send the 0 as the payload to the command status?

What happens when at the console you enter status 0

AUyttendaele, 2017-09-17:
This is the status 0 response in the console
The only difference is that I have switched topic and prefix as this is more logical to me, with various devices.

06:05:22 CMD: status 0
06:05:22 MQT: /WemosHue/stat/STATUS = {"Status":{"Module":18, "FriendlyName":"WemosHue", "Topic":"WemosHue", "ButtonTopic":"0", "Power":3, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
06:05:22 MQT: /WemosHue/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":1, "Sleep":0, "BootCount":34, "SaveCount":79, "SaveAddress":"F6000"}}
06:05:22 MQT: /WemosHue/stat/STATUS2 = {"StatusFWR":{"Version":"5.7.1f", "BuildDateTime":"2017-09-16T07:06:09", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
06:05:22 MQT: /WemosHue/stat/STATUS3 = {"StatusLOG":{"SerialLog":2, "WebLog":2, "SysLog":0, "LogHost":"domus1", "LogPort":514, "SSId1":"ASUS", "SSId2":"", "TelePeriod":300, "SetOption":"55C00009"}}
06:05:22 MQT: /WemosHue/stat/STATUS4 = {"StatusMEM":{"ProgramSize":477, "Free":524, "Heap":22, "ProgramFlashSize":1024, "FlashSize":4096, "FlashMode":3}}
06:05:22 MQT: /WemosHue/stat/STATUS5 = {"StatusNET":{"Hostname":"WemosHue-6185", "IPAddress":"192.168.1.178", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "DNSServer":"195.130.130.3", "Mac":"2C:3A:E8:0E:D8:29", "Webserver":2, "WifiConfig":3}}
06:05:22 MQT: /WemosHue/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.212", "MqttPort":1883, "MqttClientMask":"DVES_%06X", "MqttClient":"DVES_0ED829", "MqttUser":"", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
06:05:22 MQT: /WemosHue/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 17 05:05:22 2017", "Local":"Sun Sep 17 06:05:22 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
06:05:22 MQT: /WemosHue/stat/STATUS10 = {"StatusSNS":{"Time":"2017-09-17T06:05:22"}}
06:05:22 MQT: /WemosHue/stat/STATUS11 = {"StatusSTS":{"Time":"2017-09-17T06:05:22", "Uptime":1, "Vcc":2.787, "POWER1":"ON", "POWER2":"ON", "Wifi":{"AP":1, "SSId":"ASUS", "RSSI":96, "APMac":"60:A4:4C:DD:59:10"}}}

AUyttendaele, 2017-09-17:
Just in case I reverted back to %prefix%/%topic%/ but this makes no difference

arendst, 2017-09-17:
Looks fine.

Try to enable more logging with either commands seriallog 3 or weblog 3 and see if that shines some light over the situation.

AUyttendaele, 2017-09-17:
[harmony.txt](https://github.com/arendst/Sonoff-Tasmota/files/1309134/harmony.txt)

Hi, I added a log txt file. I added some comments in the txt for clarification.

weblog 3 did not give me much. after enabling weblog 4 I do see the button presses on the harmony remote in the console. So Tasmota is seeing the button presses. I do not know how to go on from here.

arendst, 2017-09-18:
Just tried on a wemos too with seriallog set to 4:
```
15:13:51 WIF: Checking connection...
15:13:51 WIF: Connected
15:13:54 HTP: Hue API (//lights/2/state)
15:13:54 HTP: Hue POST args ({"bri":1,"on":true})
15:13:54 MQT: stat/wemos5/POWER2 = {"POWER2":"ON"}
15:13:54 MQT: stat/wemos5/POWER2 = ON
15:13:55 CFG: Saved to flash at F7, Count 765, Bytes 1648
15:13:58 HTP: Hue API (//lights/2/state)
15:13:58 HTP: Hue POST args ({"on":false})
15:13:58 MQT: stat/wemos5/POWER2 = {"POWER2":"OFF"}
15:13:58 MQT: stat/wemos5/POWER2 = OFF
15:13:59 CFG: Saved to flash at F6, Count 766, Bytes 1648
15:14:02 HTP: Hue API (//lights/1/state)
15:14:02 HTP: Hue POST args ({"on":false})
15:14:02 MQT: stat/wemos5/POWER1 = {"POWER1":"OFF"}
15:14:02 MQT: stat/wemos5/POWER1 = OFF
15:14:03 CFG: Saved to flash at F5, Count 767, Bytes 1648
15:14:04 HTP: Hue API (/139f3d)
```
Seems to work fine.

As the wemos is busy receiving Hue UDP messages I suggest to close your web pages during these actions. Others disabled MQTT to have a better Hue experience.

AUyttendaele, 2017-09-18:
The thing is I don't have a hue. I only want the messages appearing on MQTT
so that I can control whatever I want from my harmony remote. I didn't see
anything appearing in MQTT. I'll try by disabling the web page this evening
when I get home.

Thanks.

AUyttendaele, 2017-09-24:
Tried again with a  fresh install on a wemos and on a sonoff with the 5.8 firmware.

No mqtt topics being sent on state change.
Upon examining your logs and my logs I see that you get.
Hue API (//lights/2/state)
Whereas I get 
Hue API (/ed829/lights/2/state)

Is it possible that this ed829 is being ignored and therefore not translated to MQTT?

arendst, 2017-09-24:
ed829 is the userid and may be omitted. As your device switches it means it has received the correct signal and should also send the mqtt message.

What is the exact output of your logging?

AUyttendaele, 2017-09-24:
That's the whole point. The wemos sees the state change but does not switch
the output. Therefore I get no MQTT.

My log: I start the log with a status 0 so that you can see it's online, I
end with a button press on my remote, as you can see, no MQTT gets sent:

10:26:27 CMD: status 0
10:26:27 RSL: Received Topic /status, Data Size 1, Data 0
10:26:27 RSL: Group 0, Index 1, Command STATUS, Data 0

10:26:27 MQT: stat/hue/STATUS = {"Status":{"Module":18,
"FriendlyName":"hue", "Topic":"hue", "ButtonTopic":"0", "Power":1,
"PowerOnState":3, "LedState":1, "SaveData":1,
 "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
10:26:27 MQT: stat/hue/STATUS1 = {"StatusPRM":{"Baudrate":115200,
"GroupTopic":"sonoffs", "OtaUrl":"
http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0,
 "BootCount":16, "SaveCount":43, "SaveAddress":"F9000"}}
10:26:27 MQT: stat/hue/STATUS2 = {"StatusFWR":{"Version":"5.8.0",
"BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0",
"SDK":"1.5.3(aec24ac9)"}}
10:26:27 MQT: stat/hue/STATUS3 = {"StatusLOG":{"SerialLog":4, "WebLog":0,
"SysLog":0, "LogHost":"domus1", "LogPort":514, "SSId1":"ASUS",
"SSId2":"indebuurt2", "TelePeriod":300, "SetOption":"55C00009"}}
10:26:27 MQT: stat/hue/STATUS4 = {"StatusMEM":{"ProgramSize":469,
"Free":532, "Heap":22, "ProgramFlashSize":1024, "FlashSize":4096,
"FlashMode":3}}
10:26:27 MQT: stat/hue/STATUS5 = {"StatusNET":{"Hostname":"hue-6185",
"IPAddress":"192.168.1.178", "Gateway":"192.168.1.1",
"Subnetmask":"255.255.255.0", "DNSServer":"1
95.130.130.3", "Mac":"2C:3A:E8:0E:D8:29", "Webserver":2, "WifiConfig":3}}
10:26:27 MQT: stat/hue/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.164",
"MqttPort":1883, "MqttClientMask":"DVES_%06X", "MqttClient":"DVES_0ED829",
"MqttUser":"AUyttendaele", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
10:26:27 MQT: stat/hue/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 24 09:26:27
2017", "Local":"Sun Sep 24 10:26:27 2017", "StartDST":"Sun Mar 26 02:00:00
2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
10:26:27 MQT: stat/hue/STATUS10 =
{"StatusSNS":{"Time":"2017-09-24T10:26:27"}}
10:26:27 MQT: stat/hue/STATUS11 =
{"StatusSTS":{"Time":"2017-09-24T10:26:27", "Uptime":0, "Vcc":2.780,
"POWER":"ON", "Wifi":{"AP":1, "SSId":"ASUS", "RSSI":84, "APMac":"
60:A4:4C:DD:59:10"}}}
10:26:32 HTP: Hue API (/ed829/lights)
10:26:35 UPP: Hue 3 response packets sent to 192.168.1.135:15674
10:26:36 UPP: Hue 3 response packets sent to 192.168.1.135:15674
10:26:39 HTP: Hue API (/ed829/lights/1/state)
10:26:39 HTP: Hue API (/ed829/lights/1/state)
10:26:40 HTP: Hue API (/ed829/lights/1/state)
10:26:45 WIF: Checking connection...
10:26:45 WIF: Connected
10:26:49 HTP: Hue API (/ed829/lights)
10:26:51 UPP: Hue 3 response packets sent to 192.168.1.170:1008
10:27:01 HTP: Hue API (/ed829/lights)

AUyttendaele, 2017-09-24:
> api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0,

One more thing which appears weird to me:

The Hue sends responses to 2 IP's
192.168.1.135 which is a raspberry pi which has no relation, the MQTT
service is not running on this PI
192.168.1.170 which is my access point.

I would expect to see the IP here of the wemos (192.168.178) or my MQTT
server (.164)

On Sun, Sep 24, 2017 at 11:31 AM, alex uyttendaele <aj.uyttendaele@gmail.com

arendst, 2017-09-25:
The reactions from 170 and 135 are the result of their uPnP requests. If you would monitor your network you would be surprised to see what information is transferred between all kind of devices...

As you do not see messages like
```
HTP: Hue POST args ({"on":false})
```
it seems your Harmony Remote fails to send parameter(s) like ON/OFF so Tasmota does not know what to do.

Try another Hue App and see if that works. On my phone I use the Philips black and White Hue App for testing.

schlams, 2017-10-07:
Hi,
I have exactly the same problem! 
Could you find a solution? 
thanks

closed 2018-05-07 by stale[bot]

ol3k, 2018-08-20:
@AUyttendaele @arendst @schlams 
I've the same situation with a S20.

State is published but no chance of switching the S20 through the harmony app itself.
Any suggestions how to fix this?

jupzup, 2018-11-07:
Hello, I've the same problem here with my Logitech Harmony Hub ( Companion Remote ). On log level 4 i can see only "HTP: Hue-API (/4e2ff4/lights/1/state)" when i press the toogle button on the hub app or the logitech remote. it seems that logitech dont send a cmmand like on / off but the logitech will switch correctly with a original hue bridge. 
The logitech app shoes the state correctly if i switch on / off on the tasmota web gui. Any ideas what is the problem here or its is possible to fix them ?