arendst/Tasmota #877: Errors on timings?

hsmptg, 2017-09-16:
Hi

I need to turn on a Sonoff POW and turn off it after a few seconds.
For that I tried 2 different approaches:
- PulseTime
- Backlog w/ delay
In both cases I sometimes get the correct time but not always! :(

For instance:
http://192.168.1.126/cm?cmnd=PulseTime 50
http://192.168.1.126/cm?cmnd=Power 1
I get a 5s delay but almost half of the times I get 13s.

http://192.168.1.126/cm?cmnd=PulseTime 100
http://192.168.1.126/cm?cmnd=Power 1
I get a 10s but almost half of the times I get 17s.

http://192.168.1.126/cm?cmnd=PulseTime 150
http://192.168.1.126/cm?cmnd=Power 1
I get a 80s delay or sometimes 90s instead of the expected 50s as described in the wiki..

http://192.168.1.126/cm?cmnd=Backlog power 1; delay 50; power 0
I get the expected 5s delay.

http://192.168.1.126/cm?cmnd=Backlog power 1; delay 100; power 0
I get the expected 10s delay.

http://192.168.1.126/cm?cmnd=Backlog power 1; delay 150; power 0
I get always an incorrect 23s delay.

What is going on?!?

Regards,
Hélio

PS - Congratulations for this great Sonoff-Tasmota project!!!

davidelang, 2017-09-16:
what shows up in the logs? I wouldn't be surprised to learn that you had 
communications problems in the middle of the time period for example.

hsmptg, 2017-09-16:
Hi
Looking into the logs (I confess that I didn't knew that I could debug using that), I found the problem:
Since the MQTT is enabled by default (I thought it was disabled), and the default Host (domus1) did not exist I was getting some additional delays caused by that unsuccessful connection attempts.
The delays seem to be OK now!
Thanks
Hélio

closed 2018-05-07 by stale[bot]