arendst/Tasmota #878: Slight relay glitch on wifi reconnect

jonesmabel, 2017-09-17:
If wifi connection is lost (router off for instance) the Sonoff relay does not change state, however once the wifi is back, a relay that was ON will, for a very brief period, switch OFF before returning to ON. If it is connected to a lamp then this is not really a problem but if it is connected to equipment, a PC for example, then this is not a friendly thing to do the the PC power supply.
Any ideas ? Many thanks.

davidelang, 2017-09-17:
see the wiki page on poweronsettings (or something like that) and look at the 
retain setting and what the mqtt server may send you

jonesmabel, 2017-09-18:
Thanks but it is not mqtt settings as I am not using mqtt and it is not poweron, the sonoff remains powered, the problem is when the sonoff wifi reconnects after loosing the wifi signal (possibly due to a router reboot or router power cycle).

arendst, 2017-09-18:
Check command wificonfig options.

stefanbode, 2017-09-23:
You can also solve this with a small capacitor delivering the 3.3V during the short down on a electrical way. Depending on your requirements it can also be helpful to invert the signal to have the ON state already during boot. This depends on what is more critical. A small ON or a small OFF. With the wificonfig option you can avoid the reboot, and just try to reconnect.

jonesmabel, 2017-09-23:
Thanks to you both. The wificonfig option 4 seems to be the most elegant solution, particularly as it is no longer necessary to open the sonoff box to upload the Sonoff-Tasmota software, the sonOTA system is a bit strange (at first) but works well.
Many thanks for your hard work.

closed 2017-09-24 by jonesmabel