arendst/Tasmota #879: Basic: Touch sensor on GPIO 14

wunderbaum, 2017-09-17:
Hi,
I connected a touch-sensor ([](https://www.aliexpress.com/item/NEW-TTP223-Touch-button-Module-Capacitor-type-Single-Channel-Self-Locking-Touch-switch-sensor/32700504848.html)) on GPIO14 of a sonoff basic. 
Unfortunately, I cannot configure it like the original push button. Press once, its on, press next its off.
I would like to have the same functionality on GPIO14, too.
Is it possible?

davidelang, 2017-09-17:
see the switchmode command

wunderbaum, 2017-09-18:
That seems to work only on the internal switch unfortunately not on a pushbutton on GPIO 14.

wunderbaum, 2017-09-18:
Ah, https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management

Seems that I need to use:
Switchmode<x> 0

And the x is the number of the gpio?

davidelang, 2017-09-18:
what do you have GPIO defined as?

davidelang, 2017-09-18:
On Mon, 18 Sep 2017, Wunder Baum wrote:

> Ah, https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management
>
> Seems that I need to use:
> Switchmode<x> 0
>
> And the x is the number of the gpio?

no, X is the number of the switch (switch1..switch4)

wunderbaum, 2017-09-18:
Thanks! I will check this when I am back to my lab again. Let you know tonight!

wunderbaum, 2017-09-18:
Now I tested it, switch1 (the one on gpio14) is configured with switchmode1 as 0 (toggle).
But, as long as I press it, is ON, when I release it, its off.
Even more strange (in my point of view) is, when I toggle with the internal pushbutton, the touch on gpio works in an inverted way.

Of course, the easiest solution would be to solder the touch directly to the same contacts as the pushbutton.

wunderbaum, 2017-09-18:
Puuuh, its just 1 and zero. Binary things can be so hard to understand.
https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

With _switchmode1 4_ it works!
Still curious, the internal pushbutton seems to be switchmode 0.

davidelang, 2017-09-18:
the internal pushbutton is a button, not a switch (yes, there is a difference)

closed 2017-09-20 by wunderbaum