arendst/Tasmota #880: Can't upload firmware to Sonoff Basic. Upload is successful

koevskinikola, 2017-09-17:
I've been trying to upload the Sonoff-Tasmota firmware on my Sonoff Basic with a FTDI programmer. The pins are soldered well, the upload shows a successful message through the PlatformIO IDE, but nothing happens when I try to turn it on.

While it had the original firmware, I tried to do an update, but failed in the middle of it. Since then I couldn't get it to work and have been trying to flash it.

Can this be fixed or it's permanently bricked?

ionciubotaru, 2017-09-18:
Try esptoool https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool

Curtisjk, 2017-09-18:
Check mode is set to DOUT - I had similar issues where Arduino Studio would reset the mode after a restart.

koevskinikola, 2017-09-19:
I tried esptool, used the commands as instructed in the wiki, but still the sonoff won't turn on. I ensured that the mode was set to DOUT in the last command (for write_flash).

closed 2018-05-07 by stale[bot]