arendst/Tasmota #881: Multi mqtt messages or alt

mydakota, 2017-09-18:
Hi guys, 
I've modified my wireless doorbell to use Tasmota and communicate with my Home Assistant 
And worked well as a binary sensor on and off.
My issue is every time I lose power or reset door bell it reverts the bell tune to default and needs resetting so I programmed one pin as switch 2
For my sensor on n off then one pin as a relay to
Change tunes on the door bell all this will works fine but I can only receive one mqtt message at a time so either I can activate the relay via the Tasmota interface for tune selection of my door bell but this also now sends the mqtt message and can no longer get a mqtt message for the input.

Ideal solution
2 x mqtt messages one for relay one for a sensor.

Alt solution no mqtt messages for relay on off but still able to turn on n off via Web interface and the input to trigger a mqtt message on off 

Any help please save me as I'm going insane trying to figure this out

davidelang, 2017-09-18:
seethe backlog command

mydakota, 2017-09-18:
Ok just upgraded my FW and have done some reading in regards to back log but I'm not quite sure how to configure a relay and sensor using backlog would you know of any examples I could try thanks in advance.

mydakota, 2017-09-19:
Has anyone got any help for getting two mqtt messages using the backlog feature thanks

ionciubotaru, 2017-09-20:
```
Backlog status 1; power 2; delay 20; power 2; status 4
Backlog ssid1 myssid; password1 mypassword
```

Via mqtt:
topic: `cmnd/sonoff/backlog`
message: `status 1; power 2; delay 20; power 2; status 4`

mydakota, 2017-09-22:
ok ive got this as below

```
binary_sensors.yaml

- platform: mqtt
  name: Door Bell Ding Dong
  state_topic: "stat/DoorBell/POWER"      # receives sat changes from pin 5 of D1 mini  
  payload_on: "ON"
  payload_off: "OFF"
  qos: 0
  retain: false

switchs.yaml

- platform: mqtt
  name: "Door Bell Tune Changing switch"
  state_topic: "stat/DoorBell/POWER"
  command_topic: "cmnd/DoorBell/POWER"
  qos: 0
  payload_on: "ON"
  payload_off: "OFF"
  optimistic: false
  retain: true
```
I am not sure where I put the backlog commands as nothing i have tried works

mydakota, 2017-09-23:
any ideas i am stuck pleases :)

ionciubotaru, 2017-09-23:
```
switchs.yaml

- platform: mqtt
  name: "Door Bell Tune Changing switch"
  state_topic: "stat/DoorBell/POWER"
  command_topic: "cmnd/DoorBell/backlog"
  qos: 0
  payload_on: "power1 1;power2 1"
  payload_off: "power1 0;power2 0"
  optimistic: false
  retain: false
```

closed 2018-05-07 by stale[bot]