arendst/Tasmota #885: Recurrent LWT Online messages

PeggyFree, 2017-09-18:
Hello,

I'm a bit lost with LWT messages. I get tons of LWT online and offline messages. Who is guilty? My MQTT broker is mosquito. The offline message comes from a renamed sonoff. 
According to each sonoff log everything is all right. 

Look at this pollution (in fact there are 7 sonoffs like this): 

01:36:38   MQTT.tele/Sonoff86LedChambre/LWT.Online u"Online"
01:36:38   MQTT.tele/sonoff2/LWT.Offline u"**Offline**"
01:36:58   MQTT.tele/Sonoff86LedChambre/LWT.Online u"Online"
01:36:58   MQTT.tele/sonoff2/LWT.Offline u"**Offline**"
01:37:08   MQTT.tele/Sonoff86LedChambre/LWT.Online u"Online"
01:37:08   MQTT.tele/sonoff2/LWT.Offline u"**Offline**"

davidelang, 2017-09-19:
check your logs, you may find that you have multiple devices with the same name, 
or a device that's disconnecting and reconnecting from the network.

PeggyFree, 2017-09-19:
Thank you for your help. The sonoffs above all show a great uptime according to my wifi router. 
In the sonoffs console there is nothing special. In mosquitto I'm unable to get any log, mosquitto is running as a service on my windows, It seems impossible to get a log.

ionciubotaru, 2017-09-20:
If sonoff switch Online and Offline check your wifi
Try a free mosquitto server like cloudmqtt.com, maybe it is a problem with mosquitto windows version.

PeggyFree, 2017-09-20:
All my sonoffs show this trace, I think it is not abnormal : 

22:17:02 Wifi: Checking connection...
22:17:02 Wifi: Connected

But only a part of them suffer from abusive LWT.
Another clue, all LWT messages are generated at the same time, so it is probably a problem with my mosquitto. I don't know how to debug it.

VasilePaduraru, 2018-01-26:
It seems that is something wrong with LWT, multiple connection attempts to io.adafruit.com generates a rc. 6 error from adafruit. Any solution to this ?

23:28:55 MQT: Attempting connection...
23:28:55 MQT: Connected
23:28:55 MQT: tele/sonoff/LWT = Online (retained)
23:28:55 MQT: cmnd/sonoff/POWER = 
23:28:56 MQT: Attempting connection...
23:28:56 MQT: Connected
23:28:56 MQT: tele/sonoff/LWT = Online (retained)
23:28:56 MQT: cmnd/sonoff/POWER = 
23:28:57 MQT: Attempting connection...
23:28:57 MQT: Connected
23:28:57 MQT: tele/sonoff/LWT = Online (retained)
23:28:57 MQT: cmnd/sonoff/POWER = 
23:28:58 MQT: Attempting connection...
23:28:59 MQT: Connected
23:28:59 MQT: tele/sonoff/LWT = Online (retained)
23:28:59 MQT: cmnd/sonoff/POWER = 
23:29:00 MQT: Attempting connection...
23:29:00 MQT: Connected
23:29:00 MQT: tele/sonoff/LWT = Online (retained)
23:29:00 MQT: cmnd/sonoff/POWER = 
23:29:01 MQT: Attempting connection...
23:29:01 MQT: Connected
23:29:01 MQT: tele/sonoff/LWT = Online (retained)
23:29:01 MQT: cmnd/sonoff/POWER = 
23:29:02 MQT: Attempting connection...
23:29:02 MQT: Connected
23:29:02 MQT: tele/sonoff/LWT = Online (retained)
23:29:02 MQT: cmnd/sonoff/POWER = 
23:29:03 MQT: Attempting connection...
23:29:04 MQT: Connected
23:29:04 MQT: tele/sonoff/LWT = Online (retained)
23:29:04 MQT: cmnd/sonoff/POWER = 
23:29:05 MQT: Attempting connection...
23:29:05 MQT: Connected
23:29:05 MQT: tele/sonoff/LWT = Online (retained)
23:29:05 MQT: cmnd/sonoff/POWER = 
23:29:06 MQT: Attempting connection...
23:29:06 MQT: Connected
23:29:06 MQT: tele/sonoff/LWT = Online (retained)
23:29:06 MQT: cmnd/sonoff/POWER = 
23:29:07 MQT: Attempting connection...
23:29:08 MQT: Connected
23:29:08 MQT: tele/sonoff/LWT = Online (retained)
23:29:08 MQT: cmnd/sonoff/POWER = 
23:29:09 MQT: Attempting connection...
23:29:09 MQT: Connected
23:29:09 MQT: tele/sonoff/LWT = Online (retained)
23:29:09 MQT: cmnd/sonoff/POWER = 
23:29:10 MQT: Attempting connection...
23:29:10 MQT: Connected
23:29:10 MQT: tele/sonoff/LWT = Online (retained)
23:29:10 MQT: cmnd/sonoff/POWER = 
23:29:11 MQT: Attempting connection...
23:29:11 MQT: Connected
23:29:11 MQT: tele/sonoff/LWT = Online (retained)
23:29:11 MQT: cmnd/sonoff/POWER = 
23:29:12 MQT: Attempting connection...
23:29:13 MQT: Connected
23:29:13 MQT: tele/sonoff/LWT = Online (retained)
23:29:13 MQT: cmnd/sonoff/POWER = 
23:29:14 MQT: Attempting connection...
23:29:14 MQT: Connected
23:29:14 MQT: tele/sonoff/LWT = Online (retained)
23:29:14 MQT: cmnd/sonoff/POWER = 
23:29:15 MQT: Attempting connection...
23:29:15 MQT: Connected
23:29:15 MQT: tele/sonoff/LWT = Online (retained)
23:29:15 MQT: cmnd/sonoff/POWER = 
23:29:16 MQT: Attempting connection...
23:29:16 MQT: Connected
23:29:16 MQT: tele/sonoff/LWT = Online (retained)
23:29:16 MQT: cmnd/sonoff/POWER = 
23:29:17 MQT: Attempting connection...
23:29:18 MQT: Connected
23:29:18 MQT: tele/sonoff/LWT = Online (retained)
23:29:18 MQT: cmnd/sonoff/POWER = 
23:29:19 MQT: Attempting connection...
23:29:19 MQT: Connected
23:29:19 MQT: tele/sonoff/LWT = Online (retained)
23:29:19 MQT: cmnd/sonoff/POWER = 
23:29:20 MQT: Attempting connection...
23:29:20 MQT: Connect failed to io.adafruit.com:1883, rc 6. Retry in 60 sec

PeggyFree, 2018-01-26:
@ionciubotaru In reply to your last answer : My sonoffs don't generate LWT Offline, but keep generating LWT Online every 15s. All are up to date. I could'nt test them with another mqtt broker until know.

ionciubotaru, 2018-01-27:
As you can see sonoff disconnects from your AP very often. Possible causes:
- poor wifi signal
- multiple APs using the same channel
- AP problem - try to make your smartphone a hotspot and connect sonoff to internet via smartphone
- sonoff hardware problem (I had some similar problems with some sonoffs standard)

PeggyFree, 2018-02-12:
Let's try to understand why it is behaving like this : what would make Mosquitto send LWT Online messages every 20s for almost all my sonoffs? I don't fully understand the LWT mechanism. All sonoffs have a good connection, AP is stable, network is stable.

BigFood2307, 2018-02-19:
I have pretty much the same problem. For me it is almost exactly 12s. I can also constant DHCP requests via my network binding:

`2018-02-19 21:36:26.032 [INFO ] [work.internal.dhcp.DHCPListenService] - DHCP request for unknown address: 192.168.2.115

2018-02-19 21:36:38.518 [INFO ] [work.internal.dhcp.DHCPListenService] - DHCP request for unknown address: 192.168.2.115

2018-02-19 21:36:50.697 [INFO ] [work.internal.dhcp.DHCPListenService] - DHCP request for unknown address: 192.168.2.115

2018-02-19 21:37:02.674 [INFO ] [work.internal.dhcp.DHCPListenService] - DHCP request for unknown address: 192.168.2.115`

Also I only have the problem with sonoff basics. My Touch and pows are fine. Have you by any chance found a solution for this?

BigFood2307, 2018-02-19:
Also this only happens as soon as I configure the mqtt part. When I reset the basic or use wrong information (Pwd etc.) for Mqtt the DHCP Requests stop

BigFood2307, 2018-02-21:
getting the newest software (development) helped for me
others seem to have had a similar problem: #1955 
Hope it works for you too!

Lukhaub, 2018-02-21:
After updating to 5.12.0 (default) a few hours ago I suddenly was observing the same behaviour on my 1CH (5V) and 4CH device. Other devices (basic, TH16, Pow) do not show that behaviour, although using the same firmware.
After reading BigFood2307's comment above, I had a look at the recent changes and found this: "Fix intermittent exception when dns lookup is used while sleep is enabled".
So I switched off Sleep for now (using command "Sleep 0" on the console) and it looks like this solved the problem. Connection has been stable on both devices for about 30 minutes now.

So, switching off Sleep might be a temporary solution when you don't want to switch to the development branch.

swampdogmash, 2018-03-17:
I'm seeing similar issues. Not sure if the Wifi IS actually disconnecting as even a rapid ping shows constant responses to the device. Tasmota Console log logs like the attached below.

When i changed the "mqtt client name" from mwemos4 to mwemos4a I stopped seeing the issue which I guess would suggest a client with the same name (I can't seem to see one publishing even with mosquitto log on full debug) or maybe a leftover retained message from an old device with the same name. I'st tried reverting the client name back to "mwemos4" and it started again, so I've changed the "TOPIC" to "WEMOS4a" to see if this relates to something leftover in a queue for WEMOS4 on my mqtt server. Same issue. Not sure i understand what is happening here.

This is version 5.12.0 on a Wemos D1 Mini. (Makerhawk D1 Mini).

Give me a shout if anything else would be more helpful

**_With Client mwemos4 (I once had a deivce named this previously) and Topic WEMOS4_**

23:39:43 MQT: Attempting connection...
23:39:43 MQT: Connected
23:39:43 MQT: homeautomation/wemos/tele/WEMOS4/LWT = Online (retained)
23:39:43 MQT: homeautomation/wemos/cmnd/WEMOS4/POWER = 
23:39:45 MQT: Attempting connection...
23:39:45 MQT: Connected
23:39:45 MQT: homeautomation/wemos/tele/WEMOS4/LWT = Online (retained)
23:39:45 MQT: homeautomation/wemos/cmnd/WEMOS4/POWER = 
23:39:46 MQT: Attempting connection...
23:39:46 MQT: Connected
23:39:46 MQT: homeautomation/wemos/tele/WEMOS4/LWT = Online (retained)
23:39:46 MQT: homeautomation/wemos/cmnd/WEMOS4/POWER = 
23:39:46 MQT: homeautomation/wemos/tele/WEMOS4/STATE = {"Time":"2018-03-17T23:39:46","Uptime":"0T00:04:15","Vcc":2.974,"Wifi":{"AP":1,"SSId":"andyVMedia2g","RSSI":90,"APMac":"0C:85:25:9E:5E:30"}}
23:39:46 MQT: homeautomation/wemos/tele/WEMOS4/SENSOR = {"Time":"2018-03-17T23:39:46","AM2301":{"Temperature":20.2,"Humidity":43.3},"TempUnit":"C"}
23:39:52 MQT: Attempting connection...
23:39:52 MQT: Connected
23:39:52 MQT: homeautomation/wemos/tele/WEMOS4/LWT = Online (retained)
23:39:52 MQT: homeautomation/wemos/cmnd/WEMOS4/POWER = 
23:39:53 MQT: Attempting connection...
23:39:53 MQT: Connected
23:39:53 MQT: homeautomation/wemos/tele/WEMOS4/LWT = Online (retained)
23:39:53 MQT: homeautomation/wemos/cmnd/WEMOS4/POWER = 
23:39:55 MQT: Attempting connection...
23:39:55 MQT: Connected
23:39:55 MQT: homeautomation/wemos/tele/WEMOS4/LWT = Online (retained)
23:39:55 MQT: homeautomation/wemos/cmnd/WEMOS4/POWER = 

_**With Client mwemos4 and TOPIC WEMOS4a**_

23:51:33 MQT: Attempting connection...
23:51:33 MQT: Connected
23:51:33 MQT: homeautomation/wemos/tele/WEMOS4a/LWT = Online (retained)
23:51:33 MQT: homeautomation/wemos/cmnd/WEMOS4a/POWER = 
23:51:33 MQT: Subscribe to homeautomation/wemos/cmnd/WEMOS4a/#
23:51:33 MQT: Subscribe to homeautomation/wemos/cmnd/grpsonoffs/#
23:51:33 MQT: Subscribe to cmnd/mwemos4/#
23:51:34 DHT: Received 01, B6, 00, C6, 7D =? 7D
23:51:34 MQT: Attempting connection...
23:51:34 MQT: Connected
23:51:34 MQT: homeautomation/wemos/tele/WEMOS4a/LWT = Online (retained)
23:51:34 MQT: homeautomation/wemos/cmnd/WEMOS4a/POWER = 
23:51:34 MQT: Subscribe to homeautomation/wemos/cmnd/WEMOS4a/#
23:51:34 MQT: Subscribe to homeautomation/wemos/cmnd/grpsonoffs/#
23:51:34 MQT: Subscribe to cmnd/mwemos4/#
23:51:35 MQT: Attempting connection...
23:51:35 MQT: Connected
23:51:35 MQT: homeautomation/wemos/tele/WEMOS4a/LWT = Online (retained)
23:51:35 MQT: homeautomation/wemos/cmnd/WEMOS4a/POWER = 
23:51:35 MQT: Subscribe to homeautomation/wemos/cmnd/WEMOS4a/#
23:51:35 MQT: Subscribe to homeautomation/wemos/cmnd/grpsonoffs/#
23:51:35 MQT: Subscribe to cmnd/mwemos4/#
23:51:36 MQT: Attempting connection...
23:51:36 MQT: Connected
23:51:36 MQT: homeautomation/wemos/tele/WEMOS4a/LWT = Online (retained)
23:51:36 MQT: homeautomation/wemos/cmnd/WEMOS4a/POWER = 
23:51:36 MQT: Subscribe to homeautomation/wemos/cmnd/WEMOS4a/#
23:51:36 MQT: Subscribe to homeautomation/wemos/cmnd/grpsonoffs/#
23:51:36 MQT: Subscribe to cmnd/mwemos4/#
23:51:37 DHT: Received 01, B6, 00, C6, 7D =? 7D
23:51:37 MQT: homeautomation/wemos/tele/WEMOS4a/STATE = {"Time":"2018-03-17T23:51:37","Uptime":"0T00:02:55","Vcc":2.977,"Wifi":{"AP":1,"SSId":"andyVMedia2g","RSSI":90,"APMac":"0C:85:25:9E:5E:30"}}
[... 28 lines cut ...]
23:51:41 MQT: Subscribe to homeautomation/wemos/cmnd/WEMOS4a/#
23:51:41 MQT: Subscribe to homeautomation/wemos/cmnd/grpsonoffs/#
23:51:41 MQT: Subscribe to cmnd/mwemos4/#
23:51:42 MQT: Attempting connection...
23:51:42 MQT: Connected
23:51:42 MQT: homeautomation/wemos/tele/WEMOS4a/LWT = Online (retained)
23:51:42 MQT: homeautomation/wemos/cmnd/WEMOS4a/POWER = 
23:51:42 MQT: Subscribe to homeautomation/wemos/cmnd/WEMOS4a/#
23:51:42 MQT: Subscribe to homeautomation/wemos/cmnd/grpsonoffs/#
23:51:42 MQT: Subscribe to cmnd/mwemos4/#

**_Client mwemos4a  and topic WEMOS4_**

23:54:05 HTP: Console
23:54:09 MQT: homeautomation/wemos/tele/WEMOS4/STATE = {"Time":"2018-03-17T23:54:09","Uptime":"0T00:00:15","Vcc":2.973,"Wifi":{"AP":1,"SSId":"andyVMedia2g","RSSI":86,"APMac":"0C:85:25:9E:5E:30"}}
23:54:09 DHT: Received 01, B7, 00, C5, 7D =? 7D
23:54:09 MQT: homeautomation/wemos/tele/WEMOS4/SENSOR = {"Time":"2018-03-17T23:54:09","AM2301":{"Temperature":19.7,"Humidity":43.9},"TempUnit":"C"}
23:54:19 WIF: Checking connection...
23:54:19 WIF: Connected
23:54:19 MQT: homeautomation/wemos/tele/WEMOS4/STATE = {"Time":"2018-03-17T23:54:19","Uptime":"0T00:00:25","Vcc":2.974,"Wifi":{"AP":1,"SSId":"andyVMedia2g","RSSI":86,"APMac":"0C:85:25:9E:5E:30"}}
[... cut ...]
23:54:19 MQT: homeautomation/wemos/tele/WEMOS4/SENSOR = {"Time":"2018-03-17T23:54:19","AM2301":{"Temperature":19.7,"Humidity":43.9},"TempUnit":"C"}
23:54:29 MQT: homeautomation/wemos/tele/WEMOS4/STATE = {"Time":"2018-03-17T23:54:29","Uptime":"0T00:00:35","Vcc":2.973,"Wifi":{"AP":1,"SSId":"andyVMedia2g","RSSI":88,"APMac":"0C:85:25:9E:5E:30"}}
23:54:29 DHT: Received 01, B7, 00, C5, 7D =? 7D
23:54:29 MQT: homeautomation/wemos/tele/WEMOS4/SENSOR = {"Time":"2018-03-17T23:54:29","AM2301":{"Temperature":19.7,"Humidity":43.9},"TempUnit":"C"}

chedxb, 2018-10-18:
I started having the same problem.

I store MQTT data onto an SQL database and SQL table is full of LWT messages, repeating every 1 (one second) and interestingly even for the devices which are not currently connected to internet (they used to send MQTT data in the past)

MQTT is working fine, I can see incoming data when I subscribe from my iPhone and Windows, however I only see recurring LWT messages on my SQL table, nothing else.

Any idea?

Best,

Savas

tomeraviram, 2019-06-20:
I had an Issue with recurring LWT connect messages.
I found out that it happened because I had 2 connections (subscribers) running at the same time with the same ClientID

Djang0Pepper, 2020-11-11:
same !! 
configuration => mqtt configuration => Client (DVES_xxxxxx)

gerdotron, 2023-03-01:
@TnDA-tech Thank you SOOOOO Much for posting your solution. It was exactly the problem I was having. It was right in front of my eyes and I would have never seen it. I've been troubleshooting this for days. FIXED!!!

eknoesbrunswick, 2023-11-28:
You saved me from a mental breakdown!

Tank you!