arendst/Tasmota #886: I got some issue with switch recently

hamwong, 2017-09-19:
HI, I recently found a strange issue,
I set 2 switch but it resufse to name it with POWER1 / POWER2 for each switch
they all name on POWER but it's not happening on all devices.

`19:20:28 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2017-09-19T19:20:28", "Switch1":"OFF", "Switch2":"OFF", "AM2301":{"Temperature":25.2, "Humidity":42.4}, "TempUnit":"C"}
19:20:49 MQT: cmnd/BedRoom-ESP-Motion/POWER = ON
19:20:53 MQT: cmnd/BedRoom-ESP-Motion/POWER = OFF
19:21:13 MQT: cmnd/BedRoom-ESP-Motion/POWER = ON
19:21:15 MQT: cmnd/BedRoom-ESP-Motion/POWER = OFF`

(above first time were switch1 and second time is switch2)
please advise is there any new command I was missing?
also something very weired, on switchmode, I set both to 2, and default should be off, and once sensor detected, change to on, as you see below, swithmode both set to 2 and result is SW1=OFF SW2=ON...
is this is bug?

```
19:26:29 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2017-09-19T19:26:28", "Switch1":"OFF", "Switch2":"OFF", "AM2301":{"Temperature":25.2, "Humidity":42.4}, "TempUnit":"C"}
19:26:38 CND: switchmode1
19:26:38 MQT: stat/ESP-Wemo1-BedRoom/RESULT = {"SwitchMode1":2}
19:26:46 CND: switchmode2
19:26:46 MQT: stat/ESP-Wemo1-BedRoom/RESULT = {"SwitchMode2":1}
19:27:12 CND: switchmode2 2
19:27:12 MQT: stat/ESP-Wemo1-BedRoom/RESULT = {"SwitchMode2":2}
19:27:29 MQT: tele/ESP-Wemo1-BedRoom/STATE = {"Time":"2017-09-19T19:27:29", "Uptime":8, "Vcc":3.032, "Wifi":{"AP":1, "SSId":"Ham2", "RSSI":40, "APMac":"64:09:80:0B:A8:DD"}}
19:27:29 MQT: tele/ESP-Wemo1-BedRoom/SENSOR = {"Time":"2017-09-19T19:27:29", "Switch1":"OFF", "Switch2":"ON", "AM2301":{"Temperature":25.2, "Humidity":42.4}, "TempUnit":"C"}
```

hamwong, 2017-09-20:
I fallback to firmware 5.0.6 and 4.11, both able have Switch1>POWER1 and Switch2>POWER2

arendst, 2017-09-20:
If there is just one relay it will perform as you state above.

Check if your device config has one or two relays. Some people seem to loose some devices after an upgrade. I don't know why but it may have something to do with own compilations and using the wrong linker file not being 1M (no SPIFFS).

closed 2018-05-07 by stale[bot]