arendst/Tasmota #887: Problem after upgrade to last firmware

aferraz, 2017-09-19:
i have upgraded a nodemcu, configured as Wenos, to the last firmware version.
Previously the mcu is on 5.6.0l release.
The mcu control 2 relay, after upgrade I can turn on the relay from mqtt and from web interface but the relays don’t turn really on.

mydakota, 2017-09-19:
Had similar issue with two of my sonoffs after upgrading found by going to the mqtt settings option and selecting where %prefix%/%topic%/
Is greyed out but not changing anything then save fixed both my issues . Worth a try

mydakota, 2017-09-19:
You can tell by going to information page before hand and full Mqtt topic is corrupted the save thing I said prior fixes it

ionciubotaru, 2017-09-20:
Try `reset 1` the reconfigure the device from the beginning.

aferraz, 2017-09-20:
This is not an mqtt problem. When I send mqtt command I can see relay switching on web interface but it don’t switch really. I’ll try the reset the config.

ionciubotaru, 2017-09-20:
Maybe your sonoff has a hardware malfunction.
Write you own small code (see blink example in Arduino library) upload it using ftdi and check if the relay works.

aferraz, 2017-09-20:
With version 5.6 it works

arendst, 2017-09-20:
Provide output of command ``status 0``.

mydakota, 2017-09-20:
Yip I'm now finding my issues are across all my sonoffs upgraded to 5.7.1 
Switches will turn on in Home Assistant but no status updates so switch just turns of in HA And can't turn switch off vi HA so light jams on. I've reset one then tried to reconfigure but has now become unresponsive I'll have to disconnect and reflash it. But still issues with 5 other units may need to roll back the FW version to a older one until issues are fixed grrrr

aferraz, 2017-09-20:
This is status 0 result after firmware update:
18:46:09 MQT: stat/esp05/STATUS = {"Status":{"Module":18, "FriendlyName":"Terrario", "Topic":"esp05", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":1}}
18:46:09 MQT: stat/esp05/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"esp", "OtaUrl":"http://esp.iltuoweb.net/api/arduino/nodemcu.ino.bin", "Uptime":0, "Sleep":250, "BootCount":35, "SaveCount":804, "SaveAddress":"3F5000"}}
18:46:09 MQT: stat/esp05/STATUS2 = {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T22:47:27", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
18:46:09 MQT: stat/esp05/STATUS3 = {"StatusLOG":{"SerialLog":0, "WebLog":2, "SysLog":2, "LogHost":"192.168.1.3", "LogPort":514, "SSId1":"XXXXXX", "SSId2":"XXXXXX", "TelePeriod":60, "SetOption":"55800229"}}
18:46:09 MQT: stat/esp05/STATUS4 = {"StatusMEM":{"ProgramSize":455, "Free":568, "Heap":23, "ProgramFlashSize":4096, "FlashSize":4096, "FlashMode":3}}
18:46:09 MQT: stat/esp05/STATUS5 = {"StatusNET":{"Hostname":"esp05-0501", "IPAddress":"192.168.1.44", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.3", "Mac":"18:FE:34:D4:61:F5", "Webserver":2, "WifiConfig":2}}
18:46:09 MQT: stat/esp05/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.iltuoweb.net", "MqttPort":1883, "MqttClientMask":"DVES_%06X", "MqttClient":"DVES_D461F5", "MqttUser":"XXXXXX", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
18:46:09 MQT: stat/esp05/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 20 16:46:09 2017", "Local":"Wed Sep 20 18:46:09 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":99}}
18:46:09 MQT: stat/esp05/STATUS10 = {"StatusSNS":{"Time":"2017-09-20T18:46:09", "DS18B20":{"Temperature":26.1}, "BME280":{"Temperature":24.7, "Humidity":60.3, "Pressure":973.7}, "BH1750":{"Illuminance":56}, "TempUnit":"C"}}
18:46:09 MQT: stat/esp05/STATUS11 = {"StatusSTS":{"Time":"2017-09-20T18:46:09", "Uptime":0, "Vcc":2.777, "POWER1":"OFF", "POWER2":"OFF", "POWER3":"OFF", "Wifi":{"AP":1, "SSId":"XXXXXX", "RSSI":30, "APMac":"EC:08:6B:36:0D:B6"}}}

arendst, 2017-09-20:
I'm triggered by NodeMCU and the fact that ProgramFlashSize is 4096.

What I see going wrong is compiling the source with a recommended linker of 1M and no spiffs which is being serial uploaded to a device using NodeMCU upload tools which try to change ProgramFlashSize to 4M.

That's probably what has happened here and will not work when an OTA upgrade is performed.

Get your compile and upload options in sync. Make sure it is always 1M as there is no need for 4M with Tasmota.

aferraz, 2017-09-20:
Hi @arendst I always compile the firmware mysqlf with 4M(3M SPIFFS) for nodemcu and 1M(NO SPIFF) for sonoff/electodragon, also the actual loaded version (5.6.1l) is compiled with the same parameters and work fine.

aferraz, 2017-10-01:
@arendst I try to recompile with  1M and no spiffs but I have the same result.
Relay 1 and 2 don't work, WS2812 work fine. I can't understand where the problem is.

davidelang, 2017-10-01:
check your module configuration and see that the relays are defined tothe 
correct GPIOs

aferraz, 2017-10-30:
@davidelang the configuration is correct. In the meantime I change NodeMcu with an Electrodragon and all work with the last firmware

closed 2017-10-30 by aferraz