arendst/Tasmota #888: Connect directly as a AP??

Neo-NL, 2017-09-19:
Hello,

When i try to connect direct to the Sonoff with Tasmota on it. i get stuck at the Wifi network scan.
Thing is, i don't want to set it up in a network but want to use it as a stand alone.

Is that possible? and if yes, how can i arrange that?

ionciubotaru, 2017-09-20:
Try `WifiConfig 2` https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wifi

Neo-NL, 2017-09-20:
Thank you for you comment.
When i Select WifiConfig 2 it starts looking again for a Network like indebuurt1 and 2.
When i go to the 192.168.4.1 screen, it shows me the Networks again.

Do i have to make some changes in the ino files perhaps?

arendst, 2017-09-20:
It's not meant to stay in AP mode. You'll have to tweak the source yourself and allow access to more configuration pages too.

closed 2018-05-07 by stale[bot]