arendst/Tasmota #889: S20 now with esp8265

MatsKarlsson, 2017-09-19:
Just info to perhaps add to the wiki.

Got a couple of S20's that didn't work after I flashed them, examined the esp-chipset and discovered it was a esp8265. So when I flashed with "Generic esp8285" board they worked perfectly.

the back says that it's `Model: PS-16`, `Max Load: 10A`
PCB says `DLX-S20_EU_V15` (or perhaps V1.5, difficult to see)

PeggyFree, 2017-09-20:
I just received 13 S20 from banggood. I Will check that.

arendst, 2017-09-20:
If so please do not use Generic esp8285 but follow the readme and wiki and my tips at https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips concerning DOUT.

I try to keep telling but people tend to stumble across this for months.....

vstonkus, 2018-02-19:
@arendst Sorry, but I dont find how to setup for flashing S20 with esp8285. Do I need to change board in platformio.ini? I use Visual Studio Code & Platformio. Thanks in advance.

arendst, 2018-02-19:
esp8285 has been taken care off. Just use the provided platformio.ini file and it should fly.