arendst/Tasmota #890: Power always on after restart: LWT Online -> Power (null) -> Result: POWER ON

tmikko, 2017-09-20:
Hi,

I know that there's almost a similar issue reported before, but this might be something new still, hence the new thread.

As the topic says, after a restart (power loss) my Sonoff basic always turns on. This is what I see in mosquitto log
09/20/17 10:21:56.874 MyHome/Bathroom/stat/sonoff_basic_1/RESULT {"POWER":"OFF"}
09/20/17 10:21:56.917 MyHome/Bathroom/stat/sonoff_basic_1/POWER OFF
09/20/17 10:22:10.341 MyHome/Bathroom/tele/sonoff_basic_1/LWT Online
09/20/17 10:22:10.398 MyHome/Bathroom/cmnd/sonoff_basic_1/POWER (null)
09/20/17 10:22:10.467 MyHome/Bathroom/stat/sonoff_basic_1/RESULT {"POWER":"ON"}
09/20/17 10:22:10.507 MyHome/Bathroom/stat/sonoff_basic_1/POWER ON
09/20/17 10:22:10.551 MyHome/Bathroom/tele/sonoff_basic_1/INFO1 {"Module":"Sonoff Basic", "Version":"5.7.1", "FallbackTopic":"DVES_F499D0", "GroupTopic":"sonoffs"}
09/20/17 10:22:10.596 MyHome/Bathroom/tele/sonoff_basic_1/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff_basic_1-6608", "IPAddress":"192.168.1.41"}
09/20/17 10:22:10.640 MyHome/Bathroom/tele/sonoff_basic_1/INFO3 {"RestartReason":"Power on"}

What happened there is that at 10:21:56 I pressed the physical button on the Sonoff to switch the relay OFF. Then I restarted the module. 
At 10:22:10 I'm getting "LWT Online" message 
and after that - here's where I suspect we have an issue - I'm getting "POWER (null)" message that is then interpreted as "POWER ON" switching the light on again.

The relevant user config parameters are;
#define SAVE_DATA              1                
#define SAVE_STATE             1  
#define MQTT_BUTTON_RETAIN     0
#define MQTT_POWER_RETAIN      0
#define MQTT_SWITCH_RETAIN     0

I'm running Tasmota v5.7.1 (20170909) and mosquito server on Raspberry PI with MQTT v.3.1 (not 3.1.1)

Any idea how to fix this? A stupid user error maybe?

BR,
Mikko

arendst, 2017-09-20:
Your assumption about the "POWER (null) message are incorrect. That is just a message to satisfy people using iobroker.

The problem is just before 10:22:10.467 where your sonoff most probably receives a retained message from your MQTT broker.

Enable more logging with command seriallog/weblog/syslog option 3 and check if a MQTT message is received.

Check for any retained messages on your MQTT broker and remove the ones addressing this sonoff.

tmikko, 2017-09-20:
Thank for a prompt reply!

Now studying this a bit further it really seems the sonoff is receiving a ON command after reboot
00:00:00 CFG: Loaded from flash at F4, Count 8
00:00:00 APP: Boot Count 3
00:00:00 Project sonoff_basic_1 Sonoff (Topic sonoff_basic_1, Fallback DVES_F499D0, GroupTopic sonoffs) Version 5.7.1
00:00:00 CFG: Saved to flash at FB, Count 9, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 XYZ in mode 11N as sonoff_basic_1-6608...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff_basic_1-6608.local with IP address 192.168.1.41
00:00:03 APP: (UTC) Wed Sep 20 08:32:22 2017
00:00:03 APP: (DST) Sun Mar 26 02:00:00 2017
00:00:03 APP: (STD) Sun Oct 29 03:00:00 2017
09:32:24 MQT: Attempting connection...
09:32:25 DNS: Query done. MQTT services found 0
09:32:25 MQT: Connected
09:32:25 MQT: MyHome/Bathroom/tele/sonoff_basic_1/LWT = Online (retained)
09:32:25 MQT: MyHome/Bathroom/cmnd/sonoff_basic_1/POWER = 
09:32:25 RSL: Group 0, Index 1, Command POWER, Data ON (ON)
09:32:25 MQT: MyHome/Bathroom/stat/sonoff_basic_1/RESULT = {"POWER":"ON"}
09:32:25 MQT: MyHome/Bathroom/stat/sonoff_basic_1/POWER = ON

The RSL line is an incoming command, right?

I also tried to remove the possible retained messages from mosquitto with a command
`mosquitto_pub -t MyHome/Bathroom/cmnd/sonoff_basic_1/POWER -r -n`

but with no success. The same thing happens again.

At least two things remain a mystery;
1. Why in the first place there was a retained command on the broker. As you see from my original post, all the RETAIN config flags are set to 0.
2. Why on earth they still keep coming after executing the command above

Also, the POWER (null) message seem to be there for removing any retained messages, or?

BR,
Mikko

arendst, 2017-09-20:
As said above, the POWER (null) message is just there to signal iobroker users what command to use to control their sonoff. It does not remove retained messages from any MQTT server.

The mosquitto_pub command line should have had removed the retained message and probably did but you still might have a retained message for POWER1 (which addresses the same relay).

tmikko, 2017-09-20:
Thanks again for your help.

I did the following. 
1. Switched off the sonoff relay (turned lamp off)
2. Unplugged the sonoff from the wall
3. run the following commands
```
mosquitto_pub -t MyHome/Bathroom/cmnd/sonoff_basic_1/POWER -r -n
mosquitto_pub -t MyHome/Bathroom/cmnd/sonoff_basic_1/POWER1 -r -n
```
But still the same behaviour. I guess I need to get some more logs on the mqtt broker to see what's going on there... 

Do you happen to know how to remove all the retained messages from all topics in mosquitto, just to make sure there's nothing left?

BR,
Mikko

arendst, 2017-09-20:
Don't know but I do use the following command to monitor any activity:
```
mosquitto_sub -t stat/# -t tele/# -t cmnd/# -v
```
and initially it shows all retained commands known to the broker.

In your case it would have been
```
mosquitto_sub -t MyHome/Bathroom/stat/# -t MyHome/Bathroom/tele/# -t MyHome/Bathroom/cmnd/#  -v
```

tmikko, 2017-09-20:
Aha, now we are talking;

```
> mosquitto_sub -t MyHome/Bathroom/stat/# -t MyHome/Bathroom/tele/# -t MyHome/Bathroom/cmnd/#  -v
MyHome/Bathroom/tele/sonoff/LWT Offline
MyHome/Bathroom/tele/sonoff_basic_1/LWT Online
MyHome/Bathroom/cmnd/sonoff_basic_1/power ON
```
Sometimes small stuff matters! :)

Thanks a lot for your help. I probably would have spent hours with this!

BR,
Mikko

closed 2017-09-20 by tmikko