arendst/Tasmota #891: Sonoff touch - physical switch delay

KaSroka, 2017-09-20:
I've flashed tasmota to my sonoff touch and I can see slight delay when using touch button. It's much faster to turn on the light by mqtt message than with touch button. Is there any option I can tweak? Stock sonoff firmware was much faster.

davidelang, 2017-09-20:
The problem is that tasmota lets you do different things if you press the button 
multiple times, so it needs to wait and see if you are only pressing it once or 
if you are pressing it multiple times.

There s an option that lets you disable this. I don't remember it at the moment

PeggyFree, 2017-09-20:
Let me help : 
**SetOption13 --> 1**   = Enable button single press only for immediate response. Disable by holding 4 x SetOption32

By the way, the "Disable by holding 4 x SetOption32" feature is curious.

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

jeylites, 2017-10-03:
@PeggyFree 

are you referring to this line to set it to "1"?

#define KEY_HOLD_TIME          40                // [SetOption32] Number of 0.1 seconds to hold Button or external Pushbutton before sending HOLD message

davidelang, 2017-10-03:
no, @PeggyFree is saying to issue the command "setoption13 1" to disable all the multipress options. This isn't a code change, it's a setting change.

jeylites, 2017-10-03:
Noted, will give it a shot...

jeylites, 2017-10-04:
@davidelang @PeggyFree @arendst 

Tasmoata excepted the command I ran "setoption13 1" but there is still a lag on the touch buttons. There are times when it will lock up the button for a few seconds. And other times especially when it comes out from being ideal, the button becomes less responsive and than it catches up few seconds later.

What can I do to resolve this?

jeylites, 2017-10-04:
I realize something while reading [#947](https://github.com/arendst/Sonoff-Tasmota/issues/947) , I did not set the MQTT IP address while running my button test. After setting the correct address it became responsive. My only concern is, what if MQTT looses connection? 

@ionciubotaru suggested there was an update on the MQTT library, I'm using the latest 5.8.0 downloaded from [Download Tasmota](https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE) regardless, it still looks up when MQTT fails.

jeylites, 2017-10-04:
@arendst 
 I noticed when all the switches are [ON] state and the Touch loose wifi connection eg: wifi turned off. The relays turn OFF and ON every few seconds. It is like the Touch is rebooting, looking for a valid wifi connection. I'm not sure if its a bug or it suppose to react this way?

davidelang, 2017-10-05:
There is a longstanding bug with the arduino libraries that if it has trouble 
making a connection, nothing else can take place.

alton76, 2017-10-05:
That explains  a lot, can a time out be setup, stop trying to communicate
after 5 attempts? And  re-attempt to connect after a reboot cycle.

jeylites, 2017-10-05:
@alton76 I was going to ask the same as well....

*Jey Kasi*

closed 2018-05-07 by stale[bot]