arendst/Tasmota #892: noob question about sonoff touch with tasmota controlling a warm/cool bulb?

kevinshane, 2017-09-20:
hey guys, sorry about the noob question, but I don't know how to set it up with tasmota, could u guys please tell me how to do
I have a sonoff touch with a warm/cool switchable bulb, the bulb has a funtion which allows me to switch between a warm and cool mode. So currently I have flashed the tasmota to a sonoff touch and I can turn the bulb on/off, it works like a charm, but how can I make the bulb "knows" that I want a different temp? Can I make some kind of "adjustment" to let the sonoff touch understand "the first touch is a warm, the second touch is a cool, the third touch is turn off" ??

PeggyFree, 2017-09-20:
Yes it is, search wiki with keywords switchmode, multipress, hold

kevinshane, 2017-09-21:
@PeggyFree thank you man, I've read the wiki page, but still I am really new to this, could u please give me some details about how exactly this switchmode and multipress works? It's really difficult for me, so sorry about the noob questions.

PeggyFree, 2017-09-21:
Off course : look at this discussion #200 , read from the bottom until you find the table "arendst commented on 6 Jun". Please read this until you understand, then you will be a multipress master.

enesbcs, 2017-12-02:
I read the table at [#200](https://github.com/arendst/Sonoff-Tasmota/issues/200#issuecomment-306617207) and if i interpret correctly, i can not use it in Domoticz MQTT?

closed 2018-05-07 by stale[bot]