arendst/Tasmota #893: 1 Channel Inching/Self-lockIing Mode WiFi Wireless Switch 5V/12V and Sonoff-Tasmota?

jannnfe, 2017-09-20:
Hallo,
sorry i never posted anything on GitHub but I have a question and can't find a solution at this time. I'm a noob and like to control my Garage Door with Siri and Homebridge. My Garage Door needs a pulse to open/close.
I bought the "1 Channel Inching/Self-lockIing Mode WiFi Wireless Switch 5V/12V" from [itead.cc](https://www.itead.cc/smart-home/inching-self-locking-wifi-wireless-switch.html) and like to use it with the Sonoff-Tasmota Firmware to connect via plugin to Homebridge.

Pictures of the Relay:
[image]
[image]

Is it possible to flash the Sonoff-Tasmota Firmware to this relay?
Where do i have to connect my USB adapter or can i just use the build in USB Port?
Where are the RX, TX, GIO Ports on this Relay?

I hope anybody can help me and sorry for my bad english.
Thanks!

ionciubotaru, 2017-09-21:
You can use Tasmota.
Connect the FTDI to the 4 pin cconector near the relay
One of 2 black buttons must be GPIO0

jannnfe, 2017-09-21:
Thank you for your reply.
Is this right in the picture?
[image]

After the flash is the Self-lockIing Mode still working?
Thanks!

ionciubotaru, 2017-09-21:
From left to right:
GND - TX - RX - 3V3
The square is 3V3

jannnfe, 2017-09-21:
Thanks!
I wrote with the itead.cc Support and they told me this device can not be flashing.
Is this right?

itsecasc, 2017-09-27:
any updates ? i have almost the same board, same problem
[image]

tomasroud, 2017-10-05:
Hi, my board does not have connected RX and TX to the chip, so I must used wires with needles and connect directly to the chip. TX and RX is 2nd and 3rd from to left corner. And be aware about changed 3,3 and gnd pin!!!!! First connect 3,3 and gnd and check without button presing if led go on. Fter that start with pressed button. Then press second button, to have both leds light on. Then you can flas with needles, I had so 70 trys, because my hand shaked a lot ;) P.S. I used autoclick with 20s delay to upload firmware ;)

tomasroud, 2017-10-05:
[image]
Here is the connect picture. That left to gpio0 is not necessary, button is working good. And for tx, rx use needles. 
[image]

jannnfe, 2017-10-05:
Related to @ionciubotaru must the occupancy look like this on my relay? (White):
[image]
@tomasroud you say it is possible to flash my relay? Does everything work the way it should or does complications occur?
Is the assignment at the top of the image or do I have to connect 3v3, RX, TX, GND in one of the three red rectangles? If yes, you can tell me which I must use and wich buttons when I must press. Many Thanks!

tomasroud, 2017-10-05:
Eat the elephant by pieces. First locate 3,3 + GND.
It should be GND and 3v3 as in your picture.
Try to connect if LED light up.
Got it? Next step.
Start flash mode ... must both of two LEDs light up.
Next step.
Locate TX + RX.
Try RX and TX on your picture .... try swap TX + RX. 
Still not working.
Try on rectangle nr 1 second and third from bottom.
And again try to swap second and third as TX + RX.
Tell how successfull you are. And be aware about switch 3,3 and GND. If you connect bad gnd to 3,3 and connect TX or RX, you just demage your USB programmer. As I twice :(

jannnfe, 2017-10-06:
Now i have tested that what you are explaining.
I found out, that this picture is the right assignment:
[image]

But i don't know how to come in flash mode. Can you tell me?

tomasroud, 2017-10-07:
Press one button nr. 1 or 2, try it and then connect to usb. If lights only 1 led, press next button till both of leds light.

Tell me, how it works.

P.S. When you connect 3,3 and GND, it will be too at bottom two pins on block nr. 3

jannnfe, 2017-10-08:
I use the same arrangement as at the beginning. In the beginning lights still lit. For some reason now lights nothing more when I connect via USB. I do not think this is broken. Can be connected where else GND RX TX and 3v3?

tomasroud, 2017-10-09:
3,3 and GND are bottom 2 pins in block nr. 3, try it there, if the board will live. Try with ohmmetter that these pins are same as 3,3V and GND.

jannnfe, 2017-10-11:
@tomasroud No light on the block 3 :(
I think the usb adapter is broken but it is light up on usb connect and it gives out 3,3v.

tomasroud, 2017-10-11:
USB adapter can be broken on TX or RX, on 3,3 and GND probably not. So it must be OK.
If you connect GND and 3,3 is voltage on block 3, bottom 2 pins?

tomasroud, 2017-10-11:
[image]
Here are TX and RX pins, as I wrote above.

jannnfe, 2017-10-12:
Idk if this is stupid but i connected 3,3v on board voltage and GND, RX, TX like this picture 
[image]
and now my mac recognize the board. The two light next to button 1 and 2 are flashing but in Arduino IDE 1.8.4 and 1.8.5 i get the same error like here: https://github.com/arendst/Sonoff-Tasmota/issues/957

tomasroud, 2017-10-12:
The board lives, that is cool, so, you should connect your USB adaptor to USB in pc in same time as btn. is pressed. Then both LEDs will light ... not flash, light still. OK?

jannnfe, 2017-10-31:
So I have now tried all sorts of combinations and pins and with nothing I can flash the Tasmota firmware. I have two pictures where you can see the LED light up. No idea what the flash mode is and if he is even in flash mode. Any ideas on how to connect to the tasmota firmware?
[image]
[image]

tomasroud, 2017-11-03:
You must flash with VCC and GND from USB in PINS only. I see you have connected main voltage, it is not possible to flash the device like that. Unplug micro USB voltage, connect voltage in pins and tell me, if you can set up both these pictures too.

jannnfe, 2017-11-04:
I'm not able to get in this light mode on picture 2 with VCC and GND connected :(

tomasroud, 2017-11-04:
Press btn1 when connected to USB and release. Then press btn 2. IT MUST STAY IN 2 LEDs light on.

jannnfe, 2017-11-08:
No it don't work. I think I have a Version of the Sonoff, that is not flash able :(

tomasroud, 2017-11-10:
I am wondered, but maybe it is possible. I am sorry, that I can not help you :(

ciotlosm, 2017-12-19:
This is how flashing mode should look like: https://www.youtube.com/watch?v=E0WBoFCcnYY

jannnfe, 2017-12-19:
Thank you for the Video. I can access this mode like in the video but im not able to flash the sonoff.
My Module ist the PSA-B01-GL

jannnfe, 2017-12-19:
Got it finally working!!!
I used Arduino IDE with ESP8285 an Flash Size : 1M (512k SPIFFS) and the latest Sonoff Firmware.
I connect VCC and GND to the pins shown above in the Photo and connect TX and RX to the Pins shown in this picture [https://user-images.githubusercontent.com/9718240/31437695-70e1f8be-ae86-11e7-94f3-d6b4c711a761.png](url). Hold the jumper cable to this pins without soldering it should work. Try swap TX and RX if its not working. After flashing you have to connect the USB to get into webinterface

badigit, 2018-01-11:
Sucsesfull flashed this with SonOTA

benjaminsonnleitner, 2018-01-12:
Yesssss also!!! but which device belongs in the configuration set? 
That's not the basic model
[https://www.youtube.com/watch?v=wSjzj-Imj0c](url)

[image]

ciotlosm, 2018-01-12:
Try to use 1 Channel (option 12)

jannnfe, 2018-01-14:
Now I have bought a new 1 channel module, which looks different again than what I had before. Since I already gained some experience, I was able to flash this module directly on the chip with RX and TX.

This Module:
[image]

However, I'm wondering now. Is there any GPIO on this module that you can connect a reed sensor, for example, or do I have to solder directly to the chip?

I found this by @LurchiStromberg in [this Issue](https://github.com/arendst/Sonoff-Tasmota/issues/453) about the PSF-B01 Chip:
[image]

Is there an easy to reach GPIO or RX TX on this module?
Thanks!

budisek, 2018-01-18:
Hey all,

please can you write any tutorial to flash and upload Tasmota firmware to 5V/12V relay?

I have right this type:

[image]

I am using FTDI 3.3v. GND and 3v3 connection is correct but everytime when i try to upload Tasmota i get an error:

> Uploading .pioenvs\sonoff\firmware.bin
> warning: espcomm_sync failed
> error: espcomm_open failed
> error: espcomm_upload_mem failed

I also tried swap TX and RX but it did not help.

Thanks

ciotlosm, 2018-01-18:
@budisek You need to connect tx and rx to the PSF-B01 correct pins. I have a video that has some details that might help: https://www.youtube.com/watch?v=vzoM4op60jo

jannnfe, 2018-01-18:
This Module is "hard" to flash.

1. Set Up Arduino IDE: https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE
2. Connect the Jumper Cables to this Sonoff:
[image]
3. Get into flashing Mode: Connect 3.3V while pressing both buttons and than press the bottom one like in this video: https://www.youtube.com/watch?v=E0WBoFCcnYY
4. Upload Tasmota Firmware with PulseTime enabled

You can just hold the jumper cables at the chip pins seen in this video: https://www.youtube.com/watch?v=vzoM4op60jo

budisek, 2018-01-18:
Great thats look not so difficult. 

Two questions:
In youtube video is used 5V or 3.3V mode?
And how can i enable pulsetime?

Thanks

ciotlosm, 2018-01-18:
I used USB 5v not the FTDI 3.3v. I also used only ground from the FTDI and tx/rx

ciotlosm, 2018-01-18:
@budisek in my video I used USB 5.5V and used FTDI for ground/tx and rx only.

benjaminsonnleitner, 2018-01-18:
Hallo! 

I doesn‘t Flash the module with FTDI...... I flashed over the Wifi with my MacBook our Windows .... its very easy .... Time what I needed: 2Minutes to Flash. Preparation: 15 minutes

budisek, 2018-01-18:
I did it. I flashed it by cables and Tasmota is online.

But I need to set the relay to 1 sec ON and then OFF. I changed this in firmware before uploading:

> #define APP_PULSETIME          10       

But it works strange. When I toggle ON, it crashes (led is blinking) and restart is necessary.

What I need to set the relay to ON for 1 sec?

jannnfe, 2018-01-18:
One time i had the same issue.
Its because Tasmota is not supporting the inching /Self-locking function of the module.
Maybe you are in the wrong mode. Press the Buttons and test different states. 
I also pressed the buttons and at some point it worked.
You can change PulseTime by typing "PulseTime 30" in the Console at the Webinterface.

budisek, 2018-01-18:
You are right. I switched mode with second button. Then it works like switch (switching between ON and OFF manually) but when i changed pulsetime, now it is ON only for adjusted time. Great.

Last question i hope:

Is it possible to set power on state to OFF always? I see that with every restart it turns ON for short time (less than 1sec) before it toggles to OFF. You know, i dont want to have the garage door opened after every restart :)

jannnfe, 2018-01-18:
Really thats the restart issue is strange. I don't have this problem.
I changed nothing in the user_config.h except PulseTime, Project Name and Wifi Options and my module is always off even after restart.
Did you configure MQTT? I think I've heard that some functions work better when an MQTT broker is configured.
Maybe someone else knows better about this issue.

budisek, 2018-01-19:
In fact, i don't have relay connected to garage door yet, so I am only guessing now. 

After every restart red LED (which blinks when I toggle ON) shortly lights and relay clicks twice. Maybe it is some part of restart and it will have no effect to opening. I will let you know.

Thanks for your help.

Alpengroove, 2018-01-22:
You can do the firmware update OTA without touching the module. 
[https://haus-automatisierung.com/hardware/sonoff/2018/01/04/sonoff-vorstellung-part-11.html](url)
The video is in german, but you can copy the commands.

Reikard80, 2018-01-22:
Dear All,

Does anywone can tell me where I find the GPIO14 on this board? would be great 

Rik

jannnfe, 2018-01-22:
> I found this by @LurchiStromberg in [this Issue](https://github.com/arendst/Sonoff-Tasmota/issues/453) about the PSF-B01 Chip:
[image]

Reikard80, 2018-01-23:
Hi Jannnfe,

Thanks a lot, indeed now I found gpio14 :) thanks for your support on this

Rik

salmanslim, 2018-02-23:
Wont this 5/12v work without flashing for garage door ?

Bull10X, 2018-03-08:
It will work but if you can get it flashed it will support an input like a reed switch to determine if your door is open or closed if you are not in line of sight of door.

Doug

sumasage, 2018-03-21:
How do I flash with pulsetime?  And what is pulsetime?

I do have adruino ide setup and able to flash tasmota to sonoff basic without any problem. 

Thanks

tuga49, 2018-03-24:
I have it working with tasmota. But how I add a reed switch to it? Thanks

rmarkle, 2018-05-02:
@tuga49 - Did you get this figured out?  I am trying to do the same thing.

salmanslim, 2018-05-02:
I am using it wothout flashing as i am using a smartthings sensor for open/close notifications.

closed 2018-07-08 by stale[bot]

gadLinux, 2018-11-24:
> This Module is "hard" to flash.
>
>     1. Set Up Arduino IDE: https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE
>
>     2. Connect the Jumper Cables to this Sonoff:
>        [image]
>
>     3. Get [...]

This board is marked with 7-32 AC/DC. Is it safe to power it up with AC power?

Martin-Dolan, 2018-12-28:
> This board is marked with 7-32 AC/DC. Is it safe to power it up with AC power?

Only if the AC is between 7 and 32 volts. I have mine powered by my door intercom transformer which is around 16-18v AC

mientki, 2018-12-31:
Hello
I ordered a similar WiFi module on Aliexpress. My one has an RF module, besides it seems very similar to Sonoff RF. Of course, it has ewelink software. I would like to reprogram it on tasmote, but I do not know how to enter it in flashing mode. Maybe someone was already trying to flash it? Will you help me?

[image]
[image]

Martin-Dolan, 2019-01-07:
It will be the same as the other modules. Power it on while holding gpi0 to ground for a few seconds

udayroy04, 2019-02-27:
Hello everyone , i am totally new on git , this board is very useful for me and i want to use this board with my own android application . I would be great full if some can help me find the android sdk for this board .

badigit, 2019-03-03:
> You are right. I switched mode with second button. Then it works like switch (switching between ON and OFF manually) but when i changed pulsetime, now it is ON only for adjusted time. Great.
>
> Last [...]

Have same problem. After soft-restart in webui, relay switch 1-0.
After hard-restart (via power on\off) - no problems, relay stay off

Bacon-Ranch, 2019-04-06:
hello,  please can anyone help me, I have tried every possible combination of wires and button holds as I power on, I am using ESP easy, lol. The Sonoff device is a 1 channel inching type, the serial device is a DSD tech FT232 TTL. I have looked for pin outs for the serial device and can not find, I have tried with the jumper on the RTS and CTS pins and without. I have tried 5v and 3.3V jumpers, I do not think I fried anything yet but the serial board did get a bit hot at first. Everything still seems to light up and at least is not completely dead yet, not even sure which button to hold down so I have tried both seperately. Please see the picture showing 2 boards, I get flash failed every time, so it at least attempts to connect. Orange and yellow are TX and RX, Red is +, brown is ground.

[image]
[image]
[image]

[28 more replies]