arendst/Tasmota #894: Pushbutton

siouxme, 2017-09-20:
Any chance to configure this firmare to act on the relay like a push button? E.g. mqtt command "on for 10ms" or "on for 500ms".
I'd like to use it like a software-based push-button. actually to turn off / on a lamp with a pulse. And next to dimm that light be simulating a long pulse.

davidelang, 2017-09-20:
see the pulsetime command

On Wed, 20 Sep 2017, siouxme wrote:

> Date: Wed, 20 Sep 2017 18:09:58 +0000 (UTC)
> From: siouxme <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24fbeebe1379eca1e91e1f49dcf6186769efbaf71bd92cf0000000115da70f
>     592a169ce0f73e40f@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [arendst/Sonoff-Tasmota] Pushbutton (#894)
> 
> Any chance to configure this firmare to act on the relay like a push button? E.g. mqtt command "on for 10ms" or "on for 500ms".
> I'd like to use it like a software-based push-button. actually to turn off / on a lamp with a pulse. And next to dimm that light be simulating a long pulse.
>
>

ionciubotaru, 2017-09-21:
see backlog to run a couple of commands in one MQTT message

Bambutscha, 2017-10-05:
Hi,
i also have a question about the PUSHBUTTON
Which Parameters must i Change to get These function?
I tryed a lot of Parameters without a result.

I Need i Switch Impulse for example 500ms.
I have a room with 4 Points to Switch a "current shock relay" for my light.

Please, which Parameters must i Change to get a Pushbutton Impulse for 500ms to Switch my "current shock relay" ???

Thanks a lot

ionciubotaru, 2017-10-06:
check the wiki for `Pulsetime`

Bambutscha, 2017-10-06:
OK, i think i testet it with it.
Now i`m not 100% shure...
5  * 0.1sec = 0.5sec are 500ms
#define APP_PULSETIME          **5**                 // [PulseTime] Time in 0.1 Sec to turn off power for relay 1 (0 = disabled)

But is that all ? Or must i also Change this one from **TOGGLE** to **PUSHBUTTON** ???

#define SWITCH_MODE            **PUSHBUTTON**        // [SwitchMode] TOGGLE, FOLLOW, FOLLOW_INV, PUSHBUTTON, PUSHBUTTON_INV, PUSHBUTTONHOLD or PUSHBUTTONHOLD_INV (the wall switch state)
Thank's for your Help. I Need this function so much.

ionciubotaru, 2017-10-06:
you can change the pulsetime via mqtt commands - this is a more flexibile and better way

Bambutscha, 2017-10-06:
I will only Programming it one time.
I don't know how i can send commands in mqtt. I have no Server or service running for that.

Question is,
#define APP_PULSETIME **5** 
...
#define SWITCH_MODE **PUSHBUTTON** 

is that OK, or must #define SWITCH_MODE **TOGGLE**

Bambutscha, 2017-10-07:
Hmmm...
i tryed different pulsetime and different Switch_mode
He do it not. He go on and stay on. He don't go  off after 2sec for himself.
Here my last Settings
`
// -- Application ---------------------------------
#define APP_TIMEZONE           1                 // [Timezone] +1 hour (Amsterdam) (-12 .. 12 = hours from UTC, 99 = use TIME_DST/TIME_STD)
#define APP_LEDSTATE           LED_POWER         // [LedState] Function of led (LED_OFF, LED_POWER, LED_MQTTSUB, LED_POWER_MQTTSUB, LED_MQTTPUB, LED_POWER_MQTTPUB, LED_MQTT, LED_POWER_MQTT)
#define APP_PULSETIME          20                 // [PulseTime] Time in 0.1 Sec to turn off power for relay 1 (0 = disabled)
#define APP_POWERON_STATE      3                 // [PowerOnState] Power On Relay state (0 = Off, 1 = On, 2 = Toggle Saved state, 3 = Saved state)
#define APP_BLINKTIME          10                // [BlinkTime] Time in 0.1 Sec to blink/toggle power for relay 1
#define APP_BLINKCOUNT         10                // [BlinkCount] Number of blinks (0 = 32000)
#define APP_SLEEP              0                 // [Sleep] Sleep time to lower energy consumption (0 = Off, 1 - 250 mSec)

#define KEY_HOLD_TIME          40                // [SetOption32] Number of 0.1 seconds to hold Button or external Pushbutton before sending HOLD message
#define SWITCH_MODE            PUSHBUTTON_INV        // [SwitchMode] TOGGLE, FOLLOW, FOLLOW_INV, PUSHBUTTON, PUSHBUTTON_INV, PUSHBUTTONHOLD or PUSHBUTTONHOLD_INV (the wall switch state)
#define WS2812_LEDS            30                // [Pixels] Number of WS2812 LEDs to start with

#define TEMP_CONVERSION        0                 // [SetOption8] Return temperature in (0 = Celsius or 1 = Fahrenheit)
#define TEMP_RESOLUTION        1                 // [TempRes] Maximum number of decimals (0 - 3) showing sensor Temperature
#define HUMIDITY_RESOLUTION    1                 // [HumRes] Maximum number of decimals (0 - 3) showing sensor Humidity
#define PRESSURE_RESOLUTION    1                 // [PressRes] Maximum number of decimals (0 - 3) showing sensor Pressure
#define ENERGY_RESOLUTION      3                 // [EnergyRes] Maximum number of decimals (0 - 5) showing energy usage in kWh`

davidelang, 2017-10-07:
did you change CFG_HOLDER each time you flashed it ith a different config? (or 
take other actions to make sure the new config took effect)?

Bambutscha, 2017-10-07:
#define CFG_HOLDER             0x20161209
#define SAVE_DATA              1
#define SAVE_STATE             1
You mean this one ?

**No, i didn't Change something from this Parameter.**
Everytime i make a complete new Image with ATOM and Flash it again with the smal USB-Reader/writer

But i Change sometimes the NAME
#define FRIENDLY_NAME        "Elefant"
to take a look, that the new config are realy inside

I'm not really a programmer. A small "user"...
May be somebody find a working confihuration and can post here
the Parameters that he changed to make the Standard sonoff Switch like a pushbutton to Switch my "current shock relay"

These two Parameters are only changed everytime to try to make it happen:
#define APP_PULSETIME 5 sometime i test it with 20 (20*0.1sec = 2sec)
 ...
 #define SWITCH_MODE PUSHBUTTON

davidelang, 2017-10-07:
On Fri, 6 Oct 2017, Bambutscha wrote:

> #define CFG_HOLDER             0x20161209
> #define SAVE_DATA              1
> #define SAVE_STATE             1
> You mean this one ?

yes that one, if you don't change it, tasmota assumes the existing config on the 
device is valid and keeps using it.

> **No, i didn't Change something from this Parameter.**
> Everytime i make a complete new Image with ATOM and Flash it again with the smal USB-Reader/writer
>
> But i Change sometimes the NAME
> #define FRIENDLY_NAME        "Elefant"
> to take a look, that the new config are realy inside

do a 'status 0' and make sure the parameters you are trying to set are really 
set.

Bambutscha, 2017-10-07:
Ohhh davidelang
You are my **## HERO** !

It works !
#define APP_PULSETIME          10 (it is OK 1sec Impulse to the "current shock relay")

I send the new Image again to the sonoff and modified also the line
#define CFG_HOLDER 0x20170001

Now everytime i want to Change some Settings inside, i Change the number one up
#define CFG_HOLDER 0x20170002
and so on... #define CFG_HOLDER 0x20170003....

I think normaly i don't must reconfigurate something... but if when... i know that now, how does it works.
Thanks again !!!

Last question
But why it works when i Change only on the WEBINTERFACE for example
the #define FRIENDLY_NAME or the #define EMULATION
These Settings are also saved or not ?

davidelang, 2017-10-07:
anything you change in the web interface or via mqtt is saved

when you change the settings in the code, that doesn't directly change the 
settings on the running machine when you load it. What happens is that when it 
boots up, the firmware looks to see if it has a valid config (is it an upgrade 
to an older tasmota version), and if the CFG_HOLDER value is in the right 
place,it assumes that the existing config is valid. If it doesn't find the right 
value, it assumes that this is not an upgrade and takes the compiled-in config 
data and writes it out to the config area.

VinceMasuka, 2017-10-13:
Pulsetime is pretty neat when it comes to triggerin relays (just like a pushbutton). It's like the code comments says: "[PulseTime] Time in 0.1 Sec to turn off power for relay 1 (0 = disabled)". So it's the time power is turned off and the relay finally triggers. Unfortunately that gives the Sonoff the state power = off. Although it's technically right it says not whether the bulb is on or off (since the Sonoff is always off after triggering a lamp on or off). Regarding voice control via Echo Dot it's quite uncool, as you can't say "turn OFF the light", you always have to say "turn ON the light" (no matter if you want it on or off).
So you see where this is leading to? Suggestions?

VinceMasuka, 2017-10-15:
OK then:
See... The pulse is triggering a relay. Therefore it has to pulse "on and then off again" the relay in order that the relay changes it's state (on or off, depending on the previous state). So the relay might be triggered on or off, but after the pulse the Sonoff is always off. And so it's recognized by the Echo Dot.

Therefore if you say  "turn on the light" it triggers the Sonoff to pulse on and off again, just to have the relay triggering. The other way around you can't say "turn on the light" as the Sonoff is recognized as off anyway. Technically it's ok, but practically I would appreciate the option that Sonoff could somehow remember the last pulsing in order to turn the relay on or off, so it could cast it's state "right".

Might look strange to you, but just think of a light bulp in a room where you have like three switches around just to trigger that bulp. Therefore you install pushbuttons in that room that just trigger that relay that actually triggers that bulp. All the pushbutton do have to provide current just like 0.1 sec or so to trigger the relay.

closed 2018-05-07 by stale[bot]