arendst/Tasmota #895: sonoff-tasmona 5.7.1j does not send cmnd messages by mqtt when the physical button is pressed

damocles-git, 2017-09-20:
Hi,

I'm using the last version of sonoff-tasmona (5.7.1j)  with mosquitto 1.4.14. 

I disabled flash writes (#define SAVE_STATE 0) and instead use the retain feature of mqtt, but whilst messages issued with the mosquitto command line send a cmnd message with the retain flag activated, the ones sent by sonoff-tasmona (by pushing the physical button) does not send the cmnd message.

using the mosquitto command line:
<pre>
$ mosquitto_pub -d -i mosquitto_cli_pub -r -q 0 -t cmnd/sonoff/POWER -m TOGGLE
Client mosquitto_cli_pub sending CONNECT
Client mosquitto_cli_pub received CONNACK
Client mosquitto_cli_pub sending PUBLISH (d0, q0, r1, m1, 'cmnd/sonoff/POWER', ... (1 bytes))
Client mosquitto_cli_pub sending DISCONNECT

# journalctl -n 99 -f
Sep 20 20:27:59 server mosquitto[8176]: Sending CONNACK to mosquitto_cli_pub (0, 0)
Sep 20 20:27:59 server mosquitto[8176]: Received PUBLISH from mosquitto_cli_pub (d0, q0, r1, m0, 'cmnd/sonoff/POWER', ... (1 bytes))
Sep 20 20:27:59 server mosquitto[8176]: Sending PUBLISH to sonoff-01 (d0, q0, r0, m0, 'cmnd/sonoff/POWER', ... (1 bytes))
Sep 20 20:27:59 server mosquitto[8176]: Received DISCONNECT from mosquitto_cli_pub
Sep 20 20:27:59 server mosquitto[8176]: Received PUBLISH from sonoff-01 (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (15 bytes))
Sep 20 20:27:59 server mosquitto[8176]: Received PUBLISH from sonoff-01 (d0, q0, r1, m0, 'stat/sonoff/POWER', ... (3 bytes))
</pre>

using the physical button:
<pre>
Sep 20 20:34:27 server mosquitto[8176]: Received PUBLISH from sonoff-01 (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (15 bytes))
Sep 20 20:34:27 server mosquitto[8176]: Received PUBLISH from sonoff-01 (d0, q0, r1, m0, 'stat/sonoff/POWER', ... (3 bytes))
</pre>

there is not cmnd message so this state is not retained by mosquitto. 

As result the state is retained when I use the cmnd/ topic but not when the button is pressed.

am I missing something? or is there a bug in the program?

damocles-git, 2017-09-21:
how to reproduce:

- disable saving status to flash (#define SAVE_STATE 0)

stefanbode, 2017-09-23:
This only happens, if you retain the cmd messages. I really recommend not to do this. I also see it as a bad behavior that the device WRITE to its own cmd queue. Normally this should not be necessary. I have set the permissions on Mqtt to avoid this in general. I also do not store the status, because the device know it normally, if you don’t play with the savestate.

damocles-git, 2017-09-25:
Although the number of writes in flash is in the range of tens of thousands, I would like avoid replace switches in difficult access areas by this motive or simply to preserve the flash. 

I think saving the state at the broker level is a valid configuration, perhaps the correct behaviour would be to read (and respect) the retained status in the "stat" topic.

damocles-git, 2017-09-26:
Although the number of writes in flash is in the range of tens of thousands, I would like avoid replace switches in difficult access areas by this motive or simply to preserve the flash. 

I'm a little confused with how sonoff-tasmona behaves, maybe I'm not understanding who it works.

if I want power-on the switch from the command line I use:

<pre>
mosquitto_pub -d -i mosquitto_cli_pub -r -q 0 -t cmnd/sonoff/POWER -m TOGGLE
</pre>

with this command the switch change the current state and it is retained, now if I disconnect the switch and connect it again it get the state retained in the topic "cmnd/sonoff/POWER", the corresponding log is:

<pre>
Sep 25 20:33:52 server mosquitto[15451]: 1506364432: New client connected from ::1 as mosquitto_cli_pub (c1, k60).
Sep 25 20:33:52 server mosquitto[15451]: 1506364432: Sending CONNACK to mosquitto_cli_pub (0, 0)
Sep 25 20:33:52 server mosquitto[15451]: 1506364432: Received PUBLISH from mosquitto_cli_pub (d0, q0, r1, m0, 'cmnd/sonoff/POWER', ... (6 bytes))
Sep 25 20:33:52 server mosquitto[15451]: 1506364432: Sending PUBLISH to sonoff-01 (d0, q0, r0, m0, 'cmnd/sonoff/POWER', ... (6 bytes))
Sep 25 20:33:52 server mosquitto[15451]: 1506364432: Received DISCONNECT from mosquitto_cli_pub
Sep 25 20:33:52 server mosquitto[15451]: 1506364432: Client mosquitto_cli_pub disconnected.
Sep 25 20:33:52 server mosquitto[15451]: 1506364432: Received PUBLISH from sonoff-01 (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (14 bytes))
Sep 25 20:33:52 server mosquitto[15451]: 1506364432: Received PUBLISH from sonoff-01 (d0, q0, r1, m0, 'stat/sonoff/POWER', ... (2 bytes))
</pre>

however if I press the physical button, I get:

<pre>
Sep 25 20:33:09 server mosquitto[15451]: 1506364389: Received PUBLISH from sonoff-01 (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (15 bytes))
Sep 25 20:33:09 server mosquitto[15451]: 1506364389: Received PUBLISH from sonoff-01 (d0, q0, r1, m0, 'stat/sonoff/POWER', ... (3 bytes))
</pre>

and if I power cycle the switch the status it forgets the previous state.

How am I supposed to do to retain the state without writing to the device flash?

stefanbode, 2017-09-27:
You should use a tool like mqtt-spy-1.0.0.jar to get a clear view what is published to you MQTT and what messages are currently there if the device reconnects. As far as I understand the TOGGLE will be replaced in DEFAULT configuration with ON/OFF to save the state to MQTT. The TOGGLE and starttime just TOGGLE the device again with an unknown state afterward. There is a known limitation of PUBSUBCLIENT. It can only execute the FIRST retained message if there are more in the cmd/# topic. This is not a limitation of MQTT, but as far as I have understood of the ESP8266 and PUBSUBCLIENT. Therefore the switch to the required state ONLY works if there are no more retained messages. The debuglog on the ESP also helps to understand what is coming in and what not. In some rare cases also this first message is dropped. Therefore there is no guarantee, that a stored message on the MQTT server will be executed at startup. This is also a reason, why I suggest NOT to use the MQTT server to retain status. It is handy, but the limitations of the ESP8266 do not allow a professional use of it.

davidelang, 2017-09-28:
it would be a very bad thing for a device to send a cmnd/ message when it turns 
on, because the mqtt server would send that message back to the device, and it 
would execute it again (and since what the button does is a 'toggle' action, 
this would turn the relay back off almost immediately)

what you can do is set buttontopic so that the button is decoupled from the 
relays and doesn't change anything locally, it just sends a message to the mqtt 
server. Your home automation software can then process the message and issue a 
matt message to the device that switches the relay.

damocles-git, 2017-09-29:
Hi Stefan,

I'm not 100% sure but I think that retaining the last message at the broker is by design and not a limitation of the PUBSUBCLIENT:

http://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages
[...]
_"For each topic only one retained message will be stored by the broker."_
[...]
_So retained messages can help newly subscribed clients to get a status update immediately after subscribing to a topic and don’t have to wait until a publishing clients send the next update._

_In other words a retained message on a topic is the last known good value, because it doesn’t have to be the last value, but it certainly is the last message with the retained flag set to true._
[...]

stefanbode, 2017-09-29:
Ok, sorry, I did not communicate it in a proper way. Yes, retaining a message is by design. No problem. The limitation is on subscribing to a WILDCARD of messages like cmd/<mydevice>/#. Here it could happen that there are more than ONE message is available (EG: POWER1, POWER2, POWER3,....) . And here the PUBSUBCLIENT just delivers the first one in the current design; if he delivers one.

damocles-git, 2017-09-29:
Stefan, thank you for the clarification.

It seems that the solution is cited in:

https://github.com/arendst/Sonoff-Tasmota/wiki/Tips
[...]
_To stop saving power changes only to Flash or Spiffs use command SaveState off. This will disable the relay from returning to the same state after power on UNLESS you use the MQTT retain flag in which case the MQTT broker will send the last known MQTT state on restart or power on. The command **ButtonRetain on** will configure the button to send a MQTT command with Topic and the MQTT retain flag set._

I was missing the ButtonRetain=on setting. 

What confuses me now is that with SwitchMode=0, SaveData=0 and ButtonRetain=on  I see this when I press the physical button is:

<pre>
Sep 29 20:03:49 server mosquitto[21301]: 1506708229: Received PUBLISH from sonoff-01 (d0, q0, r1, m0, 'cmnd/sonoff/POWER', ... (2 bytes))
Sep 29 20:03:49 server mosquitto[21301]: 1506708229: Sending PUBLISH to sonoff-01 (d0, q0, r0, m0, 'cmnd/sonoff/POWER', ... (2 bytes))
Sep 29 20:03:50 server mosquitto[21301]: 1506708229: Received PUBLISH from sonoff-01 (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (14 bytes))
Sep 29 20:03:50 server mosquitto[21301]: 1506708229: Received PUBLISH from sonoff-01 (d0, q0, r1, m0, 'stat/sonoff/POWER', ... (2 bytes))
</pre>

the switch is publishing in its own cmnd topic which is contrary to the advise of Stefan and David .. ¿?

closed 2018-05-07 by stale[bot]