arendst/Tasmota #896: Upload Error 255 (espcomm failed)

chris-008, 2017-09-20:
Hi guys,
Pretty new in this scene but with a big eager to learn I started this small homebridge-sonoff tasmota project.
Last step in the whole project is to flash my sonoff devices (material got delivered quite late :-).
I am using an FT232R at 3,3V and the latest Tasmota package V5.7.1. ATOM Platformio to get the softwareflashed on the sonoff device.
After selecting the proper serial monitor, I managed to get the sonoff environment to install with success.
However, the sonoff-NL, -minimal and -ds18x20 are giving Error 255 (espcomm_ failed)
Any idea what I am doing wrong here? Thanks in advance for your kind help!

[image]

Chris

joecotton, 2017-09-20:
By default, the PlatformIO settings compile and upload 4 variants on the firmware. See the Wiki for how to select only one:
https://github.com/arendst/Sonoff-Tasmota/wiki/PlatformIO#change-ide-parameters

chris-008, 2017-09-21:
Hello Joe, thanks a lot for your quick reply... Looking at it now, it makes perfect sense ;-)

closed 2017-11-06 by chris-008

Kepete, 2018-01-25:
I'm experiencing the same error: 

[image]

I did select only one to upload: 

[image]

What am I getting wrong?

wizzor, 2018-03-17:
I have the same issue, trying to flash the same on Sonoff Touch.

robinmanuelthiel, 2018-06-30:
Which `upload_speed` are you using? For me, changing it to baud rate 115200 fixed the problem. In `platformio.ini`, I changed line `upload_speed = 512000` to `upload_speed = 115200`.