arendst/Tasmota #898: Sonoff POW : Measurements when Off

Th1h4ck, 2017-09-21:
Hello,
First, thanks for all this work, this firmware is great.

I modified a POW Sonoff to control my stove pellets: the relay became dry contact, and I wanted to keep the power measurement to simulate the state of the pan with the different consumption (start, run, stop).
Unfortunately, the current messure only works when the relay is on.
To start and stop my stove, I have to send a 4 second pulse (On, wait 4s, off). So, I have no consumption when the relay is returned to off.
Is there a way to force the power measurement even when the relay is off, or where is it a physical limitation of the card?

Th1h4ck
(sorry for my bad english)

arendst, 2017-09-21:
See #796

Th1h4ck, 2017-09-24:
Thank you !!!

I modified the file xsns_hlw8012.ino line 228 to force the computation of w, u, i and c if power is off :)

[img]http://reho.st/self/d81a8378839558fcb2ca070cc1be9bea61e67655.png[/img]

closed 2017-09-28 by Th1h4ck

PeggyFree, 2017-10-01:
Hello, would you mind sharing the modifications to get a dry contact on the pow?

PeggyFree, 2018-01-24:
Hello @Th1h4ck , may I ask you  how you did hack the pow to get a dry contact?

Th1h4ck, 2018-01-24:
Hello,

This is my mod : 
[url=https://imgur.com/s3ToVjq][img]http://i.imgur.com/s3ToVjqt.png[/img][/url]
Red : Cutting
Blue : Soldering

PeggyFree, 2018-01-24:
THank you very much. I did slightly differently as I wanted to separate the dry contact from the line as much as possible. I will upload pics asap.

PeggyFree, 2018-01-24:
[url=https://reho.st/view/self/8e2e948c04cffafc5ef11f20ef9d58c806daef46.jpg][img]https://reho.st/self/8e2e948c04cffafc5ef11f20ef9d58c806daef46.jpg[/img][/url]

[url=https://reho.st/view/self/ed5f8ddefd4fcd5fc4ea7c304a89f3ea4f5a300a.jpg][img]https://reho.st/self/ed5f8ddefd4fcd5fc4ea7c304a89f3ea4f5a300a.jpg[/img][/url]

[url=https://reho.st/view/self/5f88fa8914eb4cfd02dff853c5dc44e706bb1eec.jpg][img]https://reho.st/self/5f88fa8914eb4cfd02dff853c5dc44e706bb1eec.jpg[/img][/url]

[url=https://reho.st/view/self/8cccc6d648e19db5f69a9f2e02346baad7d41025.jpg][img]https://reho.st/self/8cccc6d648e19db5f69a9f2e02346baad7d41025.jpg[/img][/url]