arendst/Tasmota #899: New user configuration is missing after second flash

papaiatis, 2017-09-21:
I flashed the latest Tasmota firmware to my Sonoff WiFi Relay on Windows 7 with Arduino IDE.
I forgot to update the user_config.h (because it was not mentioned in the Wiki...) so I went back to Arduino IDE and set my SSID and password accordingly then flashed the firmware again.
I connected to the chip with Serial Monitor and I saw that it still tries to connect to the old, default APs: `indebuurt1` and `indebuurt2`. There is no `indebuur` string at all in my entire sketch folder.

What am I doing wrong?

papaiatis, 2017-09-21:
Figured, I need to change the CFG_HOLDER string in user settings. 
This doesn't seem to be documented as well.

davidelang, 2017-09-21:
please update the documentation, I don't know how many more places we need to 
put this than where it currently is. If you can add it wherever you looked for 
this, please do.

papaiatis, 2017-09-22:
It's done. Will update the Wiki wherever I feel the need of it. :)
Thanks.

closed 2017-09-22 by papaiatis