arendst/Tasmota #901: Security issue

NMGAndy, 2017-09-22:
Just noticed my 2 sonoff basic devices lost wifi connection to router and created open Access Point. The problem is anybody can connect to the AP and see password of my WIFI network in web page code.
Why it happens?? I know there is a feature to configure WIFI settings this way. I used the feature before by clicking 4 short button clicks. But this time nobody pressed the button.

ionciubotaru, 2017-09-22:
set `wificonfig 4`

NMGAndy, 2017-09-22:
I see. Thank you.
A im sure such things should be active by default. It is very surprising that my wifi password shown to everybody.

davidelang, 2017-09-22:
well, it's shown to everybody who can connect to the device, which either means 
they have physical access to use the serial port, or they are connecting over 
the network, which means that they already have access to your network and the 
wifi password doesn't get them more access.

ionciubotaru, 2017-09-22:
In production  mode a good practice is to set password to your local mqtt server and to disable web interface.

stefanbode, 2017-09-23:
Agree. Then serial is the only weak point for outdoor devices. In my mod I disable serial input on seriallog 0. Then it’s getting quite harder to extract information.

closed 2018-05-07 by stale[bot]