arendst/Tasmota #902: PowerOnState - inprovement

ionciubotaru, 2017-09-22:
`PowerOnState` is a global option. For multiple relays systems (Electrodragon) a `PowerOnState<x>` will be a better option (similar with `PulseTime<x>`)

davidelang, 2017-09-22:
why would you only want to remember part of your state when power bounces?

It may be useful, but it would help to explain a situation where you want to do 
different things with different relays.
kk

ionciubotaru, 2017-09-22:
On my electrodragon there are 2 relays - one to turn on the light and the second to turn on the water pump. Afer a power loss I like to keep the light OFF and the water pump ON. Because water pump is  important I cannot use mqtt server to store the relay state (after power loss the raspberry can hang on boot)

closed 2018-05-07 by stale[bot]