arendst/Tasmota #903: Sonoff POW with Tasmota firmware and Domoticz

Jerby13, 2017-09-22:
I have flashed the Sonoff POW with Tasmota V5.6.1k and it's working great with Domoticz, type Electric (Instant + Counter). This way I can measure Energy and Power usage in Domoticz. But how can I measure the Voltage, Current and PowerFactor in Domoticz which are also available on MQTT server? I cannot configure other idx sensors in Tasmota POW.

arendst, 2017-09-22:
Will consider

PeggyFree, 2017-09-23:
I am interested by this request too.

Jerby13, 2017-09-26:
Mr Arendst,
Excellent what a fast service, A and V are working perfect. I'm very impressed about the implementation and options that are build-in to Tasmota and can be configured online. Thanks a lot.

closed 2017-09-26 by Jerby13

PeggyFree, 2017-10-02:
Where do you check V and A in domoticz ? I just see power.

arendst, 2017-10-02:
1 - Domoticz > Setup > Hardware
2 - Virtual Sensors > Create Virtual Sensors
3 - Current: Select Sensor Type Ampere (1 Phase)
4 - Voltage: Select Sensor Type Voltage