arendst/Tasmota #905: How to communicate via HTTP Requests

luisfpinto, 2017-09-22:
Hello, 

Is there any way I could communicate with sonoff to turn on/off the devices and the rest of functionalities doing HTTP Requests? I would like to integrate Sonoff into our mobile application

Thank you so much!

lgnd33, 2017-09-22:
Hi,

you might want to check Command reference out - especially the "Web" part:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#web

---
//_

luisfpinto, 2017-09-22:
I see! Thank you for pointing me to that. 

I have checked that for example:

```http://sonoff/cm?cmnd=Power%20TOGGLE``` will execute toggle in the sonoff device. 

I suppose that ```http://sonoff...``` should point to the device IP Address right? Is there any discover method to use? Or should I enter it manually?

jetema, 2017-09-22:
You could assign a static IP address for each sonoff if you want to access them quickly, so
See commands on 
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wifi

luisfpinto, 2017-09-22:
I see, so I'm looking for a solution that would scan the wifi network looking for sonoff. To do so there should be a protocol like SSDP or MDNS. I assume that, this is not currently possible, right?

jetema, 2017-09-22:
I think you're right and actually is not currently available.

jetema, 2017-09-22:
I have seen there are on web interface ( tab on information ) mDNS Discovery  Enabled and as well mDNS Advertise  Web Server, so you could try scanning your sonoff's from a PC on the same network

luisfpinto, 2017-09-25:
I see thank you so much! I'll try going on that way! Thank you so much

joydeepsaha05, 2017-12-18:
@luisfpinto Did you create the mobile app? If so, could you share the link?

I was thinking of creating an app to control all your sonoff devices over the local WiFi as I couldn't find any existing ones.

davidelang, 2017-12-18:
There are a lot of apps out there that talk mqtt, and several different ones 
that will interface with OpenHab.

joydeepsaha05, 2017-12-18:
I don't have any hardware to set up a dedicated MQTT broker (I was wondering if you could run one using an Amazon EC2 instance?).

bontas69, 2019-07-21:
I have a device (https://tinycontrol.pl/en/lan-controller-v3-3/). This controller sends http toggle commands on some events to sonoffs with tasmota firmware.Thats till the 6.4.18 firmware.The syntax is like this: http://192.168.4.47/ay?o=0   The tasmota changes relay states on each http command,also responds ,see attach. After 6.5.0 firmware tasmota does not toggle and response different. Can be reactivated ,or it should be forgetten?
[image]
[image]

meingraham, 2019-07-21:
@bontas69 

I'm unsure as to how the command worked in the first place. This definitely was not a **_documented_** means of invoking a Tasmota command via http even as far back as 5.13.0. In any case, I recommend that you update your environment to use the current documented syntax to send the commands you want. The syntax is documented in the Commands wiki article.

Regards.

Mike

bontas69, 2019-07-21:
Thank you,Michael.Unfortunately,programming is not my area of expertise.I appreciate the sonoff project. The polish Lancontroller from the link above has the option to send http commands to some devices who accept this kind of controlling.Including tasmota, but in the form  http://192.168.4.47/ay?o=0. Does not work with 192.168.4.49/cm?cmnd=Power%201 or 192.168.4.49/cm?cmnd=Power%200 or 192.168.4.49/cm?cmnd=Power%20TOGGLE.I don't know why. Until(or at least 6.4.18). Flash a sonoff with 6.4.18 and try.The manufacturer specify the command in device user manual. See attach
[image]
Of course I want to update to 6.6.0 but I loose the remote control from Lancontroller.
Sorry to bother, the issue resides probably on my device...

meingraham, 2019-07-21:
@bontas69 

`Path2` can't be updated with the proper Tasmota command syntax?

I looked for the commit history of `xdrv_01_webserver.ino` but could not find any references to `/ay` as with the other webserver commands listed [here](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_01_webserver.ino#L515).

@arendst do you have any information of the `/ay` http syntax?

Mike

bontas69, 2019-07-21:
Unfortunately, does not work from LK3(the device).The sonoff receives "unknown command".Of course ,works from browser. Kind of Weird.
[image]
[image]

meingraham, 2019-07-21:
Turn up logging levels to see what Tasmota receives - `weblog 5` and perform the test again.

The fourth entry (server address 4, Port4, Path 4) isn't interfering, is it?

bontas69, 2019-07-21:
[image]

meingraham, 2019-07-21:
Now you need to figure out why `%20T` is being converted to `-600` by your utility.

Try this `cm?cmnd=Power TOGGLE` in your path parameter

bontas69, 2019-07-21:
Tried separate commands on off
[image]
and in LK3
[image]

meingraham, 2019-07-21:
```
0 / off = turn OFF 
1 / on = turn ON 
2 / toggle = if relay is ON switch to OFF and vice versa
```
Either `Power 2` or `Power Toggle` should work much the same as using `On/Off` vs. `1/0`

bontas69, 2019-07-21:
That's for sure my device tangles commands, thank you for assistance. I'll try more tomorrow. One more wish: Do some user friendly option to this scenario: one tasmota device with a temperature sensor or button/switch controlls another tasmota device,ie turn on/off the relay. This could be the fly on the moon project. Not knx , not mqtt, just http commands or something simple.Web gui preferably
[image]
Thanks for the weblog commands...one step ahead:)

meingraham, 2019-07-21:
The device with the sensor can send to another device using MQTT `Publish` or HTTP `WebSend`.

This `/Power-60032` is a clue. It's converting `%20` to `-600` and seems to be "eating" the next character. In the first try it "ate" the `T` from "TOGGLE". In this test, it "ate" the `%` from "%32". Not sure what your utility is doing when it's decoding/encoding the path string.

bontas69, 2019-07-21:
Too many beers in the IT department.
This is the log of the tasmota 6.4.18 device receiving http commands from LK3 with http://192.168.4.47/ay?o=0

22:07:26 CMD: weblog 5
22:07:26 MQT: u9149f/568/RESULT = {"WebLog":5}
22:07:26 CFG: Saved to flash at F9, Count 482, Bytes 3584
22:07:32 SRC: WebGui from 192.168.4.44
22:07:32 MQT: u9149f/568/RESULT = {"POWER":"ON"}
22:07:32 MQT: u9149f/568/POWER = ON
22:07:33 CFG: Saved to flash at F8, Count 483, Bytes 3584
22:07:33 WIF: Checking connection...
22:07:33 WIF: Connected
22:07:37 SRC: WebGui from 192.168.4.44
22:07:37 MQT: u9149f/568/RESULT = {"POWER":"OFF"}
22:07:37 MQT: u9149f/568/POWER = OFF
22:07:38 CFG: Saved to flash at F7, Count 484, Bytes 3584
22:07:53 WIF: Checking connection...
22:07:53 WIF: Connected

bontas69, 2019-07-21:
Websend it's on my desk

meingraham, 2019-07-21:
Log isn't showing what was received, just the result. Try setting `weglob 4`. Maybe 6.4.18 is confused by the higher logging level (I think `5` may be newer).

meingraham, 2019-07-21:
`In browser we have this when refresh http://192.168.4.47/ay?o=0`

What do you mean?

bontas69, 2019-07-21:
Yes, paste http://192.168.4.47/ay?o=0 in firefox, the response is {t}{t}ON.
Refresh page, relay toggle, response is {t}{t}OFF

meingraham, 2019-07-21:
Bizarre.

If you enter `192.168.4.47/cm?cmnd=Power%202` in Firefox, does the device toggle?

bontas69, 2019-07-21:
Yes.,toggle. 
[image]

meingraham, 2019-07-21:
OK, so the device is working as expected.

Back to basics for a bit...

What kind of device is the 6.4.18?
What kind of device is the 6.5.0?

Please paste the output of `Status 0` from both devices here.

bontas69, 2019-07-21:
6418 is sonoffSV with esp8285:
22:37:54 RSL: Group 0, Index 1, Command STATUS, Data 0
22:37:54 MQT: u9149f/568/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff SV"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:37:54 MQT: u9149f/568/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T03:47:28","StartupUTC":"2019-07-21T17:50:26","Sleep":50,"BootCount":32,"SaveCount":490,"SaveAddress":"F9000"}}
22:37:54 MQT: u9149f/568/STATUS2 = {"StatusFWR":{"Version":"6.4.1.8(fc1198c-sonoff)","BuildDateTime":"2019-01-14T12:01:43","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
22:37:54 MQT: u9149f/568/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":5,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HP",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
22:37:54 MQT: u9149f/568/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
22:37:54 MQT: u9149f/568/STATUS5 = {"StatusNET":{"Hostname":"Sonoff SV","IPAddress":"192.168.4.47","Gateway":"192.168.4.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.2","Mac":"80:7D:3A:68:6C:B7","Webserver":2,"WifiConfig":4}}
22:37:54 MQT: u9149f/568/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.ats.pl","MqttPort":1883,"MqttClientMask":"sonoff sv","MqttClient":"sonoff sv","MqttUser":"bonti69","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:37:54 MQT: u9149f/568/STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 21 21:37:54 2019","Local":"Sun Jul 21 22:37:54 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:09","Sunset":"20:43"}}
22:37:54 MQT: u9149f/568/STATUS10 = {"StatusSNS":{"Time":"2019-07-21T22:37:54"}}
22:37:54 MQT: u9149f/568/STATUS11 = {"StatusSTS":{"Time":"2019-07-21T22:37:54","Uptime":"0T03:47:28","Vcc":3.292,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HP","BSSId":"58:6D:8F:95:BD:78","Channel":13,"RSSI":100}}}
650 is Psfa85 core with esp 8285:
22:36:38 RSL: Received Topic /Status, Data Size 1, Data 0
22:36:38 RSL: Group 0, Index 1, Command STATUS, Data 0
22:36:38 MQT: u9149f/fb6/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff PSF A85"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:36:38 MQT: u9149f/fb6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:00:02","StartupUTC":"2019-07-21T20:36:36","Sleep":50,"CfgHolder":4617,"BootCount":57,"SaveCount":437,"SaveAddress":"F7000"}}
22:36:38 MQT: u9149f/fb6/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":5,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:36:38 MQT: u9149f/fb6/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":5,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HP",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
22:36:38 MQT: u9149f/fb6/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
22:36:38 MQT: u9149f/fb6/STATUS5 = {"StatusNET":{"Hostname":"PSF A85","IPAddress":"192.168.4.49","Gateway":"192.168.4.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.2","Mac":"60:01:94:84:FE:90","Webserver":2,"WifiConfig":4}}
22:36:38 MQT: u9149f/fb6/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.ats.pl","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_84FE90","MqttUser":"bonti69","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:36:38 MQT: u9149f/fb6/STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 21 21:36:38 2019","Local":"Sun Jul 21 22:36:38 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:09","Sunset":"20:43"}}
22:36:38 MQT: u9149f/fb6/STATUS10 = {"StatusSNS":{"Time":"2019-07-21T22:36:38","DS18B20":{"Temperature":23.7},"BMP280":{"Temperature":30.1,"Pressure":1003.3},"PressureUnit":"hPa","TempUnit":"C"}}
22:36:39 MQT: u9149f/fb6/STATUS11 = {"StatusSTS":{"Time":"2019-07-21T22:36:38","Uptime":"0T01:00:02","Vcc":2.982,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HP","BSSId":"58:6D:8F:95:BD:78","Channel":13,"RSSI":30,"LinkCount":1,"Downtime":"0T00:00:06"}}}
22:36:53 WIF: Checking connection...
22:36:53 WIF: Connected

meingraham, 2019-07-21:
Also, you have your SV set as `Module 1` which is the Sonoff Basic module. I know, it's working, but that's not exactly right.

Please send me the template for the 6.5.0 device.

Actually, it's 49 that is the new device... Are there any relays on your 6.5.0 device? If so, try this in Firefox - `192.168.4.49/cm?cmnd=Power%202`

Does this cause the 6.5.0 device to toggle?

bontas69, 2019-07-21:
Tried websend: IT WORKS!!!!
WebSend [192.168.4.47:80] Power TOGGLE
Please give me the syntax from this simple scenario: Push the button on one device1 >websend to  device 2> toggle relay on device2
This is my long time wishful desire: simple remote contact via wifi

bontas69, 2019-07-21:
6.5.0 works like a charm in firefox:

22:52:05 SRC: WebCommand from 192.168.4.76
22:52:05 RSL: Received Topic /Power, Data Size 1, Data 2
22:52:05 RSL: Group 0, Index 1, Command POWER, Data 2
22:52:05 MQT: u9149f/fb6/RESULT = {"POWER":"ON"}
22:52:05 MQT: u9149f/fb6/POWER = ON
22:52:06 CFG: Saved to flash at F6, Count 438, Bytes 3584
22:52:12 HTP: Command
22:52:12 SRC: WebCommand from 192.168.4.76
22:52:12 RSL: Received Topic /Power, Data Size 1, Data 2
22:52:12 RSL: Group 0, Index 1, Command POWER, Data 2
22:52:12 MQT: u9149f/fb6/RESULT = {"POWER":"OFF"}
22:52:12 MQT: u9149f/fb6/POWER = OFF
22:52:13 CFG: Saved to flash at F5, Count 439, Bytes 3584
22:52:14 WIF: Checking connection...
22:52:14 WIF: Connected

meingraham, 2019-07-21:
Rule 1 ON Button#State DO WebSend [192.168.4.47:80] Power TOGGLE ENDON
Rule1 1

meingraham, 2019-07-21:
Yes, but I thought the issue was your https://tinycontrol.pl/en/lan-controller-v3-3/ setup.

bontas69, 2019-07-21:
Maybe.
TRied rule 1 but not working:
22:54:43 CMD: Rule 1 ON Button#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON Rule1 1
22:54:43 SRC: WebConsole from 192.168.4.76
22:54:43 RSL: Received Topic /Rule, Data Size 73, Data 1 ON Button#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON Rule1 1
22:54:43 RSL: Group 0, Index 1, Command RULE, Data 1 ON Button#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON Rule1 1
22:54:43 MQT: u9149f/568/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
22:54:43 CFG: Saved to flash at F7, Count 492, Bytes 3584
22:54:53 WIF: Checking connection...
22:54:53 WIF: Connected
22:55:01 APP: Button1 multi-press 1
22:55:02 SRC: Button
22:55:02 MQT: u9149f/568/RESULT = {"POWER":"OFF"}
22:55:02 MQT: u9149f/568/POWER = OFF
22:55:02 CFG: Saved to flash at F6, Count 493, Bytes 3584
22:55:04 APP: Button1 multi-press 1
22:55:04 SRC: Button
22:55:04 MQT: u9149f/568/RESULT = {"POWER":"ON"}
22:55:04 MQT: u9149f/568/POWER = ON
22:55:05 CFG: Saved to flash at F5, Count 494, Bytes 3584
22:55:13 WIF: Checking connection...
22:55:13 WIF: Connected
22:55:33 WIF: Checking connection...
22:55:33 WIF: Connected

meingraham, 2019-07-21:
YOu have to enter those two line in separate console entries.

bontas69, 2019-07-21:
Capisco.Not working

23:00:43 MQT: u9149f/568/STATE = {"Time":"2019-07-21T23:00:43","Uptime":"0T04:10:17","Vcc":3.397,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HP","BSSId":"58:6D:8F:95:BD:78","Channel":13,"RSSI":80}}
23:00:52 CMD: Rule 1 ON Button#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON
23:00:52 SRC: WebConsole from 192.168.4.76
23:00:52 RSL: Received Topic /Rule, Data Size 65, Data 1 ON Button#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON
23:00:52 RSL: Group 0, Index 1, Command RULE, Data 1 ON Button#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON
23:00:52 MQT: u9149f/568/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
23:00:53 WIF: Checking connection...
23:00:53 WIF: Connected
23:01:08 CMD: Rule1 1
23:01:08 SRC: WebConsole from 192.168.4.76
23:01:08 RSL: Received Topic /Rule1, Data Size 1, Data 1
23:01:08 RSL: Group 0, Index 1, Command RULE, Data 1
23:01:08 MQT: u9149f/568/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
23:01:13 WIF: Checking connection...
23:01:13 WIF: Connected
23:01:17 APP: Button1 multi-press 1
23:01:17 SRC: Button
23:01:17 MQT: u9149f/568/RESULT = {"POWER":"ON"}
23:01:17 MQT: u9149f/568/POWER = ON
23:01:18 CFG: Saved to flash at F9, Count 498, Bytes 3584

meingraham, 2019-07-21:
The rule is not defined. The rule set is empty:

RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":511,**_"Rules":""_**}

Go to http://192.168.4.47 in your browser.
Open the Console
Enter the rule `ON Button#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON`
Enter `Rule1 1`

Push the button on your SV. The relay on the other device should toggle.

meingraham, 2019-07-21:
BTW, I'm not sure you have your SV set up properly. And I'd like to see your Template on the new device. You may have some lurking issues due to misconfiguration.

bontas69, 2019-07-21:
psf85 new device : 
23:16:06 RSL: Received Topic /template, Data Size 0, Data 
23:16:06 RSL: Group 0, Index 1, Command TEMPLATE, Data 
23:16:07 MQT: u9149f/fb6/RESULT = {"NAME":"Generic","GPIO":[17,255,255,255,255,255,255,255,21,255,255,255,255],"FLAG":1,"BASE":18}

bontas69, 2019-07-21:
[image]

meingraham, 2019-07-21:
Something is not right. Module 0 is for a Template... And your template does not match the module in the screenshot. The template has a button on GPIO0 and a relay on GPIO12. Your module GPIO assignments are completely different.

Also, your Sonoff SV is configured as a `Sonoff Basic (1)`. The SV ought to configured as a SV `(3)`

I'm concerned that you're going to be fighting ghosts if you don't have your setup understood properly.

bontas69, 2019-07-21:
Sorry. I'm dizzy now. You're right . the sonoff sv was configured as sonoff basic , from an older backup file. Mean time I change it to sonoff sv. But the rule does not send the web command to other device. I will try tomorrow in a slower pace. Must to succeed. Great achievement, thanks to your kind patience. WIFI REMOTE SWITCH is about to be born

meingraham, 2019-07-21:
Understood.

I'd rather transition this conversation to Discord. Look me up over there.

Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

bontas69, 2019-07-21:
Agree... Have a nice beer
Thanks a LOT

bontas69, 2019-07-21:
"WebSend [192.168.4.49:80] Power TOGGLE" pasted in console  toggles relay on remote device...
sonoff sv does not recognize button push? something with "button#state"?

bontas69, 2019-07-22:
Bingo! syntax error...
**rule1 ON Button1#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON**

sender log

09:16:06 CMD: rule1 ON Button1#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON
09:16:06 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":447,"Rules":"ON Button1#State DO WebSend [192.168.4.49:80] Power TOGGLE ENDON"}
09:16:14 RUL: BUTTON1#STATE performs "WebSend [192.168.4.49:80] Power TOGGLE"
09:16:14 RSL: stat/sonoff/RESULT = {"WebSend":"Done"}
09:16:50 RUL: BUTTON1#STATE performs "WebSend [192.168.4.49:80] Power TOGGLE"
09:16:50 RSL: stat/sonoff/RESULT = {"WebSend":"Done"}

receiver log

09:16:14 SRC: WebCommand from 0.0.0.0
09:16:14 RSL: Received Topic /Power, Data Size 6, Data TOGGLE
09:16:14 RSL: Group 0, Index 1, Command POWER, Data TOGGLE
09:16:14 MQT: u9149f/fb6/RESULT = {"POWER":"OFF"}
09:16:15 MQT: u9149f/fb6/POWER = OFF
09:16:15 CFG: Saved to flash at F4, Count 456, Bytes 3584
09:16:29 WIF: Checking connection...
09:16:29 WIF: Connected
09:16:51 HTP: Command
09:16:51 SRC: WebCommand from 0.0.0.0
09:16:51 RSL: Received Topic /Power, Data Size 6, Data TOGGLE
09:16:51 RSL: Group 0, Index 1, Command POWER, Data TOGGLE
09:16:51 MQT: u9149f/fb6/RESULT = {"POWER":"ON"}
09:16:51 MQT: u9149f/fb6/POWER = ON

meingraham, 2019-07-22:
Strange, 6.4.18 has all the same rules features as 6.5. But glad it works.

But what about the tinycontrol `/ay` issue?

arendst, 2019-07-22:
/ay has gone, foetsie, away, ins blaue hinein, disparu(e)

Has always been undocumented use. The correct used html interface is /cm as you found out.

meingraham, 2019-07-22:
@arendst - well, good to know that it wasn't bizarre behavior by the device. Just making use of a feature that was never intended to be used... and now gone.

@bontas69 Do you need your tinycontrol system? If so, you'll have to follow up with them on why the `%20` is being encoded as `-600`. Otherwise, you will not be able to make use of the `/cm` syntax.

Signing off.

Mike

bontas69, 2019-07-22:
### **REMOTE WIFI SWITCH** -with two Sonoff SV
tested with tasmota 6.5.0 (6.6.0 get's some strange behaviour ,restarts but not sure)
Sonoff SV supports 5-24V DC low voltage input, have pins for additional GPIO 4.5,14.
[image]

We'll use gpio4 configured as switch2 on sender
[image]
[image]
The pushbutton controls the relay as usual and the switch attached to sonoff will send commands to another tasmota device connected to the same wifi network.On the console paste the websend commands: 
rule1 on switch2#state=1 do WebSend [receiver ip:80] Power on ENDON
rule2 on switch2#state=0 do WebSend [receiver ip:80] Power off ENDON
enable the rules:
rule1 on
rule2 on

The rule above makes the switch open = relay on (close) . With the rules bellow switch open= relay off(open):
rule1 on switch2#state=1 do WebSend [receiver ip:80] Power off ENDON
rule2 on switch2#state=0 do WebSend [receiver ip:80] Power on ENDON

Application: An alarm PIR sensor when activated sends command wiressly to a camera with DI (digital input) to record ,or some NVR with digital inputs. Or some light, siren whatever...
Make it simple!
Enjoy

meingraham, 2019-07-23:
Since there are no switches on the Sonoff SV, you can define GPIO4 as `Switch1`

First pair:
`Rule1 on switch1#state do WebSend [receiver ip:80] Power %value% ENDON`

Second pair can be combined into one rule set:
`Rule1 on switch1#state=0 do WebSend [receiver ip:80] Power on ENDON on switch1#state=1 do WebSend [receiver ip:80] Power off ENDON`

bontas69, 2019-07-23:
Look's better...thanks
Small improvements:
Suppose the receiver is offline,the sender console look's like this:
07:41:21 RUL: SWITCH2#STATE=0 performs "WebSend [192.168.4.49:80] Power on"
07:41:27 MQT: u9149f/568/RESULT = {"WebSend":"Connect failed"}
07:41:28 RUL: SWITCH2#STATE=1 performs "WebSend [192.168.4.49:80] Power off"
07:41:35 MQT: u9149f/568/RESULT = {"WebSend":"Connect failed"}
Would be nice some red flag in webgui of the sender or some kind of warning(buzzer alarm on the available GPIO5). Some rule On {"WebSend":"Connect failed"} do gpio5#state=1 endon...More fancy would be 1sec gpio on then 5sec gpio off , intermitent like an ups alarm.
Another great feature would be the status of the additional switch (gpio4) on the web interface,preferably in different color. (Tasmota automatically assign a status of a new relay on webgui if it's assigned a new relay on the generic module--- ON/off status and toggle button---)

meingraham, 2019-07-23:
The web UI is not actually intended as the "automation" UI, It's really a setup & troubleshooting interface. Only relays will have the control buttons.

Tasmota is first and foremost built as an MQTT based replacement. This is intended to send information to a "hub" and have it be the automation controller... including the UI. This automation hub can then perform all of the functions you are thinking of.

Remember that the ESP has limited program memory. The firmware has to remain small. In particular, since OTA is #2 on why Tasmota became widely used, the firmware has to stay under 650K so that it and a minimal copy can both reside in the 1MB flash concurrently during the OTA process. Thus, unless there is core automation commands that **_must_** be done on the device, then the philosophy is to offload that to a hub.

Mike

P.S. On your PSF A85, you could invert your relay component so that it follow you switch#state.
P.P.S. Why do you have 8 relays defined?
P.P.P.S. It's probably time to stop spamming this issue ;-) These kinds of philosophical conversation need to happen on Discord. I'm sure Adrian will soon lock this issue ;-)

bontas69, 2019-07-23:
Apologize.I'm just a noob.I don't have enough time to read and learn all forum,faq,documentation.I don't know much about mqtt,I have few sonoffs tasmota paired with the Lancontroller, they host a mqtt broker where device sends data.I went to discord...omg,too complicated.Allow me to give short answers here:
PSF A85 is a tiny module based on ESP8285 (https://www.itead.cc/wiki/PSF-A85)
[image]
I made with PSF A85, an adjustable 3.3V dc-dc stepdown and td62084 IC(8ch Darlington Sink Driver) a flexible test board.The gpio from psf module are connected directly to the inputs of td62084.
[image]
Datasheet:The TD62083AFNG and TD62084AFNG are high−voltage,
high−current darlington drivers comprised of eight NPN
darlington pairs.
All units feature integral clamp diodes for switching inductive
loads.
Applications include relay, hammer, lamp and display (LED)
drivers.
The outputs of the TD62084 can drive directly relays, led's whatever.
For now I have only leds to check the output state.
When 8 relays are allocated in module configuration we have 8 channels,I was just testing the environment.
Thats the board,I have all the gpio's available...
[image]
I agree esp devices are limited in program memory.I agree that every real life application demands custom solution. Would be nice a database with tested, real ,succesful projects where from a non-technical person can emulate , with step by step instruction, and use the device for it's own needs.
I hope we'll continue conversation...maybe on discord. I'll try again...
Please elaborate:"On your PSF A85, you could invert your relay component so that it follow you switch#state".

meingraham, 2019-07-23:
Discord is just an interactive chat rather than this "stuttered" chat we've been having. Use this link to join the Tasmota server. It really is simple after you create your account.

[Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4)

Best to answer the relay invert question there to step through the setup interactively with you.