arendst/Tasmota #906: Cannot OTA update

hoekbrwr, 2017-09-22:
When I try to update by pressing 7 times or ginving a MQTT command it cannot find the OTAURL and keeps on trying until a time error resets the unit.
It tries to connect to  http://domus1:80/api/arduino and when I look into the wiki there are completely different addresses for OTA. I am missing something! I tried to change the OTA URL in user_config.h and rebuild the software but that did not work.

davidelang, 2017-09-22:
see the troubleshooting page on why the change to the config didn't work

hoekbrwr, 2017-09-22:
Seems rather complicated. Best is to erase the flashrom completely,  isn't it?

hoekbrwr, 2017-09-23:
I did erase the flash according to Troubleshooting. Be sure to have the latest esptool.py, otherwise command erase_region might not be available! After the first attempt with button 1 pressed 4 times and reset the unit was "dead", so this drags the unit  - out of the mudfields - again.

closed 2017-09-23 by hoekbrwr