arendst/Tasmota #907: Momentary Button on GPIO14

pgv123, 2017-09-23:
Can anyone advise whether a momentary action button can act as a toggle on GPIO14 on the basic Sonoff? I have no problem in setting up a latching switch and setting it to 09 Switch1 and everything works as intended.
I would also like the option to use a momentary action button which would cause the relays to change state every time it is touched exactly as the GPIO0 onboard switch does.

joecotton, 2017-09-23:
Assign a switch to GPIO14, and set it to switchmode 3 or 4.
https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic#conclusion-1

pgv123, 2017-09-23:
Thanks but for me, so far, that definitely does not work with a momentary button. Have tried both switchmodes 3 & 4. Have wired in a 10k external pull up. Have tried a couple of different sonoffs.

I have SwitchTopic set to 0 as I only want MQTT to publish the state of the relay.

Any other suggestions would be great. Thanks.

pgv123, 2017-09-24:
So what I hadn't done is change the configuration back to 09 Switch1 AFTER fitting the 10k external pull up resistor. It now appears to be working correctly!

Thanks for your help.

closed 2017-09-24 by pgv123

freaktech, 2017-10-31:
i want also to use 1 momentary switch button   not a switch with permanent open or close position

1 klick = stable ON 
1 klick again = OFF

als long as you click again

how is this possible

1 resistor between 3.3V and GPIO14  brings no difference

thank you tom

davidelang, 2017-10-31:
look at switchmode, set it to toggle

Softwaremaker, 2020-05-13:
Sorry pulling an old thread here. Using Tasmota and assign a switch to GPIO14, and set it to switchmode 3 - A momentary switch wired to GPIO14 and 3.3V changes power state but reverts back to previous once the button is released.

How do I get a momentary switch to behave like a toggle switch - toggle state everytime it is pressed ?

Thanks !