arendst/Tasmota #909: Relays Inverted won´t save last state, always come back state on

fabiangl, 2017-09-23:


arendst, 2017-09-23:
See #908

liorsu, 2017-09-23:
Plus 1 to this issue.
This happens to me also when not compiling with BE_MINIMAL.

petrfaitl, 2017-09-30:
Same here. There is actually mild difference between restarting and pulling power. Regardless, it can't be relied on in live environment. 

Observation:
**PowerOnState 0 - pulling power off > Relays restart OFF (pins with 3.3V) > UI says ON/ console state message ON**
**PowerOnState 0 - cmnd/RESTART 1 from console > Relays restart ON  (pins with 0V) > UI says ON/ console state message ON** 
PowerOnState 1 - pulling power off > Relays restart ON (pins with 0V) > UI says ON/ console state message ON
PowerOnState 1 - cmnd/RESTART 1 from console > Relays restart ON (pins with 0V) > UI says ON/ console state message ON

Hope it helps finding the issue.

arendst, 2017-09-30:
Pls try to change code in file sonoff.ino around line 2836 from
```
    if ((pin[GPIO_REL1 +i] < 99) && (digitalRead(pin[GPIO_REL1 +i]))) {
```
to
```
    if ((pin[GPIO_REL1 +i] < 99) && (digitalRead(pin[GPIO_REL1 +i]) ^ rel_inverted[i])) {
```
and report if it solves your problem.

petrfaitl, 2017-09-30:
Thanks Theo @arendst for looking at the issue. We seem to have a working solution!

BTW, it helps pulling power after the flashing routine to let the module re-cycle power. Mere restart command was still causing issue as in my point 1 above.
The line for me was 2827.

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst